helgeerbe/picframe #105: Query json object from webserver returns failure for fade_time

helgeerbe, 2021-04-17:
In my proof of concept for the overlays I misuse `fade_time` to change overlays.

I found fetch `http://picframe:9000/?all` returns

````
{
  paused: false,
  subdirectory: '',
  date_from: -2147389200,
  date_to: 2145913200,
  display_is_on: true,
  shuffle: true,
  fade_time: 10,
  time_delay: 200,
  brightness: 1,
  location_filter: ''
}
````

But  `http://picframe:9000/?fade_time` returns

````
{
  fade_time: 10,
  ERROR: 'Excepton:fade_time>could not convert string to float: ;'
}
````

Not sure where the error comes from. My javascript test code

````javascript
fetch('http://picframe:9000/?fade_time')  
    .then(res => res.json())
    .then(function(data) {
      if (data["fade_time"] == 10) {
        mainWindow.loadFile('clock.html');
        console.log(data);
      }
      else {
        mainWindow.loadFile('weather.html');
      }
    })
    .catch(function() {
      console.log("Can't connect to PictureFrame webserver.");
    });
````

paddywwoof, 2021-04-17:
Helge, yes it's probably a bit sloppy code in interface_http. Part of the problem is having some methods wrapped in setter/getter decorators and some not so it's a little messy. Here https://github.com/helgeerbe/picframe/blob/de3710a1ce38ace8f15666807912ff768af1607a/picframe/interface_http.py#L84 the getter is called, but there isn't anything to filter out GET without a value. i.e. this line https://github.com/helgeerbe/picframe/blob/de3710a1ce38ace8f15666807912ff768af1607a/picframe/interface_http.py#L92 should be inside a condition of value being there. Needs fixing.

paddywwoof, 2021-04-18:
Made some tweaks but put them in the missing folder branch - hopefully merge that soon to dev https://github.com/helgeerbe/picframe/compare/dev...missing_folder

helgeerbe, 2021-04-18:
Perfect! 👍

jgodfrey, 2021-07-05:
Can this be closed?

helgeerbe, 2022-01-05:
@paddywwoof Can this be closed?

paddywwoof, 2022-01-05:
Yes.

closed 2022-01-05 by paddywwoof