helgeerbe/picframe #106: Screen flicker on a 4K

sapnho, 2021-04-20:
I am currently testing a 4K 32 inches monitor, the BenQ PD3200U, and have noticed quite a bit of screen flicker during the transitions.

Are there any recommended settings for large 4K monitors to make the transitions as smooth as possible?

helgeerbe, 2021-04-20:
Hi @sapnho,
I tested it on my tv with a pi3 and had no problems. But I resized the images to its native resolution (3840x2160) which is 16:9.

sapnho, 2021-04-20:
But the Pi3 doesn't have 4K and is capped at 1920x1200. Or did you find a workaround?

My images are all sized to 3840x2160px but still there is flicker.

helgeerbe, 2021-04-20:
Argghh. You are absolute right.

If you like, you could try this https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=79330 But I haven't tested it.

sapnho, 2021-04-20:
Sorry, forgot to specify that I am connecting a Pi4 of course.

sapnho, 2021-04-20:
I just used a 4K test image to find out if there has been scaling and that seems to be the case.
This is how it should look:
[image]
And on my 4K monitor, I can't see the checkerboard pattern.
[image]

sapnho, 2021-04-20:
@paddywwoof Does this line in config still do anything:
`  #auto_resize: True                       # default=True, set this to False if you want to use 4K resolution on Raspberry Pi 4.`

I just wonder if there are any 4k settings that I may have overlooked.

paddywwoof, 2021-04-20:
I think there are two problems here. The first is that the GPU - or the fake KMS driver - struggles to render large numbers of pixels - at least I haven't been able to get it to run very fast. When I've run simple shaders (with few vertices) I've found the frame rate to be inversely proportional to the number of pixels (fragments) so in [this test](https://www.geeks3d.com/20190930/raspberry-pi-4-vs-raspberry-pi-3-cpu-and-gpu-benchmarks/) they get 188 fps for 800x400 which would imply that the 4K speed should be 188 * (800 * 400) / (3840 * 2160) = 7.25fps and that generally seems to be something near what you actually achieve. That the Pi4 is capable of running 4k video at normal frame rate makes me think that this is just a deficiency of the GPU - probably targetted for smaller screens such as phones and tablets.

The second problem is due to me taking out the option to turn auto_resize off. What happens now is that the max texture size is checked when the pi3d.Display is created [see here](https://github.com/tipam/pi3d/blob/d63a08a593d32281d860ebe824986f515ca75bc3/pi3d/util/DisplayOpenGL.py#L154) and if that value is bigger than the list of allowed widths [see here](https://github.com/tipam/pi3d/blob/d63a08a593d32281d860ebe824986f515ca75bc3/pi3d/Texture.py#L24) it gets appended.

In the case of the Pi 4 the max texture is 4096 (and 2048 for Pi3 and below) so when pi3d is fed an image 3840 pixels wide it resizes it to 2048. There doesn't seem to be any documentation on the permitted widths - I had to find the existing list (link above) by trial and error. So it might be that 3840 is a permitted width and should also be included in the list but I will have to write a program to go through all possible values and test them. I know that some value cause seg faults from  issues that people have raised in the past, so it might be a slow process to find the allowed ones.

I will also do some trials with GPU testing software to see if there is any scope for improvement there. Otherwise the only way to stop the flicker is to make the change in alpha very small for each frame change i.e fade times three times as long as normal.

sapnho, 2021-04-20:
@paddywwoof Thanks for the explanation, Paddy. I'd be happy to test whatever genius way you come up with again! :-)

paddywwoof, 2021-04-25:
Useful experiment following from this resizing issue. I find (nearly ten years later) that the Texture widths can be any multiple of four up to max_texture_size! I will try changing pi3d to reflect that then test all the demo programs to make sure they all work.

sapnho, 2021-04-25:
Wow, Pi3D is already ten years old, and still the best around!

I dismantled my BenQ this week (couldn't help checking out the time-lapse video recording on my wife's phone: https://www.thedigitalpictureframe.com/watch-the-video-how-i-dismantled-the-benq-pd3200u-monitor-to-build-a-new-4k-digital-picture-frame-with-the-raspberry-pi/) and took it to the frame builder. With any luck i'll have it bank in ten days and will be ready to test whatever needs testing.

paddywwoof, 2021-04-25:
OK, exaggerating a bit. It was some time in 2012 - Tom made the first major refactor into different logical components [see here](https://github.com/tipam/pi3d/commits/master?after=d63a08a593d32281d860ebe824986f515ca75bc3+104&branch=master&path%5B%5D=pi3d&path%5B%5D=Texture.py)

Testing the resizing on Pi2 I noticed that it now takes ages to load - longer than it used to. I might have a look to see which change caused that.

Your wife is holding her phone *very* steadily in that video!

sapnho, 2021-04-25:
> Your wife is holding her phone very steadily in that video!

Indeed. As if she was a tripod! :-)

jgodfrey, 2021-04-26:
Nice video Wolfgang... I especially like those mid-disassembly picture frame tests... Almost like:

_Have I destroyed anything yet?
...
Nope, still seems OK.
Let's go a bit deeper
...
Have I destroyed anything yet?
...
Nope, still seems good!_

That's almost identical to my method... :)

Nice job!

sapnho, 2021-04-26:
> Have I destroyed anything yet?
> ...
> Nope, still seems OK.
> Let's go a bit deeper

@jgodfrey That is just too good! I took the liberty of adding your quote to the article. :-)

helgeerbe, 2021-04-26:
@sapnho nice job. I destroyed my first monitor. Gooing a little bit too deep. What are the sizes of the unmantled monitor?

sapnho, 2021-04-26:
> I destroyed my first monitor. Gooing a little bit too deep.

Did you ever find out what killed it?

> What are the sizes of the unmantled monitor?

If you mean the edges, it's 1 cm and the bottom one is 1.5 cm.

helgeerbe, 2021-04-26:
I tried to remove a metal frame around the panel. Unfortunately it was glued and when I pulled on it an air bubble appeared behind the panel surface. Na chance to remove it.

I was wandering about the depth. Panel plus eclectronic.

sapnho, 2021-04-26:
> I was wandering about the depth. Panel plus eclectronic.

4 cm if I remember correctly. It's currently with the frame builder.

jgodfrey, 2021-04-26:
In my case, I tore 3, 27" Dell monitors down to the bare panel and the 2 PCB's (power supply and video driver). The LCD panel itself is only about 1/4" thick (~7mm). However, after fastening the PCB's to the back of the panel, the total depth is just under 1.5" thick (~36mm). These specific monitors use a typical "PC power cord" that plugs directly into power supply PCB. The physical size of that plug makes it the thickest part of the whole design. Otherwise, it could have been somewhat thinner...

jgodfrey, 2021-04-26:
As mentioned above, the electronics of my frames are just under 1.5" thick. The frames themselves are just under 2.25" thick (~55mm). That allows for:

- Some dimensionality on the face of the frame (thickness angles back towards the screen face)
- A small air gap behind (above) the electronics
- A recessed, peg-board based back (to keep hands out of the live 110v power and provide some airflow).

sapnho, 2021-04-26:
Yes, a bit of distance to the wall is helpful for proper ventilation, and in my case, because I have a metal frame, I probably need it for my WiFi signal.

sapnho, 2021-04-26:
> I tried to remove a metal frame around the panel. Unfortunately, it was glued and when I pulled on it an air bubble appeared behind the panel surface. Na chance to remove it.

I was always most afraid to rip off a cable. I'll make three crosses when the metal frame has been mounted and it still works. Good thing that the frame builder is an electronics tinkerer, so he knows that these displays are somewhat fragile.

sapnho, 2021-05-21:
The frame is back and the display still works! If there is anything to test with 4K monitors, just let me know.
@paddywwoof The image looks much sharper than before. It seems like the resize issue was fully solved.
[image]

sapnho, 2021-05-21:
Ah, looks like I have been too quick in my verdict. According to the test image, scaling is still happening. The photo here doesn't show it too well. Everything is just grey and no checkerboard.
[image]

paddywwoof, 2021-05-21:
That is odd. The v2021.05.02 requires pi3d v2.45 which no longer does resizing *unless* the image width is not a multiple of 4 (which means 3840 and 4096 should be OK, and I assume you're not scaling it to fit inside your monitor if it's the wrong aspect ratio)

I will see if there are any debug messages we can add to see what's going on.

sapnho, 2021-05-22:
So just to make sure that I have the right settings for this test:

`  fit: False                              # default=False, True => scale image so all visible and leave 'gaps'
                                          #                False => crop image so no 'gaps'
  #auto_resize: True                       # default=True, set this to False if you want to use 4K resolution on Raspberry Pi 4.
`

"Auto resize" isn't active anymore, is it?

paddywwoof, 2021-05-22:
OK.. 

As far as I can see the only thing that would trigger resize inside picframe is if `blur_edges: True` or `mat_images` was triggered (I will assume neither apply here) However for applying the texture to the screen the fragment shader does a [look-up](https://github.com/helgeerbe/picframe/blob/5880c39d83ea47557e28309abb4a22e3f340018e/picframe/data/shaders/blend_new.fs#L7) on the sampler2d texture made from the image. And the float precision specified for GLES in [pi3d is mediump](https://github.com/tipam/pi3d/blob/d3963d7fc9171a322c7480b97e74c888e2a87818/pi3d/shaders/std_head_fs.inc#L3) which probably only has 10 bits of precision (that's minimum spec so it might be more, I will test it), and as the mapping is 0.0 to 1.0 that's only 1024 steps. highp should give 16 bits of precision which would be 65536 steps.

I don't know at all that this is the cause of the problem as I would have expected that lower precision to show on 2k screens but worth checking and easy to do. I have put a slightly modified shader into a new branch [glsl_highp](https://github.com/helgeerbe/picframe/commit/e03cc07c99309522ca082c0e26af8f4ca4eae21d) so you should be able to try that
```shell
$ cd picframe # or whatever you called it if you already cloned it to this  Pi
$ git clone https://github.com/helgeerbe/picframe.git # if you haven't already, otherwise
$ git pull
$ git checkout glsl_highp
$ sudo pip3 install -e . #NB line ends with a period!
```
PS the `mediump` was added years ago when someone was running pi3d on RPi "competitors" that used the mali GPU. I think all the broadcom GPUs can cope with highp so maybe that's something to switch back to anyway. Not sure if it has a performance impact

sapnho, 2021-05-22:
What is interesting that the sharpness of the screen is a lot better than before I updated to 05.05.

Anyway, I tried the above but I get error messages at `git pull`. It says "`fatal error: No git repository`" or so, I have the German version installed because of geolocation.

paddywwoof, 2021-05-22:
Yes I'm not sure why it's sharper (well presumably it's something to do with the resizing but then the question over not reproducing the test card properly are not clear).

Sounds like the directory isn't actually a repository... Did you git clone following my instructions above - they are actually wrong now I look at them. You need to cd to the cloned directory after cloning it! They should have been
```shell
## a) If you haven't already cloned picframe
$ cd ~
$ git clone https://github.com/helgeerbe/picframe.git
$ cd picframe
$ git checkout glsl_highp
$ sudo pip3 install -e . #NB line ends with a period!

## b) If you have already cloned picframe
$ cd ~/picframe
$ git pull
$ git checkout glsl_highp
$ sudo pip3 install -e .

## c) If you followed by previous incorrect instructions
$ mv ~/picframe ~/picframe_temp
$ mv ~/picframe_temp/picframe ~/
$ rm ~/picframe -d
$ cd ~/picframe
$ git pull
$ git checkout glsl_highp
$ sudo pip3 install -e .
```
Sorry about that!

sapnho, 2021-05-23:
Thanks, @paddywwoof but I am having difficulties starting this because the directory is a bit different from the normal installation process. I did a).
For example, there is no "run_start.py" file and the "start.py" is very different.

I tried this on a Pi4:
`sudo xinit /usr/bin/python3 /home/pi/picframe/picframe/start.py /home/pi/picframe/picframe/config/configuration.yaml`

which results in

```
INFO:start.py:starting ['/home/pi/picframe/picframe/start.py, /home/pi/picframe/picframe/config/configuration.yaml]
INFO:model.Model:Open config file: /home/pi/picframe/picframe/config/configuration.yaml:
Traceback (most recent call last):
  File "/home/pi/picframe/picframe/start.py", line 166, in <module>
    main() 
  File "/home/pi/picframe/picframe/start.py", line 130, in main
    m = model.Model(args.configfile)
  File "/home/pi/picframe/picframe/model.py", line 171, in __init__
    model_config['portrait_pairs'])
  File "/home/pi/picframe/picframe/image_cache.py", line 36, in __init__
    self.__db = self.__create_open_db(self.__db_file)
  File "/home/pi/picframe/picframe/image_cache.py", line 259, in __create_open_db
    db = sqlite3.connect(db_file, check_same_thread=False) # writing only done in loop thread, reading in this so should be safe
sqlite3.OperationalError: unable to open database file
xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.
```

paddywwoof, 2021-05-23:
Hi @sapnho I'm not sure that will work as expected. `picframe/picframe/start.py` will import from the installed picframe, so unless you did the `pip` stage to overwrite that version it do odd things.

I'm not really sure what your setup has different from the standard instructions but whatever you would normally do (or did do for your previous setup) at the point where you install using pip from pypi.org i.e. `sudo pip3 install picframe`, if you have cloned the repository and changed to the `glsh_highp` branch then you can just install from your local repository. If you previously altered the files that were installed then you would need to make the same alterations. If you have modified any of the files like start.py in `/usr/lib/python3/dist-packages/picframe` (or wherever `sudo pip3` puts them) then you will need to copy them and overwrite the new ones. But basically whatever you did in the past to upgrade picframe do exactly the same apart from pip install from local repo.

sapnho, 2021-05-23:
Hi @paddywwoof , thanks much for your help, as always much appreciated!

I think I'll work with the existing standard version for the time being. The image looks very sharp to me even if there is still some kind of scaling going on. It's much sharper than previously!

I just don't want to ruin the fresh setup that I did with Home Assistant integration and all my little extras that have accumulated over time. :-)

But I could possibly try everything with a test SD card.

paddywwoof, 2021-05-23:
I agree it's not worth breaking something that's taken effort to sort out. I can probably test the new shader on my RPi4 even though I haven't got a 4k monitor. If I define a Plane that 4k sized with a Texture made from a 4k sized image, constructed from alternate black and white pixels, then I can see what the top left corner looks like. And I will let you know what I find

Paddy