helgeerbe/picframe #107: Fade time on 4K

sapnho, 2021-04-20:
Now that I have a 4K screen full time, I happy to do some testing. Setup: Pi4 with a 4K 32 inches screen.

A fade time of 3 seconds in configuration.yaml will result in about 9 seconds.

paddywwoof, 2021-04-20:
So that sounds like you're getting just under 7fps assuming you have the default 20fps set in config. Setting fade time to 3s at 20fps means the change in alpha each frame is 1/60 or 0.01666 which is sounds to be enough to perceive quite easily. You might as well change the config fps to 7.0 seeing as that's all you're getting and change the fade time to 30s - see what the lowest value is you can get away with. At least with the fps set to the actual figure your fade times will match whatever you set. I think you can change fade_time using a Controller method (http or mqtt) so  you can play around while the pictureframe is running.

sapnho, 2021-04-21:
Yes, I guess 7 fps is about right. This is strange because the Pi4 is so much more powerful than its predecessor. And that explains the screen flicker which I noticed. So I guess, we can close this thread and have the main discussion in the "Flicker" post.

helgeerbe, 2022-01-05:
Will be handled in #106

closed 2022-01-05 by helgeerbe