helgeerbe/picframe #108: working wiht NAS

whippett2000, 2021-04-30:
I wondered if you can help me.  I want to access photos stored on a password protected share on a NAS drive on the same wifi network as the Raspberry Pi.  What lines of code do I need to insert into the script?  I am assuming these configurations will need inserting in the configuration.yaml file?  This is for the Pi3D PictureFrame image viewer installed on my Raspberry Pi 3

helgeerbe, 2021-04-30:
Hi @whippett2000 you have to mount your NAS share on your raspberry. Search on google `mounting a shared NAS folder to a Raspberry Pi` . In the configuration.yaml you can point the picture directory to your image folder on the mountpoint.

paddywwoof, 2021-04-30:
I have added some code that should make mapping drives to subdirectories within ~/Pictures (as per your (Helge's) instructions and Jeff's and my suggestions) https://github.com/helgeerbe/picframe/issues/76

I will merge those changes into the dev branch tonight or tomorrow. (Also get v2.45 of pi3d up to pypi)

Paddy

helgeerbe, 2021-05-02:
New release [2021.05.02](https://github.com/helgeerbe/picframe/wiki/Changelog) is published

helgeerbe, 2021-07-13:
No more action on this issue. Guess I can close it.

closed 2021-07-13 by helgeerbe