helgeerbe/picframe #110: Serious bug in the latest image_cache() code

jgodfrey, 2021-05-02:
Hi Paddy,

I just got around to updating a frame with the latest `dev`  branch (for the first time in a while). Running it, I'm only seeing a single image displayed. Looking around, that's because the `all_data` view only contains a single record. Digging deeper, that seems to be because each time a file is inserted, we make a pass to update the relevant folder (in `__insert_file`) and that pass is now incrementing the `folder_id` field - which it didn't used to do.

I was confused by that, but I think the culprit is the change from the original `INSERT OR IGNORE` clause in the folder update SQL to now be `INSERT OR REPLACE`. 

I haven't through through a proper fix yet and I need to step out for a while, but I'm curious if anyone else is seeing this?

And, unfortunately, I see that code was merged into master and released earlier today. Bad timing on my part I guess. 

Thoughts?

helgeerbe, 2021-05-02:
Bad timing. I run dev, but did't add images.  Should I roll back the release?

jgodfrey, 2021-05-02:
Unless @paddywwoof has some additional info to the contrary, it feels like `yes` to me...

helgeerbe, 2021-05-02:
I yanked the release on pypi

    The picframe release 2021.5.2 released on 2021-05-02 has been yanked.
    Yanked by: helge with a role: owner.
    Reason: https://github.com/helgeerbe/picframe/issues/110

So at the moment, installation from pypi will work. I will kepp main. Maybe @paddywwoof has a quick solution.

jgodfrey, 2021-05-02:
I assume the mentioned change is related to handling the new `missing` field on the `folder` table. The original `INSERT OR IGNORE` clause wouldn't have updated the record appropriately if the old value for `missing` was `1` and now needed to be `0`. The new `INSERT OR REPLACE` will properly handle that, but leads to the update of the PRIMARY KEY field, which is a deal breaker here...

I think we may have to make some more explicit checks on that record during insertion. The single clause is nice, but I don't think we can make it work properly in the face of the new `missing` field.

jgodfrey, 2021-05-02:
I think something like this is an appropriate fix...

```
INSERT INTO folder(name) SELECT 'test' WHERE NOT EXISTS(SELECT 1 FROM folder WHERE name = 'test');
UPDATE folder SET missing = 0 where name = 'test'
```

Essentially, that does the operation in 2 steps. First, it inserts the new folder into the table, but only if it's not already there. Second, it updates the `missing` field to be `0`.

I'll try to submit an update to `dev` later in the day unless @paddywwoof gets to it first. I'll be AFK for a while today though...

jgodfrey, 2021-05-02:
Just submitted an initial fix here, but it should get some additional testing before being released...

jgodfrey, 2021-05-02:
Ugh... In my haste, that got merged into `master`, not `dev`... :(

jgodfrey, 2021-05-02:
If someone can fix that for me, that'd be great. I really need to step away for a bit now... Sorry for the extra work.

helgeerbe, 2021-05-02:
Thanks Jeff for doing the analysis and the fix.  I'll move it to dev.

helgeerbe, 2021-05-02:
OK, I see. It's in dev and main. I will keep the fix in main, too.

paddywwoof, 2021-05-02:
Jeff, Helge, thanks for catching this. I've just tried the revised code and made sure I deleted the db before starting. I thought I'd done that before but obviously not.

I had naively thought that the `INSERT OR REPLACE` was equivalent to `INSERT ON DUPLICATE KEY UPDATE`, now I google I see there is an `UPSERT` or `INSERT .. ON CONFLICT(name) DO UPDATE` However having the explicit two executes: INSERT then UPDATE is clearer

helgeerbe, 2021-05-02:
Guess we need some unit-tests, when changing the code.

paddywwoof, 2021-05-02:
Yes! I did make one that clears the db and adds to it etc then compares the screen with a previous screen so it tests lots of things... But not adding multiple files to the db.

Devising tests for visual things like 3D games is quite tricky.

jgodfrey, 2021-05-02:
> OK, I see. It's in dev and main. I will keep the fix in main, too.

Are you sure it's in `dev`? I don't see it there...

paddywwoof, 2021-05-03:
No, I don't think it's in dev yet either.

Could the `folder_update` revert to `INSERT OR IGNORE INTO folder(name) VALUES(?)` as it was before I hacked it?

helgeerbe, 2021-05-03:
Indeed, it's not dev. Don't know why I thought it's in dev. I will du a cherry pick from main.

helgeerbe, 2021-05-03:
Done. Ich cherry picked the commit to dev. And reverted the last commit on main.
Hope, everything is tidy now.

jgodfrey, 2021-05-03:
> Could the `folder_update` revert to `INSERT OR IGNORE INTO folder(name) VALUES(?)` as it was before I hacked it?

Yeah, I had that same thought later too. With the addition of the secondary `UPDATE`, that should work fine I think...

helgeerbe, 2021-05-03:
Tell me, if you're done. I will update main and republish this version.

paddywwoof, 2021-05-03:
I've just tried running it on Pi3 with
```python
       # Insert the new folder if it's not already in the table. Update the missing field separately.
        folder_insert = "INSERT OR IGNORE INTO folder(name) VALUES(?)"
        folder_update = "UPDATE folder SET missing = 0 where name = ?"
...
        # Insert this file's info into the folder, file, and meta tables
        self.__db.execute(folder_insert, (dir,))
        self.__db.execute(folder_update, (dir,))
...
```
and it seems to be fine. I can start with empty db and copy a directory in and remove one and it seems OK.

There is something that isn't quite right in that when I delete a subdirectory from the main base directory (default ~/Pictures) and an image is showing from that subdirectory, it seems to be stuck (for a long time at least) just showing that same image. If I set a date that will cause the 'no images' picture then swap back it sorts itself out, eventually, most of the time. And it's fine if you stop the slide show and let it restart. I think there was something of this issue before - I don't think the new changes are impacting it but I'd like to get to the bottom of it. Does anyone else see anything like this?

jgodfrey, 2021-05-03:
@paddywwoof,

First, I think the above code should be fine. Regarding the mentioned issue, I did not notice that in my testing yesterday, but I wasn't explicitly looking for it. I'll keep an eye out for it.

I did run into a testing issue yesterday that you might have a suggestion on though... I've always used _SQLite Expert Personal_ to interact with SQLite databases. For this project, I run that app on my Windows PC and open the frame's database file on the RPi over the network. However, it seems that I don't always get a current snapshot of the db. Changes made on the PI are not always visible on the PC client. Sometimes, closing and reopening the db on the PC will fix it, but not always. So, it makes db-based debugging somewhat suspect as I never know if I'm looking at the latest version. It's almost as if there's some caching going on via the Samba connection.

I remember having a similar issue  earlier this winter and I managed to tweak some things in the Samba config that seemed to help, but I apparently didn't document them. 

Does any of this ring a bell and do you have any advice?

paddywwoof, 2021-05-03:
Sorry, I've not encountered that. But that's most likely because until quite recently the wifi connection to the Pi3 was so bad that it was impossible to notice anything beyond the connection dropping all the time. A few weeks ago I found that if I plug a USB wifi dongle into the Pi (in addition to its built in wifi) the signal strength is significantly improved and I can communicate with it reliably.

I will try watching the db over the network while picframe is running and I add and remove images and see if I can spot a delay or mismatch.

jgodfrey, 2021-05-03:
Thanks paddy. I'm not overly familiar with previous RPi hardware, but I'm testing on an RPi4 - so I think the wifi issues are less relevant, right?

paddywwoof, 2021-05-03:
Yes, I think that's right. The RPi3 I have is from the first batch and I think the aerial was particularly bad.

I've just connected to the db from my laptop while I deleted a folder of images. This time picframe seemed to respond quite quickly - only showing the pictures that remained. However the folder table in the db didn't respond with the changed missing flag. When I closed the db and re-opened it the changes were as expected (file, meta, location all unaltered - just a flag change in folder) so it looks like I have a similar caching issue, but that maybe it's fixed by a close/open.

jgodfrey, 2021-05-03:
@paddywwoof - After some digging, I may have hit upon how I fixed this earlier this winter...

Adding the following to `/etc/samba/smbd.conf` seems to have fixed my SAMBA cache latency issue:

```
kernel oplocks = yes
oplocks = yes
```

With that, a simple "refresh" of an already open, but modified, table within my management tool shows the correct db content.

sapnho, 2021-05-04:
@jgodfrey Interesting! Would it make sense that I recommend adding these lines in general in the Samba config for smooth use with Pi3D? https://www.thedigitalpictureframe.com/installing-samba-on-your-raspberry-pi-buster-and-optimizing-it-for-macos-computers/

paddywwoof, 2021-05-04:
@sapnho it might be a bit specific, i.e. where there is a file on the RPi that might have changing content that you want to view over the network. I'm not sure where that would come in useful apart from debugging - really the database should be *only* accessed by picframe.ImageCache so via the interface_mqtt and interface_http functionality. There sound to be some disadvantages to oplocks in certain circumstances so probably best to apply them when there is a reason (they sound to be more related to the windows OS so that might be relevant)

jgodfrey, 2021-05-04:
So, it looks like `dev` should be fine now, but `main` still has the original, broken code. AFAIK, we're done here, so maybe the latest `dev` changes to `image_cache` need to be cherry-picked into `main`?

helgeerbe, 2021-05-04:
If you're fine with dev. I will merge it into main and prepare a release. Do I have your go?

jgodfrey, 2021-05-04:
I haven't had a chance to test Paddy's last minor change to `dev` (that simplifies the SQL I added to fix this issue), but it looks perfectly good on paper. I can do that this evening if you want to wait for it. I'll leave that up to you.

helgeerbe, 2021-05-04:
Well, I would wait for your go. Please test it, if you have time. I think we can wait a couple of hours.

jgodfrey, 2021-05-04:
Helge - I just did some additional testing on the `dev` branch and it seems to work as expected. I think you can merge it into `main` and work on a release. 

Thanks.

helgeerbe, 2021-05-05:
#113 is done.

I released version [2021.05.05](https://github.com/helgeerbe/picframe/wiki/Changelog)

closed 2021-05-05 by helgeerbe