helgeerbe/picframe #112: Image does not change (rpi2)

ilons, 2021-05-03:
First of all, I want to sincerely thank all of the ones contributing to this project, since I think it fills an important spot!

When trying to set this up on an old RPI 2, I'm struggling with having the image to actually change. 
If I leave it at default configuration, after which I add 2 different images and then launch it, only the first one will ever be displayed.

I tried to lower the `time_delay`, at which point I can see that it tries to change the image (as the title are shown every `time_delay`seconds), however, the actual image (nor the title) changes.

Running with LOG_LEVEL = DEBUG does not really print anything which gives me a hint of what would be wrong.

Adding in a new image in the directory while picframe is running DOES force the image to change to the newly added one, which will then be the one being "stuck".

I'd really love some input on how to solve this!

ilons, 2021-05-03:
This is my configuration file used:
```viewer:
  blur_amount: 12                         # default=12,  larger values than 12 will increase processing load quite a bit
  blur_zoom: 1.0                          # default=1.0, must be >= 1.0 which expands the backgorund to just fill the space around the image
  blur_edges: False                       # default=False, use blurred version of image to fill edges - will override FIT = False
  edge_alpha: 0.5                         # default=0.5, background colour at edge. 1.0 would show reflection of image
  fps: 20.0                               # default=20.0
  background: [0.2, 0.2, 0.3, 1.0]        # default=[0.2, 0.2, 0.3, 1.0], RGBA to fill edges when fitting
  blend_type: "blend"                     # default="blend", choices={"blend", "burn", "bump"}, type of blend the shader can do
  font_file: "/home/pi/picframe_data/data/fonts/NotoSans-Regular.ttf"
  shader: "/home/pi/picframe_data/data/shaders/blend_new"
  show_text_fm: "%b %d, %Y"               # default "%b %d, %Y", format to show date over the image
  show_text_tm: 3.0                      # default=20.0, time to show text over image with file name
  show_text_sz: 30                        # default=40, text character size
  #show_text: "title caption name date folder location"  # default="title caption name date folder location", show text, include combination of words: title, caption name, date, location, folder
  show_text: "date location folder"  # default="title caption name date folder location", show text, include combination of words: title, caption name, date, location, folder
  text_justify: "L"                       # text justification L, C or R
  fit: False                              # default=False, True => scale image so all visible and leave 'gaps'
                                          #                False => crop image so no 'gaps'
  #auto_resize: True                       # default=True, set this to False if you want to use 4K resolution on Raspberry Pi 4.
                                          # You should ensure your images are the correct size for the display")
  kenburns: False                         # default=False, will set fit->False and blur_edges->False
  display_x: 0                            # offset from left of screen (can be negative)
  display_y: 0                            # offset from top of screen (can be negative)
  display_w: null                         # width of display surface (null->None will use max returned by hardware)
  display_h: null                         # height of display surface
  use_glx: True                          # default=False. Set to True on linux with xserver running

  mat_images: 0.01                        # default=0.01, True, automatically mat all images. False, don't automatically mat any images. Real value, auto-mat all images with aspect ratio difference > than value
  mat_type: null                          # default=null, A string containing the mat types to choose from when matting images. It can consist of any or
                                          # all of 'float float_polaroid float_color_wrap single_bevel double_bevel double_flat' (null or '' will use all mat types)
  outer_mat_color: null                   # default=null, Color of the outer mat as an RGB list. null will auto-select a reasonable color based on the image.
  inner_mat_color: null                   # default=null, Color of the inner mat as an RGB list. null will auto-select a reasonable color based on the image.
  outer_mat_border: 75                    # default=75, Minimum outer mat border in pixels
  inner_mat_border: 40                    # default=40, Minimum inner mat border in pixels (for styles that use it)
  outer_mat_use_texture: True             # default=True, True uses a texture for the outer mat. False creates a solid-color outer mat.
  inner_mat_use_texture: False            # default=False, True uses a texture for the inner mat. False creates a solid-color inner mat.
  mat_resource_folder: "/home/pi/picframe_data/data/mat" # Folder containing mat image files

  #codepoints: "1234567890AÄÀÆÅÃBCÇDÈÉÊEËFGHIÏÍJKLMNÑOÓÖÔŌØPQRSTUÚÙÜVWXYZaáàãæåäbcçdeéèêëfghiíïjklmnñoóôōøöpqrsßtuúüvwxyz., _-+*()&/`´'•" # limit to 121 ie 11x11 grid_size

model:
  pic_dir: "/home/pi/Pictures"                   # default="/home/pi/Pictures", root folder for images
  deleted_pictures: "/home/pi/DeletedPictures"   # move deleted pictures here
  no_files_img: "/home/pi/picframe_data/data/no_pictures.jpg" # default="PictureFrame2020img.jpg", image to show if none selected
  subdirectory: ""                        # default="", subdir of pic_dir - can be changed by MQTT"
  recent_n: 7                             # default=7 (days), when shuffling file change date more recent than this number of days play before the rest
  reshuffle_num: 1                        # default=1, times through before reshuffling
  time_delay: 10.0                       # default=200.0, time between consecutive slide starts - can be changed by MQTT
  fade_time: 5.0                         # default=10.0, change time during which slides overlap - can be changed by MQTT"
  shuffle: True                           # default=True, shuffle on reloading image files - can be changed by MQTT"
  sort_cols: 'fname ASC'                  # default='fname ASC' can be any columns in the table with optional ASC or DESC separated by commas
[... cut ...]
  use_http: False                         # default=False. Set True to enable http NB THIS SERVER IS FOR LOCAL NETWORK AND SHOULD NOT BE EXPOSED TO EXTERNAL ACCESS
  path: "/home/pi/picframe_data/html"            # path to where html files are located
  port: 9000                              # port used to serve pages by http server < 1024 requires root which is *bad* idea
  use_ssl: False
  keyfile: "path/to/key.pem"              # private-key
  certfile: "path/to/cert.pem"            # server certificate```

paddywwoof, 2021-05-03:
Hi, did you by any chance install picframe over the last day or two? In which case you will have a version with known bug. My fault, sorry. In a terminal do
`pip3 install picframe --upgrade`
And see if that does anything

ilons, 2021-05-03:
So I installed it from source instead, but still have the same issues. Version as below:
```picframe version:  0+untagged.315.g882082a

Checking required packages......
PIL :  5.4.1
exifread :  2.3.2
pi3d :  2.45
yaml :  5.4.1
paho.mqtt :  1.5.1
iptcinfo3 :  2.1.4
numpy :  1.16.2
ninepatch : installed, but no version info

Checking optional packages......
pyheif : Not found!```

jgodfrey, 2021-05-03:
> Adding in a new image in the directory while picframe is running DOES force the image to change to the newly added one, which will then be the one being "stuck".

Yeah, this exactly describes a problem that was fixed over the weekend (that paddy mentions above). Depending on the details and timing of your source installation, I assume you still have the broken build. If you installed from the `dev` branch prior to about 4 hours ago, it was still broken. Now, it should be fixed in both `dev` and `main`. Sorry for the confusion.

paddywwoof, 2021-05-03:
@ilons, Hopefully you've managed to sort this out. pip3 install will get the last version as Helge reverted the installation package on pypi.org However, if you do
```shell
picframe --version

picframe version:  2021.04.02

Checking required packages......
...
```
note version from 2nd April, and still get the problem you reported before then there is some other issue that we need to sort out.

Paddy

ilons, 2021-05-03:
I'm just happy about all the swift help you are providing, much appreciated!

Explicitly installing `pip3 install picframe==2021.04.02` did solve it, now it seems to be working as intended!

closed 2021-05-04 by ilons