helgeerbe/picframe #115: Unable to toggle directory name text via MQTT

jgodfrey, 2021-05-07:
Using the latest `dev` branch, I'm unable to toggle the directory name text overlay via MQTT. Poking around, that's because we're using the wrong key in the call to `__controller.set_show_text`. We're using `directory` instead of the expected `folder` key. I have a local fix that I'll submit next.

jgodfrey, 2021-05-07:
This should be fixed in `dev` via https://github.com/helgeerbe/picframe/commit/59fc665df9b52485b71074e1d85b5a9d6414c8a5

jgodfrey, 2021-05-08:
I know there's been some confusion in the code ever since I added that new text type. When I originally added it, I used "folder" as the key, which is still the key used in the text mapping function here:

https://github.com/helgeerbe/picframe/blob/19eb16225ec1316a439619eb9402835c44d45fbc/picframe/viewer_display.py#L19

At some point later, there were some changes made that started using `directory` - which caused some breakages. 

I just noticed last night that the key `directory` was being passed to the viewer's `set_show_text` method to activate the folder's text overlay string from the MQTT interface. However, that key doesn't match anything, as the key is `folder` (as linked above) so it didn't work.

I don't really have a preference between `folder` and `directory`, though I tend to use ` folder` as it's shorter. However, you're right that we now have some discrepancies in the code regarding which word we're using in various places.

That said, I don't think this change breaks anything - it should only fix the activation of the text overlay from MQTT. Though, there may still be some places that need a similar fix - I haven't looked into it any deeper.

paddywwoof, 2021-05-08:
Yes, this does need to be tidied up. Originally in Pictureframe2020 there was a "subdirectory" filter, plus the slightly different use between the `txt_map` key which is just a convenience for controlling the text overlay, as you say, and the SQL filter mechanism needing a variable with name `folder` or `directory` or `subdirectory` being set to some string value. I will look through the code later and see if there is anywhere else that's muddled.

Paddy

closed 2021-07-05 by jgodfrey