helgeerbe/picframe #119: Strange "flash" on the frame followed by picture repeat

jgodfrey, 2021-05-14:
Using `dev`, I've noticed a few split-second "flashes" right at the end of an image display cycle, followed by the same image being shown a second time - after which things advance as intended.

Doing some digging, I _think_ that's being caused when the slideshow's internal playlist references images that are missing on disk. That causes the call to `__tex_load()` to fail and trigger the catch-all exception. That, in turn, causes `slideshow_is_running()` to return early. 

I think, rather than simply allowing `__tex_load()` to attempt to load a missing image and trigger an exception in the process, we should vet the availability of the images we're about to display earlier in the process - at a point where we can handle the situation more gracefully.

On the surface, I'd think that should play out something like this:

- If `image2` is defined but missing (the paired portrait), it should be _blanked out_ in the playlist, as this is already a normal case when a paired portrait doesn't exist or wasn't requested.
- If `image1` doesn't exist but `image2` does, `image2` should be shifted to the `image1` slot and the `image2` should be blanked out. This would result in the same situation as the previous item. This would be a worst case where a portrait image that should have been part of a pair is displayed singly.
- If `image1` doesn't exist and `image2` either doesn't exist or isn't defined, that record in the playlist should just be skipped as there's nothing we can display there...

Without a deeper dive, I'm not sure where the most appropriate place for something like this is, but perhaps it belongs in `model.get_next_file()`, which is responsible for returning the next 1 or 2 (for portrait pairing) image references to the slideshow...

I may look a little closer into this over the weekend, but maybe @paddywwoof has some additional thoughts?

paddywwoof, 2021-05-14:
@jgodfrey well done for discovering this. I suppose the issue is with files disappearing after the list has been returned by ImageCache. The advantage of doing the check in Model is that it could move on to the next image as it holds the full list. The disadvantage is that there is an additional chunk of file reading and processing code in that file so it's a bit hard to keep track. At the moment it looks like a missing file should make ```__tex_load()``` return None which makes ```slideshow_is_running``` return early, maybe we should change that so it reverts to same foreground and background images it currently holds. Something like (not tested)
```python
...
        if pics is not None:
            #self.__sbg = self.__sfg # if the first tex_load fails then __sfg might be Null TODO should fn return if None?
            self.__next_tm = tm + time_delay
            self.__name_tm = tm + fade_time + float(self.__show_text_tm) # text starts after slide transition
            new_sfg = self.__tex_load(pics, (self.__display.width, self.__display.height))
            if new_sfg is not None: # this is a possible return value which needs to be caught
                self.__sbg = self.__sfg
                self.__sfg = new_sfg
            else:
                #return (True, False) # return early
                (self.__sbg, self.__sfg) = (self.__sfg, self.__sbg) # swap existing images over
...
```
Of course this would just make the previous image re-run so wouldn't be as nice as swapping a new one if one or both are missing.

jgodfrey, 2021-05-14:
> I suppose the issue is with files disappearing after the list has been returned by ImageCache.

I'm not sure that's true, is it? It looks like we don't ever verify that an image referenced in the db exists once it's returned in the playlist. So, any images in the db that are removed from disk could cause this problem at any time I think. That's because we no longer reconcile the db file records with the disk unless we set the `__purge_files` member in `image_cache.py`. 

So, from the point where files are deleted (or a folder is deleted) until things get resynced with a forced _purge_, I think the missing files will cause glitches in the slideshow (unless I've missed something).

> The disadvantage is that there is an additional chunk of file reading and processing code in that file so it's a bit hard to keep track.

Sorry, I didn't follow this. It seems like we could just adjustment the file references as necessary in `get_next_file` to ensure the ones returned are valid. But, maybe I'm over-simplifying it. I haven't looked closely, and it is fairly complex...

jgodfrey, 2021-05-14:
Though, short of anything better, your above-mentioned solution is definitely 1) simple and 2) much better than what we have today.

paddywwoof, 2021-05-14:
Ah yes, the whole idea is that files stay in the db. If a directory is moved though the files don't get include in the view, I think. The ```__purge_files``` system needs writing up, I think I added a method to the http and mqtt classes.

My comment on the disadvantage was really just thinking aloud: It seemed slight untidy to read the disk for files (as a check) in Model then have ViewerDisplay read them for real. Though checking files exist is pretty light weight and is only going to happen infrequently so probably not worth worrying about.

jgodfrey, 2021-05-15:
> Ah yes, the whole idea is that files stay in the db. If a directory is moved though the files don't get include in the view, I think.

Yeah, that's correct. However, in my case, the _files_ themselves were removed, but they were in a folder that still existed. So, the records for those files *are* include din the `all_data` view of the cache. If the _folder_ was gone, I think the system would generally handle itself, but it's just individual files in this case. 

And, maybe there's a better place to make a `file.exists` check than the `model`, but it seems that it'd be nice to verify the files we're about to display actually exist at a point where we could do something graceful if they didn't...

sapnho, 2021-07-05:
Has this issue been dealt with? I also noticed it a few times now, i.e. my wife did.. :-)

jgodfrey, 2021-07-05:
@sapnho - I don't think we've fixed it (at least, I haven't). Since it's caused by deleted images, for now, I've been getting around it with the following procedure:

If I delete any images (for me, that's manually, outside of the picture frame)...
- I run the "Purge Files" from my MQTT remove
- I toggle "Shuffle" OFF and then back ON via my MQTT remove

The purge removes the missing image references from the DB
Toggling shuffle rebuilds the playlist, now *without* the missing image references. I toggle it twice to leave it in SHUFFLE mode, which is how my frame runs...

We do need to find and fix this one, but I've found it easy to manage using the above (really, just being lazy I guess...). :)

sapnho, 2021-07-05:
Thanks, @jgodfrey , yes, fixing that would be great. I also got the comment from two readers in the past. We should not ask users to run maintenance commands on their picture frames. I sense strong spousal opposition.. :-)

jgodfrey, 2021-07-05:
> I also got the comment from two readers in the past. We should not ask users to run maintenance commands on their picture frames. I sense strong spousal opposition.. :-)

@sapnho - Sure, I agree 100%. Really, just pointing out one potential mitigation method until we get it fixed...

jgodfrey, 2021-07-08:
FYI - I've been experimenting with a fix for this. Basically, the changes I've made handle missing images mostly as I described in the opening post above. It seems to be working, but I need to set up some specific "missing image" scenarios for better testing. If it pans out, I'll check in a branch for others to test sometime soon.

paddywwoof, 2021-07-09:
@jgodfrey great, I will test it too once it's pushed to github.

jgodfrey, 2021-07-09:
Doing some additional testing here over lunch... Generally, everything works as expected and the slideshow looks really solid now - even in the face of randomly missing images. I do still have one (expected) issue to contend with though... That's the case where there are `0` images on disk to display, but there are still records in the database (pointing to nonexistent images). Currently, that'll get stuck in a flat spin trying to find an image that can be displayed - forever... :(

Due to the current structure of `get_next_file()`, it'll be a bit sticky to deal with. I have an idea, but haven't implemented it yet in order to test. So, stay tuned I guess....

Unrelated, but I've also found an issue in the same area that's causing (at least one case) where the entire playlist doesn't get displayed an equal number of times - as discussed in [this issue](https://github.com/helgeerbe/picframe/issues/128). I'll try to take a closer look at that one once I'm finished up with this one.

jgodfrey, 2021-07-09:
Alright, I think I have this fixed, although the code in `get_next_file()` was (and still is) a bit messy. It could probably stand some refactoring at some point. As I mentioned, above, there's another issue lurking in that method too...

Anyway, see https://github.com/helgeerbe/picframe/issues/119.

jgodfrey, 2021-07-10:
FYI - I've closed the above-mentioned PR in favor of a new one (not yet pushed) that should fix both this issue and issue https://github.com/helgeerbe/picframe/issues/139

jgodfrey, 2021-07-10:
Created PR https://github.com/helgeerbe/picframe/pull/140 for issue https://github.com/helgeerbe/picframe/issues/139 and https://github.com/helgeerbe/picframe/issues/119.

I'll wait to merge in case anyone has input.

jgodfrey, 2021-07-13:
I'll likely merge this into dev sometime this evening unless anyone plans to test it independently. I'm happy to wait if that's the case - just let me know.

closed 2021-07-14 by jgodfrey