helgeerbe/picframe #120: Full insert of files into DB failing

TraffordSt, 2021-05-16:
Hi.  I'm having trouble getting a full library of images into the file & folder tables.  I find that the inserting stops on a seemingly random image (i.e.  image will load one time but not another, in separate DB's).  Background is:
I'm using a Pi Model 2.
I've moved approx 32k Images from the SD drive onto a USB drive, changed pic_dir to point to USB and I've renamed the old DB to get a fresh one. Upon startup of PicFrame it's created a new db and started populating the table.  It stops inserting on image 12,036, but by looking at the old DB I can see that this image was successfully inserted into the File table (when the images resided on the SD).
I've tried deleting the image it stopped on and restarting but that doesn't trigger a continuation of the load, so I guess the image wasn't the cause of the probem.  Upon investigation I have found that the old DB was never fully populated for all 32k images and that some folders are missing (approx 1,000 images weren't included, the app was running on this pi continuously for a couple of weeks).
Do you have any ideas what might be causing the DB load to stop?  Or is there a way to see what might be causing the failure e.g. logs? At the moment it seems to have stopped loading.

Thanks

paddywwoof, 2021-05-16:
@TraffordSt thanks for the feedback on this, it definitely sounds like a bug. I will try to think of a way to get more specific info on the cause but it might be a time issue. i.e. there is a flag set and unset in ImageCache that stops the db activity while the image transition is taking place (to keep it smooth). However looking at the code https://github.com/helgeerbe/picframe/blob/5880c39d83ea47557e28309abb4a22e3f340018e/picframe/image_cache.py#L86 it should only add files to the modified_files list only once that list has been emptied and there isn't a mechanism to stop files being added to the list for each folder identified as containing missing files https://github.com/helgeerbe/picframe/blob/5880c39d83ea47557e28309abb4a22e3f340018e/picframe/image_cache.py#L316 - even if that list-building process would take a long time and eat into the slide transition time.

If you set log_level to DEBUG then you should end up with a line reported for each file added to the db. Which is a lot so maybe better to log to a file.. but you would have to hack the picframe code to do that, poss easiest https://github.com/helgeerbe/picframe/blob/5880c39d83ea47557e28309abb4a22e3f340018e/picframe/start.py#L84 changing the line
```python
   logging.basicConfig(stream=sys.stdout, level=logging.INFO, filename="/home/pi/picframe.log")
```
Finally, you could try decreasing and increasing the time for each slide and see if that has any effect on the number of files entered into the db

TraffordSt, 2021-05-16:
Hi Paddy
I should say that I'm no Python programmer.  I replaced the the logging line as you suggested but it stopped the service from starting
[image]
As you can see from the image I tried a variation I found online, to no avail.  I'd be quite keen to know what I've done wrong here and get it working.

I tried changing the timings.  I normally have time_delay set to 60.  I changed it to 30 with no change and 90, again no change.  So 'select count(*) from file' is still as 12,036.

Thanks
Richard

jgodfrey, 2021-05-16:
Hi @TraffordSt,

The below logging line works for me:

`logging.basicConfig(level=logging.INFO, filename="/home/pi/picframe.log")`

Regarding your reported issue... It feels to me like the db loader thread might be dying for some reason (though, certainly just guessing). A likely candidate in my view would be some bad EXIF data that our EXIF reader code isn't handling very well. I fixed a similar issue a while back where an improperly formatted EXIF date in an image was causing the thread to die and other images to not load. 

If you can get the logging working, hopefully that'll point us to something more definitive.

Related, you can set the log level in the frame's configuration file. However, you cannot set it to write to a file. We should probably make that possible, as it'd prevent the need to modify the code for situations like this...

jgodfrey, 2021-05-16:
@TraffordSt,

As mentioned above, I'm not overly familiar with Python's ` logging` module. But, until someone has a better / cleaner suggestion, I think something like this might be what you're after for logging. Essentially, it adds both a file handler and a stream handler to the logging mechanism...

```
    file_handler = logging.FileHandler(filename='/home/pi/picframe.log')
    stdout_handler = logging.StreamHandler(sys.stdout)
    handlers = [file_handler, stdout_handler]
    logging.basicConfig(level=logging.DEBUG, handlers=handlers)
```

Additionally, I think you'll need to set the `log_level` option in the frame's config file (`configuration.yaml`) to `DEBUG` too, as the value from the log file will overwrite the above setting later in the code...

TraffordSt, 2021-05-16:
Hi @jgodfrey, thanks, the initial logging line did the trick and wrote the first 4 lines in the attached log.
The 4 lines of code in your 2nd post wrote the remaining information (I also update config as suggested).  There doen't appear to be anything related to updating teh DB though.
[picframe.log](https://github.com/helgeerbe/picframe/files/6489854/picframe.log)
I'll try some things with the images next.
Cheers

TraffordSt, 2021-05-16:
Re image library, I've tried 3 things:
1. changed pic_dir back to the SD drive.  The DB had no SD drive images in it as I created a new DB when I pointed to USB.  After a while the images started to scroll but no inserts into file table
2. stopped the service and swapped the old DB to be aactive, which contains data entries for images on the SD drive. No inserts into file table but then the SD drive has been (almost) fully indexed into the old (now active) DB so no supprises there
3. changed pic_dir to point to USB drive, no inserts into file table but then could be stuck on same image problem so not supprising

I would have expected inserts from change number 1 if it was working correctly.

So I'm thinking I have 3 options at this stage:
1. update app, which I haven't done for a few weeks
2. start again with a new DB
3. reinstall the app and start from scratch with default pic_dir.  get it all working again and have another go at moving to USB drive
Updated log file attached FYI
[picframe.log](https://github.com/helgeerbe/picframe/files/6489947/picframe.log)
 
Thanks

jgodfrey, 2021-05-16:
@TraffordSt,

Lines like the following (in your log) are related to the image caching system (so, the db):

```
DEBUG:image_cache.ImageCache:Creating an instance of ImageCache
DEBUG:image_cache.ImageCache:Updating cache
DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
DEBUG:image_cache.ImageCache:Found 0 new files on disk
```

I see a number of places in the latest log file you attached where we get a:

```DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk```

...but then no record of the number of images found (like the last line in the above block of messages). I'm not sure what that means, unless you were stopping and restarting the service prior that check returning. The log file certainly indicates the service was restarted a number of times. I assume you did that manually. If not, maybe it's dying while checking for disk changes and restarting itself?

TraffordSt, 2021-05-16:
I did a restart of the service each time I made a change to the code or config - do i not need to do this?  I assumed the in-memory version would need refreshing. So I would say the stop/starts are all me as there was no unusual behaviour when it was running.

jgodfrey, 2021-05-17:
> I did a restart of the service each time I made a change to the code or config - do i not need to do this?

No, you're correct. You would need to restart it after making code or config changes.

TraffordSt, 2021-05-17:
Hi.  Update on this..  I've come back to it a few hours later and I've found that images are being inserted while pointing to the old DB (SD images) and the USB drive (updating for images on USB and SD).  Possibly an issue with the new DB?  I've set it back to the new DB (USB images only) while still pointing to the USB drive.  Crossing fingers..

TraffordSt, 2021-05-17:
So it seems to be working ok when creating a new DB, or using the old DB, which suggests a problem with the DB.  When setting up the USB (with the new DB) I will have reset the service a few times before I had it working.  Do you think it possible I could have interrupted processing at a critical point leaving data in a broken state causing the problem?  Taking this line further, is there's an alternative 'safe' way to shut the app down or is 'sudo systemctl stop picframe.service' etc the best way?
Thanks

jgodfrey, 2021-05-17:
@TraffordSt,

While it's certainly possible that an interruption of the db work at a critical stage could have caused some corruption, I'd say it's unlikely. SQLite is typically very resilient. That said, I'm happy to take a look at the offending DB file if you'd like - just let me know.

Regarding how to shut the service down... I typically just stop the service (as you mention above). However, both the MQTT and HTTP interfaces offer a `STOP` command that will properly shut down various systems, including the db cache. So, that's probably the *best* way.

TraffordSt, 2021-05-18:
Hi.  Thanks for the info.  I think you were on the right track when you mentioned failing on a problem image, since the DB populated to the same number of images and failed again.  Although, because I earlier deleted the potentially problem image it's failing on the next one.  
Looking at the log file, I can see the loading of the last successful image, it then loads seven more before stopping, but these one's aren't in the DB - I've double checked this.  Additionally there are no errors in the log indicating what it failed on.  It just appears to have stopped.  I could remove the entire folder for now just to get them loaded but I'd like to understand what has happened for future and I wondered if you would mind looking at the log to see if you can see an issue?
Line 18511 has the insert of the last successful DB entry and there are 7 apparently successful insert after this, then it stops inserting.  I've uploaded an screenshot of the last image in file table and also one showing that the following 7 images didn't make it into the DB.
Thanks

[picframe.log](https://github.com/helgeerbe/picframe/files/6497550/picframe.log)
[image]
[image]

jgodfrey, 2021-05-18:
Hi,

When images are cached into the db, they're loaded in "chunks". That's because we only load the DB during the time where there's a static image on the screen (so, not during a transition between images). That's because the transition effect is relatively compute-expensive and timing specific and we didn't want the background caching process to interfere with it. 

When a "chunk" of images is cached, it's committed to the db at the end of the chunk load. In the event that something goes wrong, any images that were loaded but not committed would be lost. 

Looking at your log file, it looks like the file that was inserted at line `18511` was the last file in a "chunk". It seems that the remaining images in the log were part of the next chunk, and were (presumably) not committed - due to whatever went wrong. So, with that in mind, it makes sense that the log shows that they were inserted, but they're not actually in the db file.

So, that's sensible I think - but it doesn't really help us to isolate the issue. So far, I haven't found anything in the log file that's indicative of the problem.

TraffordSt, 2021-05-18:
That all makes perfect sense.  Thanks for looking at the log and confirming there's no error to be seen.  I'll see if removing that folder wil continue the load and I'll look to narrow it down a bit later by maybe gradually adding them back or something along those lines.
Thanks again

TraffordSt, 2021-05-18:
Hi.  Do you have any tips on how to trap this error?  I put a try-except around the logic in update_cache() but still no errors appearing, despite it failing to commit the last batch of images.
Thanks

jgodfrey, 2021-05-18:
Hmmm... Your above attempt seems like a good place to start, though it apparently didn't help...

Assuming you're at the point where additional images are failing to load... I'd be curious what's in the log if you:

- Stop the service
- Remove the current log file (to get a new one)
- Start the service
- Let it run for a bit (5-10 minutes)

If you can, give that a try and post the new log file.

TraffordSt, 2021-05-18:
Here, attached..
[picframe.log](https://github.com/helgeerbe/picframe/files/6504298/picframe.log)

TraffordSt, 2021-05-18:
Worth noting it can take up to a couple of hours before the DB inserts begin.
Cheers

jgodfrey, 2021-05-18:
> Worth noting it can take up to a couple of hours before the DB inserts begin.

Hmmm... That doesn't seem right either. Looking at your latest log, I see where it started to check for modified folders here:

```2021-05-19 08:24:19 DEBUG    image_cache.update_cache:74 No unprocessed files in memory, checking disk```

The next thing it should have logged is the number of new files if found in the modified folders using the below, but it hasn't gotten there (yet) in your log. 

```self.__logger.debug('Found %d new files on disk', len(self.__modified_files))```

So, it would seem that either the folder or file look up is taking much longer than expected... I guess we could see which by adding the additional log line noted below in the code where we're reading folders/files. I assume the new log line will get recorded fairly quickly. If that's the case, then the call to `__get_modified_files()` is the slow one (so far).

```
        if not self.__modified_files:
            self.__logger.debug('No unprocessed files in memory, checking disk')
            self.__modified_folders = self.__get_modified_folders()
            self.__logger.debug('Found %d modified folders on disk', len(self.__modified_folders)) # <--- add this
            self.__modified_files = self.__get_modified_files(self.__modified_folders)
            self.__logger.debug('Found %d new files on disk', len(self.__modified_files))
```

Based on what that shows, I'd be inclined to add some more logging within the loop found in `__get_modified_files()` to see what that kicks up.

Let me know if the above isn't enough for you to do further testing...

TraffordSt, 2021-05-18:
Thanks.  I'll try this and post back soon

paddywwoof, 2021-05-18:
We have slowed down `_get_modified_files()` by making it fetch single records from the database and the select is matching a string constructed in the view column. Maybe it does get impacted by very large table size. I don't think we can put an index on the view but we could do the checking with a specific `JOIN` on `file` and `folder` using the folder, basename and extension directly.

But... this might be jumping the gun -  we don't know yet that the issue is caused by slow `__get_modified_files()`

TraffordSt, 2021-05-18:
ok, attached is a new log file after above changes.  Still no errors. 
[picframe.log](https://github.com/helgeerbe/picframe/files/6504524/picframe.log)
 Worth checking by errorhandling logic perhaps:

    def __get_modified_files(self, modified_folders):
        **try:**
            out_of_date_files = []
            sql_select = "SELECT fname, last_modified FROM all_data WHERE fname = ? and last_modified >= ?"
            for dir,_date in modified_folders:
                for file in os.listdir(dir):
                    _base, extension = os.path.splitext(file)
                    if (extension.lower() in ImageCache.EXTENSIONS
                            and not '.AppleDouble' in dir and not file.startswith('.')): # have to filter out all th$                        full_file = os.path.join(dir, file)
                        mod_tm =  os.path.getmtime(full_file)
                        found = self.__db.execute(sql_select, (full_file,mod_tm)).fetchone()
                        if not found:
                            out_of_date_files.append(full_file)
            return out_of_date_files
        **except:
            e = sys.exc_info()[0]
            self.__logger.error(e)**

jgodfrey, 2021-05-18:
Based on your latest log...

I see that the folder check found `434` modified folders, as shown in this log line...

`2021-05-19 09:20:19 DEBUG    image_cache.update_cache:76 Found 434 modified folders on disk`

And, it looks like it did so in about `7` seconds (based on time stamps), which seems reasonable.

And, sorry for not being clear. I didn't intend for you to try and catch errors in `__get_modified_files()`. Really, I just wanted you to sprinkle some log lines into the loop there. More than anything, just to 1) see that it is running as expected and 2) get a feel for how long it's taking to check each file against the db.

TraffordSt, 2021-05-18:
Sorry. I didn't read that properly - rushing..  I'll do that and come back to you in probaly around an hour.

TraffordSt, 2021-05-18:
So it's now quite verbose :)  Attached is a new log and below the logic, so you can see what you're working with.  I need to spend some time with the logic to understand it but I can see it's processing files that already exist in the DB, which is no doubt correct behaviour e.g. this is file_id 1 in the table:
_2021-05-19 10:23:07 DEBUG    image_cache.__get_modified_files:305 Processing: 2004SellwoodFamilyRog&Liz50thWeddingAnn.jpg_
[picframe.log](https://github.com/helgeerbe/picframe/files/6504776/picframe.log)

    def __get_modified_files(self, modified_folders):
        try:
            self.__logger.debug('Entering __get_modified_files') # <--- add this
            out_of_date_files = []
            sql_select = "SELECT fname, last_modified FROM all_data WHERE fname = ? and last_modified >= ?"
            for dir,_date in modified_folders:
                for file in os.listdir(dir):
                    self.__logger.debug('Processing: %s', file)
                    _base, extension = os.path.splitext(file)
                    if (extension.lower() in ImageCache.EXTENSIONS
                            and not '.AppleDouble' in dir and not file.startswith('.')): # have to filter out all the Apple junk
                        self.__logger.debug('Extension found for: %s', file)
                        full_file = os.path.join(dir, file)
                        mod_tm =  os.path.getmtime(full_file)
                        found = self.__db.execute(sql_select, (full_file,mod_tm)).fetchone()
                        if not found:
                            out_of_date_files.append(full_file)
                            self.__logger.debug('Out of date file appended: %s', file)
            self.__logger.debug('Exiting  __get_modified_files') # <--- add this
            return out_of_date_files
        except:
            e = sys.exc_info()[0]
            self.__logger.error(e)

Thanks

TraffordSt, 2021-05-18:
Would the fact that it's looping through all the files explain why it takes so long to return?

paddywwoof, 2021-05-18:
It does look rather slow to me. 200 to 400ms per file just to check if it's in the db. I would expect single digit ms

TraffordSt, 2021-05-18:
Noting this is Pi model 2 I'm using

paddywwoof, 2021-05-18:
Hmm, I wonder. It's all pretty quick for me but I'm not getting it to search through tens of thousands of records. It might be faster to check like this, as folder.name, file.basename and file.extension are all indexed
```python
    def __get_modified_files(self, modified_folders):
        out_of_date_files = []
        sql_select = """
        SELECT file.basename, file.last_modified
            FROM file
                INNER JOIN folder
                    ON folder.folder_id = file.folder_id
            WHERE file.basename = ? AND file.extension = ? AND folder.name = ? AND file.last_modified >= ?
        """
        for dir,_date in modified_folders:
            for file in os.listdir(dir):
                base, extension = os.path.splitext(file)
                if (extension.lower() in ImageCache.EXTENSIONS
                        and not '.AppleDouble' in dir and not file.startswith('.')): # have to filter out all the Apple junk
                    full_file = os.path.join(dir, file)
                    mod_tm =  os.path.getmtime(full_file)
                    found = self.__db.execute(sql_select, (base, extension.lstrip("."), dir, mod_tm)).fetchone()
                    if not found:
                        out_of_date_files.append(full_file)
        return out_of_date_files
```

jgodfrey, 2021-05-18:
> Hmm, I wonder. It's all pretty quick for me but I'm not getting it to search through tens of thousands of records. It might be faster to check like this, as folder.name, file.basename and file.extension are all indexed (I assume, as they're defined as unique)

@paddywwoof - Definitely worth a try, though all of those fields are not defined as unique... Really, just `folder.name` is unique in the above references. I'd be surprised if retrieving more fields and assembling that string dynamically during the query is faster, but I really don't know...

jgodfrey, 2021-05-18:
> It does look rather slow to me. 200 to 400ms per file just to check if it's in the db.

Agreed - this is much slower than I'd expect too.

paddywwoof, 2021-05-18:
I think with the view lookup the db has to assemble *all* of the `fname` fields to try the match whereas the JOIN system ought to be clever enough to skip over records with any of the components not matching. We have a `UNIQUE(folder_id, basename, extension)` in the `file` table so I would expect that to involve indexation in some way.

Anyway, proof of the pudding is whether it makes any difference. It runs without error on my computer at least (on the basis of copying in one fairly small folder of pictures)!

jgodfrey, 2021-05-18:
I wonder if we could be a little more clever with finding _possibly_ new or modified files... For example, maybe we could...

- In the folder table, record the timestamp when we last scanned it for changes
- Then, for every folder that comes up as `modified` from the `__get_modified_folders` call...
   - From disk, retrieve only the files that have newer timestamps than the recorded stamp for the folder
   - Then, from that (hopefully much smaller) set of files, check them against the individual file db records as we do today.
   
I assume that be faster in many circumstances. Especially where a small number of files is changed in a folder that contains lots of files. In that scenario, we'd have to check them all today. With the above, we'd only have to check the subset...

jgodfrey, 2021-05-18:
> We have a `UNIQUE(folder_id, basename, extension)` in the `file` table so I would expect that to involve indexation in some way.

I think that creates an index on that _combination_ of fields. Though, I don't think it creates any index for them individually.

> Anyway, proof of the pudding is whether it makes any difference.

Agreed. :)

jgodfrey, 2021-05-18:
Another optimization that might (significantly?) improve the initial load process...

Today, we treat both _modified_ files and _new_ files the same. That is, they're both picked discovered in `__get_modified_files()`. However, if we split that into two methods - one for _new_ files and one for _modified_ files, we could skip the initial db lookup for the new files.

That is, if we are given a folder that isn't yet recorded in the `folder` table, we know that its files won't exist in the `file` table, so really, there's no need to look for them. They could simply be inserted.

There may be some details there I'm not considering, but maybe there's an opportunity there...

TraffordSt, 2021-05-19:
Not logic related, but it's interesting to me that when the images resided on the SD I didn't encounter this problem.  This has only started when moving the images to a USB and creating a new DB.  I read online that a USB should be quicker than the SD to access.  But regardless, the fact remains that with the images on the SD ~31k of ~32k were indexed in the DB whereas I'm only indexing ~12k on the USB.  I'm assuming here that this isn't an images causing this issue since I've deleted potentially problematic images but the problem still occurs around the same image number.  And of course, these images have been indexed before when on S drive.

TraffordSt, 2021-05-19:
Anyway, happy to run anything through this image set that you want to try

jgodfrey, 2021-05-19:
@TraffordSt,

You could certainly try the version of `__get_modified_files()` Paddy posted above. Though, you'll probably have to sprinkle your logging additions back in to get a feeling for whether it's more performant or not (than the original).

TraffordSt, 2021-05-19:
It doesn't start, may be something I did. I just can't dig into it right now but will later

TraffordSt, 2021-05-19:
New log attached.  it appears to be finding only a single file.
[picframe.log](https://github.com/helgeerbe/picframe/files/6505542/picframe.log)

TraffordSt, 2021-05-19:
I found a cut and paste error.  New log attached.  
[picframe.log](https://github.com/helgeerbe/picframe/files/6506284/picframe.log)

paddywwoof, 2021-05-19:
Well that looks a lot (10 to 15x) faster, but has it make any difference to the actual problem!

>I'm not even sure it creates an INDEX for the combination of fields

[It looks to do](https://sqlite.org/lang_createtable.html#uniqueconst) I don't really know the internals of how db engines do this kind of thing  I just imagine I was writing something in python or Rust and using lists and dictionaries for tables and indexes and that's what I would do! (But UNIQUE without an index becomes an extraordinarily heavy load as the whole table needs traversing for every insert or modify.) i.e. first get the matching folder_id values, then using those in conjunction with basename and extension, look up records using the index.

TraffordSt, 2021-05-19:
Hi.  I turned off DEBUG logging as it was growing so quickly, so i restarted 3 hours ago.  I would have normally started to see some growth in the count of the file table by now.  Will let you know how it's looking. On the subject of speed, would it be an option to store the folder and file tables in caching to speed it up further?

TraffordSt, 2021-05-19:
No movement in the file count.  I might put some more logging in tomorrow so we can see if it's making it to some of the other methods and perhaps add some error handling in case it's erroring out.

paddywwoof, 2021-05-19:
And, presumably, it's not put all the image files into the db. Maybe it's just getting to a problem file a bit quicker!

I imagine that sqlite does as much as possible in memory - if you close the db without committing it will lose the data. However, to test this you could do something like this in configuration.yaml
```yaml
  db_file: '/dev/shm/pictureframe.db3' # database used by picture frame
```
which not use the SD card so will vanish after reboot.

As Jeff said, the exif checking is the most likely place for something to go wrong in the image_cache process and the fact that if there is an error it doesn't crash the application means it must be in, or a function call from within, a try block. However I can't see what might be doing that. All the except blocks apart from the image size log a message.

jgodfrey, 2021-05-19:
> [It looks to do](https://sqlite.org/lang_createtable.html#uniqueconst)

Nice - well done Paddy!

> No movement in the file count.

It won't start inserting any files until it's _discovered_ all of the files it needs to insert. So, it'll first have to crawl all of the _modified_ or _new_ folders and find all of the _modified_ or _new_ files. Once it's built that list, it'll begin inserting the file records. However, if you stop the process before the disk crawl finishes, it'll have to start all over next time. With the number of files you're looking at, it may take it a while (though, it seems that Paddy's change should have speeded that up considerably).

> And, presumably, it's not put all the image files into the db.

Right. With respect to @TraffordSt's current log messages, it should only (eventually) insert the files where you see an `Out of date file appended` message...

TraffordSt, 2021-05-20:
Hi.  I've riddled the file in logging and I've had it runnning for a few hours.  It appears to fail on insert/metadata but there are no exceptions.  There's no increase in count of file table.  It's continuing to run at this stage
Cheers
[picframe.log](https://github.com/helgeerbe/picframe/files/6512148/picframe.log)

jgodfrey, 2021-05-20:
Yeah, it definitely looks like the thread handling the image loading died... Here's what I see...

- 8:52:05, Service started 
- 8:52:27, 434 modified folders found (22 seconds)
- 8:52:27, Begins searching for modified files
- 9:16:56, 20777 new or modified files found (~24 minutes)
- 9:17:05, Begins to insert new files
- 9:17:15, Starts to insert the 5th file (of the 20,777 found)
- 9:17:16, Exif retrieved for 5th file (File: DSCF2152)
- Line 88402, About here, the image load thread seems to die (but nothing logged)

There are no further messages logged from the thread managing image load. Only messages from the running slideshow.

From this point on, nothing will be loaded. The only way to start the load process again is to restart the service. But, then all of the disk crawl info has been lost and needs to be done again, right up to where it'll likely die again... Grrr...

It's not clear whether its dying in `__get_exif_info()` or sometime soon after. I assume you've attempted to catch errors in `__get_exif_inf()` but haven't seen any, correct?

TraffordSt, 2021-05-20:
Yes, correct.  I wonder about my error handling syntax but then I'd have thought there would be an empty 'error'  log with no message if the exc_info() returned nothing.  I might review that and run it again. Tempting to just delete the file and see what happens but I'm keen to understand it, if you don't mind being patient on this..

jgodfrey, 2021-05-20:
Yeah, I'm definitely game to continue trying to assist. Though, I am sorry you're having to spend so much time on it too... :(

We'd definitely like to get to the bottom of this one. Thanks for hanging in there... :)

jgodfrey, 2021-05-20:
Related, I'd be curious what's in these two variables (from your above code):

        meta = self.__get_exif_info(file)
        meta_insert = self.__get_meta_sql_from_dict(meta)

If you're going to run it again, can you maybe log those too? Since it's only processing a handful of files before dying, it shouldn't cause too much log file congestion...

TraffordSt, 2021-05-20:
It's no problem, it's helping me to brush up on Python.  I see I should have included a couple of libraries for that excepton handling to work!  Sorry about that.  I'll add some more logging and kick it off.  Judging by the times it took only 40 mins to get to to the point of failure last time.
Yes, I'll see if I can add those

TraffordSt, 2021-05-20:
So the 'file' variable is a string of path/file so easy to add.  The 'meta' variable is the metadata array, if it comes back and doesn't fail (I'm sure you're aware of all this!).  Isn't the metadata already logged though?  e.g,
88016 2021-05-20 09:17:05 DEBUG    classes._process_tag:289  Software: (0x0131) ASCII=Digital Camera X-T10 Ver1.01 @ 190

jgodfrey, 2021-05-20:
I'm mostly interested in the contents of  `meta` and `meta_insert` in the above code. `meta` will be the dictionary that's constructed in `__get_exit_info()`, which isn't the same as the log record you reference above. `meta_insert` will be the SQL statement that's constructed from the meta dictionary. And, I'm sort of expecting the problem to be in one of those two functions, so one of them may very well fail and not give us the results we're after. But, even that's a step in the right direction...

TraffordSt, 2021-05-20:
oh you mean those variables :)
It's up and running now.

TraffordSt, 2021-05-20:
Ok, we're getting somewhere.  We have an error (division by zero).  There must be something wrong with my logic iterating the dict since we're not getting those values.  I'll have a look at that.  Here's the log, zipped this time:
[picframe.zip](https://github.com/helgeerbe/picframe/files/6512572/picframe.zip)

jgodfrey, 2021-05-20:
Nice!

Yeah, so I'd guess it's probably this line from `get_image_meta.py` in `get_exif`:

`val = round(val.values[0].num / val.values[0].den, 1)`

And, it's being fed this data (from your log):

`FNumber: (0x829D) Ratio=0/0 @ 870`

jgodfrey, 2021-05-20:
So, I'd say definitely invalid EXIF data, but we certainly need to handle that better...

jgodfrey, 2021-05-20:
Although, that error doesn't appear to have stopped the file loading thread...

[20 more replies]

closed 2021-07-06 by helgeerbe