helgeerbe/picframe #125: After deleting an image, the previous image is shown

sapnho, 2021-05-31:
I noticed that after you execute the MQTT command to delete an image, the previous, not the next, image is shown.

Is that a bug or a feature?

paddywwoof, 2021-05-31:
Sounds slightly illogical but I can imagine why it might happen. I'm away at the moment but will have a look at this and the edge wrapping when I'm back later this week.

sapnho, 2021-07-05:
Am I the only one where this happens? Would be great to have this fixed if it's easy to do. Thanks!

jgodfrey, 2021-07-05:
While I'm not sure _why_, it seems the code intentionally rewinds to the previous image upon deletion as can be seen [here]([url](https://github.com/helgeerbe/picframe/blob/b0fddbb9df1b771a9133dc3dd5ea92c18b9165df/picframe/controller.py#L81)).

paddywwoof, 2021-07-05:
From the comment and similar in ```Model.delete_file()``` Maybe there was an edge case, if the deleted image was last in list or there was only one image. However I can't see either of those causing a problem. I will try commenting out the ```back()``` on my frame here and see if there are any issues.

sapnho, 2021-08-21:
Has this fix been included in the latest PictureFrame version by any chance?

sapnho, 2022-06-04:
Works wonderfully in the 2022.06.03 version. I also updated the blog post (https://www.thedigitalpictureframe.com/integrate-raspberry-pi-picture-frame-home-assistant/).
Thanks much!

closed 2022-09-22 by helgeerbe