helgeerbe/picframe #126: picframe losing track of images

jeffalperin, 2021-06-01:
After running successfully for many hours, picframe will abruptly "forget" where the image files are located and show only the no_pictures.jpg image.  After this happened several times, I ran picframe in DEBUG mode and captured all the output to a log file (attached).  See line 116752 for the first occurrence of no_pictures.jpg. 
[picframe-command-log.zip](https://github.com/helgeerbe/picframe/files/6577258/picframe-command-log.zip)

I've also attached my configuration.yaml file and the pictureframe.db3 file.
[config and sqlite.zip](https://github.com/helgeerbe/picframe/files/6577328/config.and.sqlite.zip)

In case it matters, I have integrated picframe with the latest version of Home Assistant as described in thedigitalpictureframe.com.  I'm still testing, but it may be that the problem occurs sometime after I switch image directories in Home Assistant using the directory select helper entity ```input_select.picframe_directory```.

One other odd thing I noticed:  Although I have ```pic_dir``` set to ```/home/pi/4kpix``` in my ```configuration.yaml``` file, the sqlite db also has entries for files located in ```/home/pi/Pictures```.  (That directory **does** exist and contains lower resolutions of the same files found in ```/home/pi/4kpix/Pictures```.  

Is is possible that picframe is getting confused and trying to find image files at ```/home/pi/Pictures/Pictures```?

--Jeff

jgodfrey, 2021-06-01:
Hi Jeff,

Looking at your log, it looks like (as you mentioned), you've switched the subdirectory assignment a few times prior to this happening. What are you sending as the "subdirectory" when you attempt to return to the top-level `Pictures` folder? Ideally, you should just send an "empty" folder name I think...

Related, it looks like we should improve the logging in this area as it only lists the specified subdirectory when it's changed. It'd be nice to see the full, absolute path it's looking at in this case...

Related to the having DB records for 2 top-level folders (`/home/pi/Pictures` and `/home/pi/4kpix/Pictures`)... I assume you've switched the top-level folder since the DB was created? In that case, I'd expect the DB to still retain records for both folders - as is shown in your DB file.

jeffalperin, 2021-06-01:
Jeff:  Thanks for the most helpful response!  My image directories are structured like this:
```/home/pi/4kpix
/home/pi/4kpix/Pictures
/home/pi/4kpix/travel
/home/pi/4kpix/travel/London
/home/pi/4kpix/travel/NewYork
```
so I think you're right:  I should only be using these options:  ```Pictures```, ```travel``` and ```""``` if my root directory is ```4kpix```.
Instead I had ```4kpix``` specified instead of   ```""``` and that would cause picframe to look for images in the non-existent ```/home/pi/4kpix/4kpix```.

What's still a little odd is that even if I try to switch subdirectories in Home Assistant by changing the input select to ```travel```, Home Assistant remains "stuck" showing no images.  (This also happens even if I send an MQTT message manually.) Perhaps it's not handling a file not found error and just hangs?  (Maybe this is such an extreme edge case it's not worth dealing with in the code?)

I agree that the logging enhancement you suggest would be helpful.  What about adding a date/time stamp, too?

jgodfrey, 2021-06-01:
Hi Jeff,

Are you in a permanently broken state ATM, or can you get your frame working again? That is, does the problem only crop up when you're switching subfolders, or is it always broken for you now? And, if the latter, does either restarting the frame service (or, simply rebooting the frame) get it back into working order?

jeffalperin, 2021-06-01:
Jeff:  I can always get the frame working again by re-starting the frame service, but it will eventually fail.  The problem seems to be related to switching subfolders, but on at least one occasion, it ran without problem for more than a day, then "broke" without my having changed subfolders.

Two other odd things I noticed after a little more testing:

1)  I stopped and restarted picframe and then deliberately switched the subdirectory in HA to ```4kpix``` which is the root directory, expecting to have the no images problem -- but everything worked fine.  I switched subdirectories several times, but still no problem. 
2) When I re-checked the log that I attached to this issue, I found two instances of the following when I switched to the ```4kpix``` directory:
```
INFO:interface_mqtt.InterfaceMQTT:Recieved subdirectory: 4kpix
INFO:model.Model:Set subdirectory to: 
```
Note that the new subdirectory name is blank, which is appropriate since ```4kpix``` is the root directory.  
So, it looks like the picframe code already "knows" to adjust for this circumstance?

jgodfrey, 2021-06-01:
Jeff,

Thanks for the reply, and glad to know you're not "permanently" broken ATM. 

Looking at the code, it does seem that some attempt was made to figure out (and nullify) the case where you specify the root folder name as the subfolder, though without a deeper dive (or better logging), it's difficult to be 100% sure.

It definitely seems like there's a bug here somewhere.

Are you running from the packaged installation or from source code? Also, are you running the most current version?

helgeerbe, 2021-06-02:
Hi Jeff,

I guess we are facing two independent issues here? The first was the setting of the subdirectory 

> so I think you're right: I should only be using these options: Pictures, travel and "" if my root directory is 4kpix.

This is correct. I hope  this is solved now?

The second issue

> it ran without problem for more than a day, then "broke" without my having changed subfolders.

Have you done any interaction with PictureFrame? Or did it broke while just showing images?
What do you mean with broke? Complete crash, or some strange behavior (like no-pictures image)?

I run the dev branch for a couple of weeks and it appears rock solid to me.But my only interaction is to turn display on/off.

jeffalperin, 2021-06-02:
Helge:

No, I think there's **one** primary issue:  picframe gets 'confused' and shows either the no-pictures image or a black screen -- without any immediate interaction from me.  (Whether a black screen or the no-pictures image, Home Assistant is showing that the current image is the no-pictures image.)  Once this has happened, picframe is no longer responsive to mqtt commands to switch to another subdirectory; to resume the slideshow, the process must be terminated from the command line (with CTRL - C) and restarted.  It appears to me that this "no-pictures behavior" is somehow related to switching subdirectories in Home Assistant, but it does not occur immediately after a switch.

As to the issue setting the subdirectory using Home Assistant:  it's true that I was using the root **directory** name as a **subdirectory** option.  But in my testing, this did not immediately cause the "no-pictures behavior" problem (as I would expect).  This may be because in the log file, it appears that picframe recognized that the root directory name should be converted to a blank string:
```
INFO:interface_mqtt.InterfaceMQTT:Recieved subdirectory: 4kpix
INFO:model.Model:Set subdirectory to: 
```

Also, when I wrote that picframe "broke," I meant that it showed the "no-pictures behavior" not that it crashed.

Can I add some additional logging code somewhere as Jeff G. suggests above?

Thanks for your help looking into this!

--Jeff

jgodfrey, 2021-06-02:
I'm curious if rebuilding the cache has any impact on the issue? If you're game to experiment, you could:

- Stop the `picframe` service
- Remove (or, simply rename) the existing cache database file (`pictureframe.db3`)
- Start the `picframe` service

That will cause the cache to be rebuilt from scratch.

If you try that, let me know if it changes anything...

jeffalperin, 2021-06-02:
Experimenting underway!  I'll report back.

jeffalperin, 2021-06-02:
I just had a thought . . . Although I set up a picframe.service to run on boot, I've mostly been running picframe manually from the command line like this:  `sudo xinit /usr/bin/python3 /home/pi/picframe_data/run_start.py /home/pi/picframe_data/config/configuration.yaml`.  Should I be starting/stopping the service instead?  (I assumed either way would be equivalent, and I figured out how to view and capture the console output when I run picframe manually; not sure how to do that if it's running automatically as a service.)

jgodfrey, 2021-06-02:
Yeah, you're correct. Either way should be equivalent, and running manually certainly makes it easier to capture log output. 

Related, @paddywwoof made some changes recently to make capturing log info to a file easier (via the config file) - which should also work when running as a service, but I don't think those changes have been merged yet.

Bottom line - what you're doing sounds fine to me.

helgeerbe, 2021-06-02:
I would also recommend to delete the cache (delete the database) to have a clean start. 

Hope that would do it. Actual I don't have an idea why the display should become black. 

The new logging changes are not merged to main, yet.

jeffalperin, 2021-06-02:
Well, I did delete the cache and restart picframe, but the "no pictures" behavior recurred within 2 hours.   I changed the directories through HA several times, but the problem did not recur immediately after doing so.  In fact, looking at the debug log, it seems that it displayed about 54 images after the last directory switch before suddenly it showed the no-pictures image.

jgodfrey, 2021-06-02:
Hmmm... And, once you get the "no pictures" image, it's stuck at that point, requiring you to restart the software?

You don't, by chance, have a copy of the "no pictures" image mixed in with your other pictures and are being fooled into thinking there's a problem when it gets displayed?  

(wishful thinking I'm sure)  :)

jeffalperin, 2021-06-02:
Correct.  And no, there's only one "no pictures" image, and it's in its default location.

Two other random facts:

- I just noticed that picframe isn't totally hung because I caught it displaying the text "no_pictures.jpg * data" on the otherwise blank display.  I then used Home Assistant to request that it refresh the text, and it did so.  So it's definitely still alive and listening and responding to some things.
- The root directory and subdirectories are being synced to a Win 10 desktop using SyncThing.  3 new files were synced, but, according to the log, only after the most recent problem.  Also, I didn't sync any photos in any of the prior instances of the problem.

jgodfrey, 2021-06-02:
Thanks for the additional info.

I'm not a Home Assistant user, but I do control my frame via the MQTT interface frequently (from my Android phone). I assume you have access to the "Shuffle" command via Home Assistant? I'm curious if that makes the frame work again once it's gotten stuck? I ask as that should cause the "playlist" to be rebuilt from scratch using the current contents of the DB.

While it's certainly not a solution, it could provide another data point for us...

jeffalperin, 2021-06-02:
Sadly, turning shuffle off did nothing to revive picframe.  Neither did turning shuffle back on.  I tried both through HA and by sending mqtt messages directly.

paddywwoof, 2021-06-03:
Hi, this sounds like a frustrating bug. I'm going to be out again today but might get a chance to look at it later. Here are some thoughts off the top of my head:

a) It's possible to get a list of ```systemd``` caught output by typing ```journalctl``` at the command prompt - but I'm not sure you will learn anything extra by running picframe as a service rather than from cl
b) It would be nice to isolate the problem a bit more. I would i) in ```configuration.yaml``` set ```shuffle: False``` so it always runs in the same order from start and don't send *any* command by MQTT, HA etc. ii) *after* running in fixed order and finding the picframe freezes at a different picture in random way, turn HA off or disconnect it by whatever means iii) if the problem continues turn SyncThing off
c) The fact that picframe continues to run means that ```Controller.loop()``` is still OK but possibly one of the other loops is broken, ```ImageCache.__loop()``` had a problem before because of an error getting exif info in the background but this seems to be different. The only work done by ImageCache once the db has been filled is the location info. Anyway if the show always stops on the same image you will know this kind of thing is probably the cause.

Paddy

helgeerbe, 2021-06-03:
Paddy is right. 

a) stop picframe
b) shuffle: False
c) delete database
d) stop SyncThing (to ensure, it does not interfere)
e) start picframe, but don't send any commands

Is this stable?

Then send commands?

Does it break? If not, enable SyncThing. Don't send further commands

Does ist break?

jeffalperin, 2021-06-03:
Testing now per your protocol.

FWIW, I ran it overnight (before receiving your protocol) and it stopped after about 380 images.

Will report back.

helgeerbe, 2021-06-03:
Jeff, Thanks for testing.
I'm wondering, if picframe stops always on the same image. We were told recently about an image with corrupted exif information which results in a div by 0 error. Paddy fixed this in the actual dev branch.

jeffalperin, 2021-06-03:
Each log shows a **different** image as the last successfully found file, but there's no information in the logs about the filename picframe is seeking when it fails.  Is there a logging statement I can add?

jgodfrey, 2021-06-03:
> We were told recently about an image with corrupted exif information which results in a div by 0 error. Paddy fixed this in the actual dev branch.

Though, this specific issue resulted in the image caching thread dying, which stopped the caching of additional images, but the slideshow continued to run fine. So, as @paddywwoof said above, this seems to be a different issue.

> Each log shows a different image as the last successfully found file, but there's no information in the logs about the filename picframe is seeking when it fails.

Does that imply that you ran all of @paddywwoof's suggested scenarios, and each one died?

> Is there a logging statement I can add?

It'd be easy to add some additional logging if you were running from source code. It should still be possible when running the packaged version, but I'm not sure where everything lives in that case. @paddywwoof or @helgeerbe probably know the secret sauce here.

If you can locate (and impact) the running code, we can definitely come up with some more targeted logging.

jeffalperin, 2021-06-03:
Status report.  picframe has been running without stopping for several hours now.  However, there's another odd thing I discovered in the current log:  Although there are 177 files it should be displaying in fixed rotation (since ```shuffle``` is off), the log is showing that only 40 of those are actually being displayed.  Indeed, ```image_counter``` shows '40' throughout the log.  Here is a sample entry:
```
INFO:interface_mqtt.InterfaceMQTT:Send state: {'directory': '4kpix', 'image_counter': '40', 'image': 'Olympic Peninsula_20140901_0656-4k.jpg', 'date_from': -2147367600, 'date_to': 2145934800, 'time_delay': 60.0, 'fade_time': 5.0, 'brightness': 1.0, 'location_filter': '', 'tags_filter': ''}
```
Here's how the image directories are configured now:
```
/4kpix
/4kpix/Olympic Peninsula (4 files in directory; 4 files shown on frame*)
/4kpix/Pictures (67 files in directory; 0 files shown on frame)
/4kpix/travel/Canadian Rockies (71 files in directory; 36 files shown on frame)
/4kpix/travel/Europe_2019 (29 files in directory; 0 files shown on frame)
/4kpix/travel/test (6 files in directory; 0 files shown on frame)
```

*"shown on frame" really just means there's a log entry for the image

So, there are 177 images in the directory structure, but only 40 are being displayed.  (The sqlite db also shows entries for 177 files.)

jeffalperin, 2021-06-03:
Also, looks like the picframe code files are located at ```/usr/local/lib/python3.7/dist-packages/picframe```, so presumably I could stop picframe, edit those files to add additional logging code, and then restart picframe from the command line, right?

jeffalperin, 2021-06-03:
ok.  picframe **stopped again** without any intervention from me (no Syncthing syncing, no Home Assistant commands, no direct mqtt messages).  Looks like it displayed about 495 images (always showing ```image_counter``` = 40 before getting confused and showing ``` 'image_counter': '0', 'image': 'no_pictures.jpg' ```

jeffalperin, 2021-06-03:
**Major progress:**

I just realized that the log also shows that the ```date_from```/```date_to``` filter values change from ``` 'date_from': -2147367600, 'date_to': 2145934800 ``` to ```'date_from': -68400, 'date_to': -68400``` when picframe stops.  If that filter is being applied, then of course there are no files found with dates between those identical values.  As a test, I changed the ```date_to``` value via Home Assistant, and the frame started showing images again!  (It also showed 177 images available in image_count in HA.

So now we know why no images are displayed, but we still need to determine why the date_filter values are changed without user action.

jgodfrey, 2021-06-03:
Ah, great find @jeffalperin!

And, yep, that bad date range would definitely cause the results you reported... 

I'll poke around a bit in the code to see if anything jumps out at me based on the new info.

jgodfrey, 2021-06-03:
Looks like you're narrowing it down (based on your above edit). Looking in a log you posted much earlier in this thread, I see the same issues evident there. For example, I see this:

```
INFO:interface_mqtt.InterfaceMQTT:Recieved date_from: -68400
INFO:interface_mqtt.InterfaceMQTT:Recieved date_to: -68400
```

So, the errant date is definitely being published via the MQTT interface (as you noted in your above edit). Not being an HA user myself, I probably won't be of much help debugging the posted automation, but I bet @helgeerbe or @paddywwoof will have some input.

Thanks for sticking with this.

jeffalperin, 2021-06-04:
Looking through the Home Assistant logbook, it seems that the various picframe entities _and other mqtt entities_ are marked in the logbook as "became unavailable" at the same time that ```automation.picframe_date_from_set``` fired and changed the from_date to -68400.  

Here's what I suspect is happening:

If the mqtt service is briefly unavailable for some reason, the state of ```sensor.picframe_date_from entity changes``` (to 'unavailable'), triggering the ```automation.picframe_date_from_set``` automation.  And I bet that the automation incorrectly handles the 'unavailable' state, perhaps assuming that it is zero, when computing the value of the mqtt payload.    Instead, the automation's condition section needs to check to be sure the state has really changed and isn't just unavailable.

If this is correct, not sure exactly what changes are necessary for the automation scripts.

helgeerbe, 2021-06-04:
Hi Jeff,

Thanks for figuring it out. It seems indeed that your network becomes for a very short period unavailable. The script reads the (wrong) sensor value (date) and sends it to picframe (which is available again).

I have to figure out, how to detect this in home assistant. 

A workaround could be, to 
- disable the date getter automations
- remove the condition check in the setter automations. 
- set input_helpers to a good starting value

That means home assistant becomes the master. And will not use the dates from configuration.yaml. And you won't be able keep  the web interface and home assistant in sync.

helgeerbe, 2021-06-04:
Funny.

```
{{ "unavailable" | int | timestamp_custom('%d.%m.%Y')
    }}
```

is `01.01.1970` 

So I propose, to test for "unavailabe". Can you try this.

````YML
alias: picframe.date_from_set
description: ''
trigger:
  - platform: state
    entity_id: input_text.picframe_date_from
condition:
  - condition: and
    conditions:
      - condition: template
        value_template: >-
          {{
          as_timestamp(strptime(states("input_text.picframe_date_from"),"%d.%m.%Y"))
          | int != states("sensor.picframe_date_from")|int }}
      - condition: template
        value_template: '{{ "unavailable" != states("sensor.picframe_date_from") }}'
action:
  - service: mqtt.publish
    data:
      topic: picframe/date_from
      payload: |-
        {% if states("input_text.picframe_date_from") != "" -%}   
          {{ as_timestamp(strptime(states("input_text.picframe_date_from"),"%d.%m.%Y")) |     int }} 
        {%- endif %}
  - condition: template
    value_template: '{{ states("input_text.picframe_date_from") == "" }}'
  - service: input_text.set_value
    data:
      value: >-
        {{ states('sensor.picframe_date_from') | int |
        timestamp_custom('%d.%m.%Y') }}
    entity_id: input_text.picframe_date_from
mode: single

````

jeffalperin, 2021-06-04:
Helge:  I just tested by stopping my mosquitto broker and observing the state changes of the ```sensor.picframe_date_to``` and ```sensor.picframe_date_from``` entities.  When mqtt is stopped, the sensors each show a state of 'unavailable' and when mqtt is restarted, they revert to their prior values.  So couldn't the getter/setter automations check to be sure the changed state is **not** 'unavailable' before continuing?

helgeerbe, 2021-06-04:
Perfect.

As addition. I added a second action. If you enter a blank date, input_text ist set to sensor value

jeffalperin, 2021-06-04:
Helge:  Just tested your new code by stopping the mqtt service.  The frame continues to run, even with mqtt unavailable and the state of ```sensor.picframe_date_from``` as 'unavailable'.  The only "issue" is that the HA interface shows to/from dates as 1969-12-31, but I assume that's because there need to be corresponding changes made in the getter automations?

helgeerbe, 2021-06-04:
Jeff: Thanks for the feedback. You're right. The getter automations also require the check, not to change the input_text fields, so they keep their actual values.

I wonder, if this check is needed for all automations like brightness, fade time, ......

Guess, this is something for the weekend. Thanks again for figure it out. I will provide a new version of the yaml automation for home assistant.

jeffalperin, 2021-06-04:
See a similar suggestion in the Home Assistant documentation [here](https://www.home-assistant.io/integrations/template/#considerations) which discusses adapting templates to check for ```unknown`` state on startup.

helgeerbe, 2021-06-06:
Jeff: I updated the automations scripts. Would be good, if you can test it.
Every single automation is now checking for sensor state ('unknown', 'unavailable' and 'none') before changing the values. I also did some minor bug fixes. Brightness slider should work now correctly.

You have to remove the picframe automations in your automations.yaml file and add the content of the attached file. And you have to change the date format(it's German) to your local date format.

[automations.zip](https://github.com/helgeerbe/picframe/files/6604835/automations.zip)

jeffalperin, 2021-06-06:
thanks!  I will start testing now.

jeffalperin, 2021-06-07:
Helge:  Seems to be working great!  I ran it overnight with no problems and have stopped and started mqtt and the frame keeps functioning.

helgeerbe, 2021-06-07:
I updated the [wiki] (https://github.com/helgeerbe/picframe/wiki/Homeassistant) and download file.

closed 2021-06-07 by helgeerbe