helgeerbe/picframe #127: Directory listing in HomeAssistant drop-down menu is not alphabetized

jeffalperin, 2021-06-14:
The drop-down menu for ```input_select.picframe_directory``` appears to be in an arbitrary order rather than alphabetical.

I've not tried yet, be perhaps this could be addressed by adding ```|sort``` in the getter (`picframe_directory_get`) and/or setter (`picframe_directory_set`) automations for ```input_select.picframe_directory```?

helgeerbe, 2021-06-25:
Hi Jeff,

sorry for the delayed answer. Currently picframe just reads the folder structure and doesn't do any kind of sorting. So I guess to do the sort in in the home assistant template would be fine. Have you done it already and can you provide the code snippet? Otherwise I could do it.

jeffalperin, 2021-06-25:
Hi, Helge.  I have not done it.  (I started playing with it, but got distracted with another project.)  Thanks for taking a look!

helgeerbe, 2021-06-26:
Hi Jeff,

try this in ``picframe_directory_get`` automation

````yaml
service: input_select.set_options
data:
  options: '{{ state_attr(''sensor.picframe_directory'', ''directories'') | sort }}'
entity_id: input_select.picframe_directory
````

Is this working for you?

jeffalperin, 2021-06-26:
Yes, that works!  But only after a complete restart; just reloading the automations didn't change anything.

helgeerbe, 2021-06-27:
That's seems to be correct. The automation checks if the option list and the folder list differs. So creating / deleting a folder would trigger the automation.

But interesting. The comparison seems to work as long the lists containing the same elements. So no need to sort the sensor list before comparison.

````YAML
{{ ( state_attr("sensor.picframe_directory", "directories") ) != ( state_attr("input_select.picframe_directory", "options") ) }}
````

jgodfrey, 2021-07-05:
Can this be closed?

jeffalperin, 2021-07-05:
Yes, thank you!

helgeerbe, 2021-07-05:
I reopen this, until I provide a new yaml automation script for downloading in the wiki.

helgeerbe, 2021-07-06:
Updated automation.yaml in [wiki](https://github.com/helgeerbe/picframe/wiki/Homeassistant#20210706)

closed 2021-07-06 by helgeerbe