helgeerbe/picframe #128: Proving that every picture is shown..

sapnho, 2021-06-20:
I guess everybody is enjoying the nice summer weather and does not feel like tinkering with their picture frame.

But when the rain and nasty weather comes back, I was wondering if we could tackle an issue that should be easy to prove now that we have a database within Pi3D.

The issue is to prove that statistically there is an equal distribution of the images shown. My wife is convinced that our frame has "favorite" images and I can't say for sure that this is not the case because at least psychologically, it does indeed feel that way! :-)

So, I was wondering if it was possible to add a counter column to each image in the database which adds +1 every time an image comes up. If only for debugging purposes, it would be fun. :-)

jgodfrey, 2021-06-20:
Interesting that you mention this Wolfgang, as I have similar feelings too - though I really do think it's psychological...

I've had a slideshow that runs as the screensaver on my Roku-connected TV for years. It's connected to a number of Google photo albums and I *swear* that it only shows a subset of those images. I've had several conversations about that with the author over the years and he claims that's just not the case.

Now, fast forward to using picframe - where I'm relatively familiar with the inner workings, and... I still have the same feeling. Several things make me think there really isn't a problem though, including:

- I've written the "playlist" used by the frame to a file at the start of a show. This really just consists of one or two image ID's per record (due to single or paired images). I've taken that list, split it into single ID's, and sorted it. The results were as expected - just a sequential list of image ID's that covered my entire image set.
- When I "browse" the images that are stored on the frame (using separate software), I generally can say "Oh yeah, I've seen that one recently on the frame".

So, while I don't think there's an actual problem here, I've definitely considered adding some kind of (persistent) mechanism to the code to prove exactly that to myself....

helgeerbe, 2021-06-21:
Good point. My  family has the same feeling. On point could be the shuttle function with n recent days? Or the shuffle ist not a real shuffle. Guess, some statistics would be nice.

sapnho, 2021-06-21:
Funny, indeed. I would expect this to be purely coincidental but having some stats would indeed be helpful to remove any doubt. :-)

jgodfrey, 2021-06-22:
I'm currently working on this... 

I've added `displayed_count` and `last_displayed` fields to the `file` table. The `displayed_count` gets incremented each time a file is displayed and `last_displayed` records the date/time stamp for when the file was last displayed.

Does that sound reasonable? Is there anything else that'd be helpful here?

Also, I don't have plans to provide a way to reset the `displayed_count` via MQTT or otherwise. For now, it'll have to be reset via a manual SQL statement. Do you think that's OK?

jgodfrey, 2021-06-22:
I've created a feature branch containing the initial code changes. That's here:

https://github.com/helgeerbe/picframe/tree/Issue-128-Image-Display-Stats

I'm not ready to merge it yet, but if anyone wants to try it out and report results back, that'd be great.

If you do use it, the following query should provide enough info to judge whether the image display is working correctly or not (spoiler, it does indeed seem to be working OK here so far).

`SELECT displayed_count, count() FROM file GROUP BY displayed_count`

sapnho, 2021-06-22:
Hi Jeff, that is really cool that you already implemented this. No need for MQTT remote control and other gimmicks, just being able to check it in an SQL viewer is good enough for me.

Would be great to have it in a prod version whenever ready.

Thanks!

jgodfrey, 2021-06-22:
These changes have been merged into `dev` via 1cafb772f1b17d1e2c5d8d6e37dbc2b744647868

I'll let @helgeerbe decide when they should get merged into `master`.

jgodfrey, 2021-06-23:
Normally, I do quite a bit of experimentation with my frame, which could very definitely throw off the recorded stats I added for this issue. However, after having cleared the stats, I've intentionally just "let it run" since sometime last night, and it's now on its 5th complete pass through my entire playlist (~670 images, 30 seconds per frame, with some paired portraits). 

Looking at the new "displayed" stats, there does seem to be some (very) subtle problem, though it's too minor to be "noticed" by a casual observer of the frame.

Specifically, on the 5th pass, I currently have the following:

- 442 images displayed 4x
- 226 images displayed 5x
- 2 images displayed 6x

So, those last 2 images have been displayed 1 extra time each, and I'm not sure why. I suppose it could be a discrepancy between when I record the image as having been "displayed" and when it actually is (or isn't displayed), but I don't see any obvious issue there.

Bottom line, I'm not sure what's causing the minor discrepancy (likely some subtle bug), but there certainly doesn't seem to be any big issues here...

I'm curious what others find.

sapnho, 2021-06-24:
This is interesting data. I guess you need to let it run a bit longer. With no external interaction, the way the playlists are built, it should be exactly the same number of times, shouldn't it?

jgodfrey, 2021-06-24:
> This is interesting data. I guess you need to let it run a bit longer. With no external interaction, the way the playlists are built, it should be exactly the same number of times, shouldn't it?

The numbers shouldn't ever be off by more than 1. So, in the above stats, the 4x and 5x display values are fine as it was partially through the 5th loop of the playlist. However, those two 6x display values do (seemingly) indicate an issue.

However, I've now completely invalidated my numbers by experimenting with other things that have required restarts and advancing / rewinding the playlist - which will obviously cause the displayed count values to drift apart.

I may configure the playlist to run faster (only a few seconds per image) to speed up the process and let it run overnight sometime soon. Though, I expect I'll see similar results to the above.

sapnho, 2021-06-24:
Curious to see the results!

jgodfrey, 2021-06-25:
So, after running overnight using only 3 seconds per image, the results are as follows:

- The frame was on its 21st pass through the playlist
- 492 images were displayed 21x
- 160 images were displayed 20x
- 18 images were displayed 22x

So, based on that, it seems that an image gets shown an extra time (roughly) once per pass.

I haven't yet analyzed this any further...

sapnho, 2021-06-25:
That's still pretty ok, but a bit strange given the algorithm.

jgodfrey, 2021-06-25:
> That's still pretty ok, but a bit strange given the algorithm.

Yeah, agreed. While it's not that bad, I'd say it does indicate some subtle bug. Either in the slideshow itself or my data collection code...

jgodfrey, 2021-07-05:
> So, I was wondering if it was possible to add a counter column to each image in the database which adds +1 every time an image comes up. If only for debugging purposes, it would be fun. :-)

I think this request is resolved (adding image display stats to the db). As noted in the above conversation, there does seem to be a minor issue with a few images being shown more than expected - which should be looked into, but that seems to be a separate issue from this. Do you agree @sapnho?

sapnho, 2021-07-05:
Absolutely! My wife still doesn't believe the stats but that is my problem! :-)))))))

closed 2021-07-05 by jgodfrey

jgodfrey, 2021-07-10:
So, I've discovered one situation that causes an extra play of a single image in a complete cycle of the playlist. It's only triggered (I think) by having the `shuffle` active though - which is how my frame runs by default. So, a question for anyone watching this thread...

If you've also noticed this _extra_ display  cycle (ideally, by looking at the db stats), do you also run your frame with `shuffle` active?