helgeerbe/picframe #137: picframe with running xserver?

ethpi, 2021-07-09:
Is there any possibility to start picframe on a running xserver? Using Xephyr or anything else?
I set the option `use_glx=True` but I don't know how to start picframe on a running xserver.

Background:
I run my RPI4 as a media center and try to use picframe as an imageviewer. It's really inconvenient to shut down the running xserver, then start picframe with its own xserver, shutdown picframe and restart the first xserver.

Many thanks.

paddywwoof, 2021-07-10:
I'm not really sure what 'other' xservers might be but I would have thought they were all using the same libraries underneath. pi3d (which picframe uses for the graphics stuff) just loads the libX11 as in the code below so I would expect it to just work. How have you tried running it so far and what errors do you get? What happens if you start the xserver you want to keep running then open a terminal and:
```bash
$ python3
from ctypes import CDLL
from ctypes.util import find_library
libx11_name = find_library('X11')
libx11 = CDLL(libx11_name) 
print(libx11)
```
or better
```bash
$ python3
import pi3d
disp = pi3d.Display.create(w=100, h=100) # try without first then add ,use_glx=True
for i in range(100):
    disp.loop_running()
disp.destroy()
```
Are you starting your media centre with systemctl? Are you using glx because you want the picframe window to have a transparent background? I see on the freedesktop page on Xephyr that as at the last time the page was updated (2013!!!) glx wasn't supported.

ethpi, 2021-07-10:
Thanks for your reply.

OK, too less info for you, so I try to add some. Actually I'm busy, so it may takes a time.

ethpi, 2021-07-14:
I'm sorry, but I was busy and it took a while to check everything again.

Resume:

I don't know why, but now it's no problem to start picframe within a running xserver. Perhaps it was a typing mistake in the configfile or anything else. 

Setting:
I don't use a mediacenter like kodi or xmbc but a "normal" desktop setting: 
loginmanager: sddm
windowmanager: lxde
software: chromium with drm support, mediathekview, vlc, gwenview
That's all I need.

I installed picframe as descrided in: https://www.thedigitalpictureframe.com/how-to-add-crossfading-slide-transitions-to-your-digital-picture-frame-using-pi3d/
In this article it's suggested to start picframe by:

```
xinit /usr/bin/python3 /home/pi/picframe_data/run_start.py /home/pi/picframe_data/config/configuration.yaml
```
This command doesn't work (because `xinit` can't start its own xserver?)
Ouput:
```
_XSERVTransSocketUNIXCreateListener: ...SocketCreateListener() failed
_XSERVTransMakeAllCOTSServerListeners: server already running
(EE) 
Fatal server error:
(EE) Cannot establish any listening sockets - Make sure an X server isn't already running(EE) 
(EE) 
Please consult the The X.Org Foundation support 
         at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
      after 7 requests (7 known processed) with 0 events remaining.
```
If I drop the leading `xinit` and set in the configfile `/home/pi/picframe_data/config/configuration.yaml` the option `use_glx: True` it's possible to start picframe with

```
/usr/bin/python3 /home/pi/picframe_data/run_start.py /home/pi/picframe_data/config/configuration.yaml
```
Sorry for the trouble.

closed 2021-07-14 by ethpi

helgeerbe, 2021-07-14:
@ethpi great that it works now. If I got it right, you have already a graphical environment running?! That means, that the server is already running and you don't need the `xinit` call.
`xinit` is used, if you boot to the text console, to start the xserver.

ethpi, 2021-10-17:
If anybody is interested:

I wrote a extreme simple "gui" to select the photo directory before starting picframe. It' a bash-script, that uses "kdialog". I think it's no problem to adapt it to "dialog", if "kdialog" is not installed. I'm sorry, I renamed the script to "start_picframe.sh.txt", because the suffix *.sh isn't supported.

But so far there is no possibility to stop picframe. You can do this job by using the picframe webserver.

[start_picframe.sh.txt](https://github.com/helgeerbe/picframe/files/7359714/start_picframe.sh.txt)