helgeerbe/picframe #139: Image played twice during slideshow cycle if shuffle=True

jgodfrey, 2021-07-10:
As discussed [here](https://github.com/helgeerbe/picframe/issues/128), there has been a general feeling that some images are being played more than others during a single cycle of the playlist. I added some db-based play stats to each image record (noted in the previously linked issue), which proved there was a legitimate, (although minor) issue here. According to my initial investigation, it seemed I was seeing (roughly) 1 image get an extra play per playlist cycle.

After some digging, I've found the cause (at least for my specific issue), which is only evident when the configuration option `shuffle = True`.

The problem is in `model.get_next_file()`. [Here](https://github.com/helgeerbe/picframe/blob/b0fddbb9df1b771a9133dc3dd5ea92c18b9165df/picframe/model.py#L285), we've just displayed the last image. Assuming `shuffle=True`, we set the flag to shuffle the playlist on the next pass and set `__file_index=0` (to point back at the first image in the playlist).

However, since the playlist won't actually get shuffled until the *next* cycle, we end up playing the first image in the playlist one additional time at the end of the cycle. 

So, per playlist cycle, the first image displayed is also played one more time at the end. Once that's done, the playlist is reloaded, and the (likely different) "image 0" is played at the start of the playlist (and will be replayed at the end).

I've made changes to the same area in the open PR https://github.com/helgeerbe/picframe/pull/138 to fix issue https://github.com/helgeerbe/picframe/issues/119 and am now testing an experimental fix for this issue.

If anyone has input (testing or otherwise) on PR https://github.com/helgeerbe/picframe/pull/138 that'd be great. I'd like get it merged and then build upon it for this fix.

paddywwoof, 2021-07-10:
When I first did the random play option I think I imagined that the shuffling would only happen every five run throughs. The reason for that is because using the same sequence keeps the image repeat a full N apart. If the shuffle process is done every loop then there are bound to be some images that are shown quite soon after the previous showing.

The problem is mainly human psychology. We are just super sensitive at spotting patterns, even when they don't exist, as evidenced by our perceptions of the slide frequency. Simplest way to make it seem **more** random is to shuffle it **less** often!

jgodfrey, 2021-07-10:
@paddywwoof - yep, I agree with all of that. When starting any shuffled cycle, it's always possible for an image near the end of the previous cycle to now be near the start of the current cycle - making it display twice in a short time interval.

And, as you say, increasing the shuffle interval helps to make that situation occur with less frequency.

Regardless though, I think that showing the first image *twice* during a single cycle of the playlist (as happens now) is unintended and really is a bug that should probably fixed. Do you agree?

Related, now that we have a "last displayed" time stamp for each image, it should be possible to prevent the re-display of images in a short time interval - although I don't think it'd be worth the trouble.

sapnho, 2021-07-10:
> Related, now that we have a "last displayed" time stamp for each image, it should be possible to prevent the re-display of images in a short time interval - although I don't think it'd be worth the trouble.

That would be super neat. Admittedly, nice-to-have only but maybe there is a clever way to achieve this.. :-)

sapnho, 2021-07-10:
> The problem is mainly human psychology. We are just super sensitive at spotting patterns, even when they don't exist, as evidenced by our perceptions of the slide frequency. 

No doubt, it's all in the mind. 

> Simplest way to make it seem **more** random is to shuffle it **less** often!

@paddywwoof If the shuffling is set to more often, it repeats the same order, doesn't it? That may not be ideal, either, that's why I always set it to "1". Or did I misunderstand this parameter?

jgodfrey, 2021-07-10:
> @paddywwoof If the shuffling is set to more often, it repeats the same order, doesn't it? That may not be ideal, either, that's why I always set it to "1". Or did I misunderstand this parameter?

I'll answer for paddy (correct me if I missed something)...

Assuming `shuffle=True`, it'll run `reshuffle_num` complete iterations of the slideshow before reshuffling the image order. And, those iterations (up until the next shuffle) will run in the same order.

jgodfrey, 2021-07-10:
Created PR https://github.com/helgeerbe/picframe/pull/140 for issue https://github.com/helgeerbe/picframe/issues/139 and https://github.com/helgeerbe/picframe/issues/119.

I'll wait to merge in case anyone has input.

jgodfrey, 2021-07-13:
I'll likely merge this into `dev` sometime evening unless anyone plans to test it independently. I'm happy to wait if that's the case - just let me know.

closed 2021-07-14 by jgodfrey