helgeerbe/picframe #142: Mat error on Mac

helgeerbe, 2021-07-12:
Got an exception in viewer_display.py  line 290
````phyton
            if self.__mat_images and diff_aspect > self.__mat_images_tol:
                if not pics[1]:
                    im = self.__matter.mat_image((im,))
````
WARNING:viewer_display.ViewerDisplay:Can't create tex from file: "/Users/herbe/Pictures/Wallpapers/Berlin.jpg" or "None"
WARNING:viewer_display.ViewerDisplay:Cause: __array__() takes 1 positional argument but 2 were given

Going deeper I see that:

Exception in mat_image.py line 506.

````python
def run(self, image, start_clusters=None):
        image = image.copy()
        image.thumbnail(self.size)
        im = np.array(image, dtype=np.float)[:,:,:3]
````
'`np.float` is a deprecated alias for the builtin `float`. To silence this warning, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.\nDeprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations'

numpy version 1.21 is installed

paddywwoof, 2021-07-12:
@helgeerbe, thanks for finding that. We should change it to `np.array(image, dtype=float)` as suggested. Also on line 519.

There only seems to be one other place I've used `np.float` in pi3d (TextBlock) so patched that in develop branch for the moment.

PS I really hate the vagueness of C leaving the choice up to the operating system or compiler or whatever. In Rust you specify f32, f64, i32 etc. that way you always get what you expect!

paddywwoof, 2021-07-12:
pushed change here https://github.com/helgeerbe/picframe/commit/cb4d8c1cd36b6d674ace91e77544f095ac79a42f

helgeerbe, 2021-07-12:
Thanks @paddywwoof, but this doesn't solve the root cause.

` im = np.array(image, dtype=float)[:,:,:3]`still throwing an exception.

Image.py line 546 raises AttributeError.

````python
    def __getattr__(self, name):
        if name == "category":
            warnings.warn(
                "Image categories are deprecated and will be removed in Pillow 10 "
                "(2023-01-02). Use is_animated instead.",
                DeprecationWarning,
                stacklevel=2,
            )
            return self._category
        raise AttributeError(name)
````

paddywwoof, 2021-07-12:
That looks like an error from `pillow` what do you get if you open a python terminal and
```python
from PIL import Image
im = Image.open('/Users/herbe/Pictures/Wallpapers/Berlin.jpg')
```
my version of pillow is 8.3.1

helgeerbe, 2021-07-12:
Now it works. I can't tell what happened. I uninstalled PIL and installed the latest version again. Running from command line everything works. In visual studio code I had to select Python 3.9.6 which is located in /opt/homebrew/bin. Using python3.82 which is located at /usr/bin/phython3 can't find PIL.

So it seems to be maybe a problem with my  environment. Using libs and interpreter from /opt/homebrew/... works now.

helgeerbe, 2021-07-12:
Guess we can close this issue.

closed 2021-07-12 by helgeerbe

paddywwoof, 2021-07-12:
Sure. Well done for sorting it out. Python is still a bit of a minefield with different versions (quite apart from 2 v 3) existing in different places and different modules being installed in /home/user/.local as well as in the root of the system.