helgeerbe/picframe #143: AttributeError: 'NoneType' object has no attribute 'iy'

davidcmoody, 2021-07-12:
Trying to setup a digital picture frame.  I have 4 images in my Pictures directory, but I keep getting this odd message.

My Pictures directory is being loaded via cron job to rotate the pictures around, but the pictures are viewable after transferring to another computer no program?

INFO:start.py:starting ['/home/pi/picframe_data/run_start.py', '/home/pi/picframe_data/config/configuration.yaml']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
Traceback (most recent call last):
  File "/home/pi/picframe_data/run_start.py", line 2, in <module>
    start.main()
  File "/usr/local/lib/python3.7/dist-packages/picframe/start.py", line 157, in main
    c.loop()
  File "/usr/local/lib/python3.7/dist-packages/picframe/controller.py", line 304, in loop
    (loop_running, skip_image) = self.__viewer.slideshow_is_running(pics, time_delay, fade_time, self.__paused)
  File "/usr/local/lib/python3.7/dist-packages/picframe/viewer_display.py", line 474, in slideshow_is_running
    wh_rat = (self.__display.width * self.__sfg.iy) / (self.__display.height * self.__sfg.ix)
AttributeError: 'NoneType' object has no attribute 'iy'

jgodfrey, 2021-07-12:
Hi @davidcmoody - welcome!

I'm not sure I understand exactly what you're CRON job is doing so perhaps some more details there are warranted...

In the mean time, a few questions:

- What happens if you eliminate the CRON job from the mix and just manually place some images in your `Pictures` folder?
- What are the file types of the images in question?
- Do any of the images display before you get the above error?
- What version of picframe are you running?

jgodfrey, 2021-08-09:
@davidcmoody - is this still an issue? If so, do you have any additional information?

davidcmoody, 2021-08-11:
Hi Jeff,

I needed it for a project, so I created a way to do a simple slideshow that
I was looking for via feh.  I will try the project again later with your
code when I have some more time available.

Thanks,
-David

shedzong, 2021-08-18:
I had this issue too. It occurred when displaying full size pics (20MB) that had a caption present. It seemed to be fine without captions.

I've worked around it by resizing all pics to 1920. The captions are still there but not a problem at that pic size

Hope that's helpful

Peter

NateEaton, 2024-08-04:
It doesn't appear this has gotten any attention in a long time so I'm not sure whether it's a case of no one else having the issue or what. However, as I try to switch over from PictureFrame2020.py from pi3d_demos to Picframe, this issue has me completely blocked since my intent is to use the same form of edge blur I had with PictureFrame20202.py. 

Things were working great after the initial setup. However, once I set blur_edges=True, I get the errors below and a blank screen on the monitor. It doesn't appear to matter what other options are set to, for example, whether fit=True or False. The one exception is that if mat is not False, the error doesn't occur. 

`INFO:start.py:starting ['/home/pi/venv_picframe/bin/picframe']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
gl_id  b'GL' b'2.1 Mesa 23.2.1-1~bpo12+rpt3'
WARNING:viewer_display.ViewerDisplay:Can't create tex from file: "/home/pi/Pictures/Samples/IMG_3812.JPG" or "None"
WARNING:viewer_display.ViewerDisplay:Cause: argument 1 must be 2-item sequence, not generator
Traceback (most recent call last):
  File "/home/pi/venv_picframe/bin/picframe", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/start.py", line 136, in main
    c.loop()
  File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/controller.py", line 293, in loop
    (loop_running, skip_image) = self.__viewer.slideshow_is_running(pics, time_delay, fade_time, self.__paused)
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/viewer_display.py", line 487, in slideshow_is_running
    wh_rat = (self.__display.width * self.__sfg.iy) / (self.__display.height * self.__sfg.ix)
                                     ^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'iy'`

I'm attaching my configuration.yaml. 

Hopefully this provides enough information for this bug to get actively worked and resolved.

[configuration_yaml.txt](https://github.com/user-attachments/files/16485286/configuration_yaml.txt)

paddywwoof, 2024-08-04:
That missing `self.__sfg` and the conditions you describe should make this diagnosable. I will look at this later.

Paddy

paddywwoof, 2024-08-04:
Hi, I couldn't get your error after a quick check on my running picframe but when I `python -m pip install picframe --upgrade`ed it, there it was. It seems to be due to PIL tightening up some of its checking, and maybe trying to speed itself up. Previously I passed a generator where the `resize` function needed a list and it happened to work. Obviously a bit of slack coding on my part. I will sort out a proper update in the main branch, but in the meantime could you test that this fixes your problem?

In `/home/pi/venv_picframe/lib/python3.11/site-packages/picframe/viewer_display.py` change a couple of lines (317 and 322) so they have square brackets. i.e. `(int(...) for x in im.size)` becomes `[int(...) for x in im.size]` Like this

``` python
                    blr_sz = [int(x * 512 / size[0]) for x in size] ############################### this line ###############
                    im_b = im.resize(size, resample=0, box=box).resize(blr_sz)
                    im_b = im_b.filter(ImageFilter.GaussianBlur(self.__blur_amount))
                    im_b = im_b.resize(size, resample=Image.BICUBIC)
                    im_b.putalpha(round(255 * self.__edge_alpha)) 
                    im = im.resize([int(x * sc_f) for x in im.size], resample=Image.BICUBIC) ###### this line ##############

```
You will have to restart picframe for this to have any effect.

Paddy

PS @helgeerbe and @saphno I also noticed that I never pushed the `use_sdl2` fixes up to main so that might make a difference to the heat generation of the RPi5, if it's trying to do CPU emulation of the graphics processor, or such like. I will have a play with that later too.

sapnho, 2024-08-04:
> PS @helgeerbe and @saphno I also noticed that I never pushed the use_sdl2 fixes up to main so that might make a difference to the heat generation of the RPi5, if it's trying to do CPU emulation of the graphics processor, or such like. I will have a play with that later too.

Thanks, @paddywwoof It would be great to get the Pi 5 working with the same set of instructions.

NateEaton, 2024-08-04:
Thanks, @paddywwoof. I'll give that a try this evening when I'm home.

NateEaton, 2024-08-04:
@paddywwoof, home earlier than planned so just tested it and that fixed the problem. Thanks!

L0rz, 2024-08-13:
Same error from NatEaton here on a Pi3 b with a fresh installation of Bookworm.

Changing the Brackets also worked for me. THX

closed 2025-02-07 by helgeerbe