helgeerbe/picframe #147: Raspi Debian Desktop installation No Images shown as a Service

Dimasua, 2021-07-14:
Hej, thanks for your work, amazing what you have done! I'm trying to run the application on my Raspberry Pi 3+ with desktop version of Debian, however the images are shown only when I run the command and not in the service mode. I have followed all the steps besides SAMBA as I use Rclone to sync the images with my Google Photo.

helgeerbe, 2021-07-14:
Hi @Dimasua I'm assuming you are following the instructions from the [wiki](https://github.com/helgeerbe/picframe/wiki/Running%20As%20Service)?

What happens if you start the service from a console?
`systemctl --user start picframe.service`

What does the logging say?
`journalctl --user-unit picframe.service`

Dimasua, 2021-07-14:
Yes, that is correct, I'm following the wiki.
When I start the service from a console simply nothing happens, I still see my desktop default background image.

Please see the logs below. There is one error at the very bottom.
[image]

helgeerbe, 2021-07-14:
OK, at least the service seems to start. Can you set the log level in `configuration.yaml` to `DEBUG` and restart the the service.

`systemctl --user restart picframe.service`

Dimasua, 2021-07-14:
Done, please see the logs below.

[image]

helgeerbe, 2021-07-14:
In this log I see a mqtt error. Don't know why the images are not shown. Can you set mqtt and http to false. Just to be sure it is the display thread.

Does nothing happened and you still see your desktop or does the screen goes blank?

As a last try set use_glx to true. Does this change anything?

Do you have a link to the debian Version you used. I could set up a pi3 for further testing.

helgeerbe, 2021-07-14:
Are you starting the service through a ssh connection, or directly on a terminal on the raspi?

Dimasua, 2021-07-15:
So, I have disabled mqtt and http and it works!! :) Thank you for your support! Can you please share some documentation how to install the Home Assistant? I would like to try it as well.

helgeerbe, 2021-07-15:
Good to see, that it works now. If you enable http you will have a simple web interface to control the picframe.

As a good starting point I would recommend Wolfgang's blog [www.thedigitalpictureframe.com](https://www.thedigitalpictureframe.com/integrate-raspberry-pi-picture-frame-home-assistant/)

closed 2021-07-17 by helgeerbe