helgeerbe/picframe #148: Enhanced home assistant integration

helgeerbe, 2021-07-17:
With release 2021.7 home assistant mqtt device discovery offers (besides switches and sensors) two new types which could be used by picframe to avoid manual configuration.
- **number** for
  - fade time
  - time delay
  - brightness
- **select** for
  - directory selection

I put the mqtt changes in a separate branch [ha_mqtt_integration](https://github.com/helgeerbe/picframe/tree/ha_mqtt_integration). No changes on topics for commands to picframe.

So only input text helpers remain for text based filters
- tags
- location
- date to
- data from

To avoid further manual configurations in the automations. I created a blueprint.  The [gist](https://gist.github.com/helgeerbe/78f66ecf1f5206ce50ef9599892f202a) url can directly used in home assistant.

[image]

So most of the time. Only the date format string has to be entered to reflect the local date format. For the other settings, the default values could be used and need only to be changed, if you are using more than one picframe. They are linking picframes mqtt sensors to the input text helper ui elements.

helgeerbe, 2021-07-20:
I pulled #150 into dev. This is a breaking change for home assistant configuration. 

You have to uninstall the old integration.
1. Open mqtt integration and delete picframe device.
2. Delete all picframe automations
3. Delete all helpers **except** 
  - input_text.tags
  - input_text.location
  - input_text.date_to
  - input_text.date_from

To setup the new integration:
1. run picframe
2. wait for a pic change (this will send proper values to the ui elements) or press next image
3. Open ui cards and and replace the old (now) unreferenced elements with the newly created elements
4. press the above blueprint button to load the blueprint. That's it.

sapnho, 2021-08-21:
@helgeerbe I guess that means we should update your article: https://www.thedigitalpictureframe.com/integrate-raspberry-pi-picture-frame-home-assistant/ as well.

helgeerbe, 2021-08-21:
Yes. Sorry for that. But this makes the setup much easier. Also if you would like to run more than one picframe in home assistant.

closed 2021-09-20 by helgeerbe

sapnho, 2022-01-02:
Hi @helgeerbe, I was working on the rewrite of the https://www.thedigitalpictureframe.com/integrate-raspberry-pi-picture-frame-home-assistant article today. With your new blueprint, the integration should be a lot easier indeed.
However, I am struggling with the text entry for tags, dates, and locations where I cannot enter any values.
It looks like this:

[image]

My helpers are

[image]
Any idea where I might have taken a wrong turn? Thanks!

helgeerbe, 2022-01-04:
Hi @sapnho looks good for me. If I remember right, you will get the first controls, when picframe starts. But you have to wait at least for the second image. While changing image the attribute controls are send to picframe (you can speed this up, clicking on next image).

sapnho, 2022-02-06:
@helgeerbe I only now got around to updating the blog article but I still have the same issue that I cannot enter any text in location, tags, date from, or date to.
This is the draft update article: https://www.thedigitalpictureframe.com/?p=19080&preview=1&_ppp=cd318d49da
I guess I am doing something wrong with the Helpers but I have no idea what.

helgeerbe, 2022-02-07:
Hi @sapnho, what do you mean with: " I cannot enter any text....". On the home assistant webpage, you can't enter any text?

Check helpers and Folder-Card.
[image]

````YAML
entities:
  - entity: sensor.picframe_image_counter
    name: Bilder
  - entity: select.picframe_directory
    name: Directory
  - entity: input_text.picframe_tags_filter
    name: Stichwortsuche
  - entity: input_text.picframe_location_filter
    name: Ortssuche
  - entity: input_text.picframe_date_from
    name: Bilder ab (TT.MM.JJJJ)
  - entity: input_text.picframe_date_to
    name: Bilder bis (TT.MM.JJJJ)
show_header_toggle: false
theme: Backend-selected
title: Filter
type: entities
state_color: true
````

Entity Ids must be the same.

sapnho, 2022-02-07:
Perfect, thanks @helgeerbe.
I had "sensor" instead of "input_text" in my automatically created fields. Now it works!