helgeerbe/picframe #149: typo in code - should be 'utf-8'

junglebells, 2021-07-19:
https://github.com/helgeerbe/picframe/blob/b0fddbb9df1b771a9133dc3dd5ea92c18b9165df/picframe/get_image_meta.py#L40

helgeerbe, 2021-07-19:
@junglebells Thanks for finding that. But if I understand the definition correctly it should be 'utf_8' but 'utf8' and 'utf-8' are valid aliases.

Das that line makes any problems? Or does it work correctly in your environment?

junglebells, 2021-07-19:
I am not sure about captions, but my IPTC keywords (tags) are not coming through (they are there in pics, but not in the db)
I was trying to find where the problem could be and noticed the typo.

Can you confirm that if there are multiple keywords on a picture, it would work. Here is a sample picture, if you want to test.
[image]

jgodfrey, 2021-07-19:
Hi @junglebells,

I can confirm that multiple keywords load correctly into the db's `meta` table if that's what your asking. Also, I can confirm that the keywords in your above sample image do get stored the db as shown below:

[image]

Is that not the case for you?

junglebells, 2021-07-19:
No, unfortunately it’s not working for me. I don’t see the keywords in the db. 
I have the latest code base. Any tips on how to troubleshoot?
Thanks so much. 
Ashu

jgodfrey, 2021-07-19:
> I have the latest code base

What are you running, specifically? Source code (`main`, `dev`, ??) or a packaged release (what version)? There were some meta-data collection issues fixed in the past, but I think they should be widely available now...

junglebells, 2021-07-19:
pi3d-2.46 picframe-2021.7.5

on Raspberry Pi, installed via python3 pip

jgodfrey, 2021-07-19:
> pi3d-2.46 picframe-2021.7.5

Hmmm... Yeah, that should definitely have the _meta-data fixes_ I mentioned above. I'm not sure why it's not working for you. Is it failing on all images, or just some subset? Also, do the offending images have a `meta` table record with no `tags` data, or is there no `meta` table record for them at all?

Does the logging show anything interesting?

junglebells, 2021-07-19:
It's failing for all (except 1) images. These images do have a record in
the meta table, but no tags data.
The only difference in the image that's working and others is in the way
the tag/keyword data is stored. Working image has a bunch of keywords
stored as 1 string, while all other offending images have multiple
keywords. But I saw your screenshot - that one offending image worked at
your end.

I'd love to explore logs, but could not find where they are stored?

jgodfrey, 2021-07-19:
> I'd love to explore logs, but could not find where they are stored?

You'll probably need to:

- Stop the frame service
- Activate the logging
- Restart the service
- Check the logs

To activate the logging, you'll need to open the configuration file (`configuration.yaml`) and set the following two options:

- `log_level` (defaults to WARNING, which might be enough. For full logging, set it to DEBUG)
- `log_file` - set this to a fully qualified path of the log file you want to create (`/home/pi/picframe.log` for example)

Also, if you don't have much in your cache database at the moment, I might recommend that you just delete it (or rename it) so it gets recreated. That should force the the caching process to run on all images again, which will hopefully log something helpful should it fail.

junglebells, 2021-07-20:
I enabled the logs and also recreated the db (deleted the existing one). To my surprise the tags are showing up now!
Not sure what may have been wrong, but this is great... thanks so much for your help!

On a side note, I noticed in the db screenshot you sent earlier, you had people names. I am trying to convert face recognition tags to keywords (in Apple Photos.app) - any suggestion on what might be a good way to do it?

Thanks again for all the help

jgodfrey, 2021-07-20:
Hey, that's great news @junglebells. Glad things are working now (though I agree, it's a mystery why it didn't work before).

One thing I should have mentioned earlier... You'll want to be sure to undefine the log file name in the config file (just set it back to `""`). Otherwise, it'll continue to concatenate log info to your file and it'll get huge pretty quickly.

Regarding face recognition.... Unfortunately, I don't really have any particular advice here. The tags you saw in my earlier screenshot were added manually, although my DAM software can use Google API's for auto face recognition (but I've never used it).

Perhaps @helgeerbe or @paddywwoof  have some input?

helgeerbe, 2021-07-20:
@junglebells good to hear that it is working now. I don't use apple photo. But have a look at [https://discussions.apple.com/thread/8595733](https://discussions.apple.com/thread/8595733).

junglebells, 2021-07-21:
Thank you @jgodfrey @helgeerbe @paddywwoof - I consider this issue resolved. Thanks so much your help and a wonderful product.

closed 2021-07-21 by junglebells