helgeerbe/picframe #151: exif_datetime format

junglebells, 2021-07-21:
I wanted to bring another topic to your attention. The following MQTT command is not working for me: 
mosquitto_pub -h localhost -t picframe/date_from -m "2021-01-01"

Looking at the DB, I noticed the exif_datetime to be in a numeric format (see screenshot)
[image]

Is this considered normal? Any tips of debugging the issue?
Thanks again for all your help.

jgodfrey, 2021-07-21:
Hi @junglebells,

The dates in the database are stored in a Unix epoch time format, so what you show above is indeed normal. I don't have access to the code ATM so I can't offer much advice right now. I'll try take a look soon unless someone beats me to it.

paddywwoof, 2021-07-21:
It would be easy to fix it to use - as well as : and / https://github.com/helgeerbe/picframe/blob/b0fddbb9df1b771a9133dc3dd5ea92c18b9165df/picframe/controller.py#L10

junglebells, 2021-07-21:
Thanks @paddywwoof and @jgodfrey - I figured out the problem. It was an error at my end, I had a filter on, that I didn't realize.

Appreciate all your help and prompt responses.

closed 2021-07-21 by junglebells