helgeerbe/picframe #152: Exiting App via Touchscreen

bobcoup, 2021-07-28:
Hi

Liking the app! I run it on a Pi 3b using an Official Raspberry Pi 7" Touchscreen. Currently the only way I can exit the app and return to the desktop is either by pressing Ctrl + C or via the [Stop] button on the html menu.

Would it be possible to create a script that when the screen is touched would enable an "Exit to desktop" option?

Thanks.

paddywwoof, 2021-07-28:
Glad it's working OK. This has cropped up before, and it would be quite easy to change https://github.com/helgeerbe/picframe/blob/main/picframe/viewer_display.py to do what you want, just add a pi3d.Mouse instance, check for clicks in the loop function and return False if one happens. However the functionality should really be made to enable other things to happen such as going to the next/previous slide or showing/hiding the clock. I think the previous suggestion was for a long press to stop the show.

I will try to think of a more flexible way to do this - maybe something along the lines of the `interface_kbd.py` but for mouse would work.

helgeerbe, 2022-01-05:
There is a pending Pull Request #179 
@bobcoup are you able to test it?

bobcoup, 2022-01-13:
@helgeerbe Sorry for the delayed response. Does this still need testing?

helgeerbe, 2022-01-14:
Hi @bobcoup would be great, if you have time.

helgeerbe, 2022-01-20:
Will work in release 2022.01.22

closed 2022-01-20 by helgeerbe