helgeerbe/picframe #159: Adjust Black screen time between images?

Cinderhaze, 2021-08-19:
Hello!  I have been using your software to set up a pictureframe, and adjusting the time_delay and fade_time settings, I still get about a 5~10 second pause/black screen between images.  I'm assuming time_delay is the period between image starts.  

Is there any way to reduce the 'blank' time between images?  It almost acts as if every fade is done between an image and a black screen, and then a fade is done between the black screen and the next image - I'm not seeing the transition between two images.  

Thank you in advance for your help, I hope I'm not missing anything to obvious!

jgodfrey, 2021-08-19:
That should not be the case. Generally, one image should fade out while the next one fades in, with no "black screen" between them. A few things...

- What hardware are you running the software on?
- Maybe post your configuration file in case it has a problematic setting?

Cinderhaze, 2021-08-19:
I'll grab my configuration files when I head home from work.  It is currently running on a Raspberry Pi B+, with a freshly installed Rasbian Buster OS.

jgodfrey, 2021-08-19:
One more thing. How big are the images your running on the frame (both in file size and resolution)? If they're larger than necessary, that'll cause lots of extra work for the hardware. If you are using large images, does it perform better if you resize images  to be more in line with the resolution of your display?

Cinderhaze, 2021-08-19:
I'll double check image sizes - I first started out with three random XKCD comics to prove out that it worked, and later added in actual photographs.  I saw the same black/delay in between each transition

paddywwoof, 2021-08-19:
Definitely shouldn't have a blank between images. The OpenGL render shader has two texture2d samplers - one for the front and one for the back image and it should be running all the time (at whatever FPS the Pi can manage). So a blank period must mean that something is stopping the rendering completely. Very curious.

And, yes, the delay value is the time after an image is loaded and starts to show that triggers the next image to be loaded.

Also, [this tangential reference to Raspberry Pie](https://xkcd.com/2504/) [sic] appeared today!

Cinderhaze, 2021-08-20:
Here's my config file, comments removed

```
viewer:
  blur_amount: 12                         # default=12,  larger values than 12 will increase processing load quite a bit
  blur_zoom: 1.0                          # default=1.0, must be >= 1.0 which expands the backgorund to just fill the space around the image
  blur_edges: False                       # default=False, use blurred version of image to fill edges - will override FIT = False
  edge_alpha: 0.5                         # default=0.5, background colour at edge. 1.0 would show reflection of image
  fps: 20.0                               # default=20.0
  background: [0.2, 0.2, 0.3, 1.0]        # default=[0.2, 0.2, 0.3, 1.0], RGBA to fill edges when fitting
  blend_type: "blend"                     # default="blend", choices={"blend", "burn", "bump"}, type of blend the shader can do
  font_file: "/home/pi/picframe_data/data/fonts/NotoSans-Regular.ttf"
  shader: "/home/pi/picframe_data/data/shaders/blend_new"
  show_text_fm: ""               # default "%b %d, %Y", format to show date over the image
  show_text_tm: 20.0                      # default=20.0, time to show text over image with file name
  show_text_sz: 40                        # default=40, text character size
  show_text: ""  # default="title caption name date folder location", show text, include combination of words: title, caption name, date, location, folder
  text_justify: "L"                       # text justification L, C or R
  fit: True                               # default=False, True => scale image so all visible and leave 'gaps'
  kenburns: True                          # default=False, will set fit->False and blur_edges->False
  display_x: 0                            # offset from left of screen (can be negative)
  display_y: 0                            # offset from top of screen (can be negative)
  display_w: null                         # width of display surface (null->None will use max returned by hardware)
  display_h: null                         # height of display surface
  use_glx: False                          # default=False. Set to True on linux with xserver running

  mat_images: 0.01                        # default=0.01, True, automatically mat all images. False, don't automatically mat any images. Real value, auto-mat all images with aspect ratio difference > than value
  mat_type: null                          # default=null, A string containing the mat types to choose from when matting images. It can consist of any or
  outer_mat_color: null                   # default=null, Color of the outer mat as an RGB list. null will auto-select a reasonable color based on the image.
  inner_mat_color: null                   # default=null, Color of the inner mat as an RGB list. null will auto-select a reasonable color based on the image.
  outer_mat_border: 75                    # default=75, Minimum outer mat border in pixels
  inner_mat_border: 40                    # default=40, Minimum inner mat border in pixels (for styles that use it)
  outer_mat_use_texture: True             # default=True, True uses a texture for the outer mat. False creates a solid-color outer mat.
  inner_mat_use_texture: False            # default=False, True uses a texture for the inner mat. False creates a solid-color inner mat.
  mat_resource_folder: "/home/pi/picframe_data/data/mat" # Folder containing mat image files

  show_clock: False                       # default=False, True shows clock overlay. False does not show clock overlay
  clock_justify: "R"                      # default="R", clock justification L, C, or R
  clock_text_sz: 120                      # default=120, clock character size
  clock_format: "%I:%M"                   # default="%I:%M", strftime format for clock string

model:
  pic_dir: "/home/pi/Pictures"                   # default="/home/pi/Pictures", root folder for images
  deleted_pictures: "/home/pi/DeletedPictures"   # move deleted pictures here
  no_files_img: "/home/pi/picframe_data/data/no_pictures.jpg" # default="PictureFrame2020img.jpg", image to show if none selected
  subdirectory: ""                        # default="", subdir of pic_dir - can be changed by MQTT"
  recent_n: 0                             # default=7 (days), when shuffling file change date more recent than this number of days play before the rest
  reshuffle_num: 1                        # default=1, times through before reshuffling
  time_delay: 10.0                       # default=200.0, time between consecutive slide starts - can be changed by MQTT
  fade_time: 50.0                         # default=10.0, change time during which slides overlap - can be changed by MQTT"
  shuffle: True                           # default=True, shuffle on reloading image files - can be changed by MQTT"
  sort_cols: 'fname ASC'                  # default='fname ASC' can be any columns in the table with optional ASC or DESC separated by commas
  image_attr: [                           # image attributes send by MQTT, Keys are taken from exifread library, "PICFRAME GPS" is special to retrieve GPS lon/lat, "PICFRAME LOCATION" is special to retrieve geo reverse (load_geoloc hast to be True)
    "PICFRAME GPS",
    "PICFRAME LOCATION",
    "EXIF FNumber",
    "EXIF ExposureTime",
    "EXIF ISOSpeedRatings",
    "EXIF FocalLength",
    "EXIF DateTimeOriginal",
    "Image Model",
    "IPTC Caption/Abstract",
    "IPTC Object Name",
    "IPTC Keywords"]
[... cut ...]
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 00000000d7904c56
Model           : Raspberry Pi 3 Model B Rev 1.2
```
```
$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Cinderhaze, 2021-08-20:
https://user-images.githubusercontent.com/7217800/130170167-cecb64ea-46df-4c87-9411-d5e195af7b0a.mov

Video of the action - it looks like the fade in starts very slowly, and then suddenly jumps up to most of the brightness, and then when it fades it does the same thing (quick fade from full to almost dark, then long gradual fade to pitch black)

jgodfrey, 2021-08-20:
@Cinderhaze - yeah, that video shows the problem well, and that's certainly not expected.

I'm curious if turning off the `kenburns` effect changes anything? To do that, just set:

`kenburns: False`

... in the config file and restart the software.

Cinderhaze, 2021-08-20:
Same behavior.. slow gradual fade in that jumps to full brightness :/   Are there any other things that should be done, setting graphics memory, or overlay, or something?  I'm running this with the full desktop loaded

helgeerbe, 2021-08-20:
@Cinderhaze have a look at Wolfgang's [blog](https://www.thedigitalpictureframe.com/how-to-add-crossfading-slide-transitions-to-your-digital-picture-frame-using-pi3d/). Just to be sure, did you following his `raspi-config`settings regarding memory split and gl driver?

Cinderhaze, 2021-08-20:
I went back through and re-ran the commands (this is technically a pi3), but when I tried to configure them as Wolfgang said, I always get 'The GL driver is disabled.' when I try to select the legacy driver... but if I select any other driver, I get the following error when I try to run picframe
```
  File "/usr/local/bin/picframe", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/picframe/start.py", line 136, in main
    c.start()
  File "/usr/local/lib/python3.7/dist-packages/picframe/controller.py", line 312, in start
    self.__viewer.slideshow_start()
  File "/usr/local/lib/python3.7/dist-packages/picframe/viewer_display.py", line 420, in slideshow_start
    display_config=pi3d.DISPLAY_CONFIG_HIDE_CURSOR, background=self.__background, use_glx=self.__use_glx)
  File "/usr/local/lib/python3.7/dist-packages/pi3d/Display.py", line 570, in create
    display_config=display_config, window_title=window_title, use_glx=use_glx)
  File "/usr/local/lib/python3.7/dist-packages/pi3d/util/DisplayOpenGL.py", line 119, in create_display
    assert self.context != EGL_NO_CONTEXT and self.context is not None
AttributeError: 'DisplayOpenGL' object has no attribute 'context'
```

paddywwoof, 2021-08-20:
Hi, I think you want the GL driver to be disabled. It might work on the older Raspberry Pi models but it's excruciatingly slow if it does. So pi3d will work on the basis that you are using the original OpenGLES broadcom drivers, and generate some kind of error as you found.

In `raspi-config` -> 4 Performance Options -> P2 GPU Memory, what value to you have set?

Also, could you let me know what the dimensions are of the images in the slideshow.

Paddy

paddywwoof, 2021-08-20:
On the RPi here in the kitchen running picframe, if I turn the time delay below about 10s it starts to be limited by file loading and resizing work - and none of these picture are too big. I seem to remember that the fade time has a minimum positive value to avoid division by zero style errors.

However I've not managed to generate your blank screen effect :(

Cinderhaze, 2021-08-20:
I initially observed this behavior using 72px x 72px images (with the delay).  I'm currently seeing it (with the video clip above) with a variety of images... Here's a portion of the output

```
$ exif ~/Pictures/Customer\ pictures/*/*.jpg | grep Pixel
Pixel X Dimension   |4000
Pixel Y Dimension   |3000
Pixel X Dimension   |4000
Pixel Y Dimension   |3000
Pixel X Dimension   |4000
Pixel Y Dimension   |3000
Pixel Y Dimension   |2336
Pixel X Dimension   |4160
Pixel Y Dimension   |2336
Pixel X Dimension   |4160
Pixel Y Dimension   |2336
Pixel X Dimension   |4160
Pixel Y Dimension   |2336
Pixel X Dimension   |4160
Pixel Y Dimension   |2336
Pixel X Dimension   |4160
Pixel Y Dimension   |2336
Pixel X Dimension   |4160
Pixel X Dimension   |4032
Pixel Y Dimension   |3024
```

Cinderhaze, 2021-08-20:
GPU memory is set to 128 from running Wolfgang's earlier commands

Cinderhaze, 2021-08-20:
Hah, I needed to toggle use_glx to true, as in my case the X server is being used.. That appears to have solved my issue!  Sorry for all the confusion!

closed 2021-08-20 by Cinderhaze

paddywwoof, 2021-08-20:
Daryl, I'm obviously pleased you've sorted this out... but I'm not really sure why that fixed it. I've used the RPi 3 with X running at the same time as pi3d and it hasn't caused that odd black screen behaviour in the past.

Happy to close the issue but I will chalk this up as something to watch out for in case it's caused by some change in raspbian or the mesa GL driver.

If anything odd crops up let us know.