helgeerbe/picframe #160: Feature request: custom massage

wormiedk, 2021-08-20:
It would be great to be avle to display a custom massage from homeassistant via mqtt. Further showing select sensors from homeassistant such as room temperature, currently playing music and so on

helgeerbe, 2021-08-20:
I like that idea to send a simple custom string to picframe via MQTT. We could use the same setup like the clock string?

But for more complex layouts, I'm still not convinced this should be part of picframe. For that I'm using [picframe-overlay](https://github.com/helgeerbe/picframe-overlay). Using html + css you could render the most complex overlays.

wormiedk, 2021-08-21:
Sound like a plan. MQTT messages would be great for status messages but could potentially also be used for simple measurements such as room temp or similar

wormiedk, 2021-09-06:
I have problems getting picframe-overlay to work on my raspberry pi as picframe stops working as soon as I enable the fake gl driver. Is there a way to overlay something else e.g. conky on top of the picframe?

helgeerbe, 2021-09-07:
What kind of Pi do you have? I'm using picframe on a pi 3B+ with G1 legacy driver. But on Pi4 the recommendation for picframe is G2 GL Fake KMS.

If you boot to a graphical desktop, you could put any program on top of picframe. But the problem is, how to control it. picframe-overlay based on electron is simply spoken just a full screen transparent chrome browser, which could remotely be controlled. So you would be able to load any webpage you like. In default, it shows an empty (transparent background) page, so picframe is visible.

wormiedk, 2021-09-07:
I have a raspberry pi zero w running the picframe. The picframe seems to be running "above" the desktop. It is for example not visible when connecting using x11vnc and everything I run runs "below" it. Maybe it is because it is launched using systemd according to the guide here https://www.thedigitalpictureframe.com/how-to-add-crossfading-slide-transitions-to-your-digital-picture-frame-using-pi3d/#Automatically_start_the_Pi3D_slideshow_at_boot

Create a new service file

`mkdir ~/.config/systemd/user/ -p && nano ~/.config/systemd/user/picframe.service`
and paste the following text into the file

```
[Unit]
Description=PictureFrame on Pi3

[Service]
ExecStart=/home/pi/.local/bin/picframe ~/picframe_data/config/configuration.yaml
#Restart=always

[Install]
WantedBy=default.target
```

helgeerbe, 2021-09-07:
It's been a while since I set it up.

On raspi you need to run a composite manager to get a transparent window. By default it's xcompmgr. Unfortuantely activation of xcompmgr over raspi-config is not working. Instead append @xcompmgr in file /etc/xdg/lxsession/LXDE-pi/autostart. This will start xcompmgrwhen user piis logged in to the desktop.

You have to enable fake KMS GL driver in raspi-config

There is no default hook to run the services at login, so you must trigger it yourself. I do it from my ~/.config/autostart folder. If it does not exist. Create the folder autostart.  

My ~/.config/autostart/xsession_tartget.desktop

````
[Desktop Entry]
Type=Application
Name=xsession.target
Exec=/bin/systemctl --user start xsession.target
````

My ~/.config/systemd/user/xsession.target file
````
[Unit]
Description=Xsession running
BindsTo=graphical-session.target
````

My ~/.config/systemd/user/picframe.service
````
[Unit]
Description=Pi3D on Pi3

[Service]
ExecStart=/home/pi/.local/bin/picframe ~/picframe_data/config/configuration.yaml
Restart=always

[Install]
WantedBy=xsession.target
````

This means that picframe ist only started, if user is logged in to the graphical session!

My ~/.config/systemd/user/picframe-overlay
````
[Unit]
Description=overlay for picframe

[Service]
WorkingDirectory=/home/pi/dev/picframe-overlay
ExecStart=/usr/bin/npm start
Restart=always

[Install]
WantedBy=xsession.target
```

Don't forget to enable your services picframe and picframe-overlay

paddywwoof, 2021-09-07:
I think the pi zero will struggle to run a desktop. On the earlier pi versions pi3d uses the dispmanx broadcom display driver, which does have layers - pi3d defaults to 128 (top) and x defaults to middle (0). You can make pi3d display on other layers but then you have the issue of making the x desktop and window transparent. Not possible in my experience. So you probably have to display the picframe in an x window underneath a window with transparency. (i.e. as Helge did) But you will need to use the fake GL driver and I'm pretty sure that doesn't work on < pi 3.

So, TLDR; It would be relatively easy to show text as per the time or picture name boxes. The main issues are 1) A system for updating the text and/or switching with time display 2) I haven't got round to doing anything yet!