helgeerbe/picframe #161: Screen going black

sapnho, 2021-08-25:
Didn't we have a thread here where the user described his screen going black? Well, I had it now for the second time in a few days and I have no idea what could cause this. It's not connected with a `vcgencmd display_power` command, I suspect it comes from Pi3D.

I had it first with the 5.5 version. I then upgraded to 7.5 but it happened again. Very strange. In all these years, there was never a similar issue. Any files I should check? Maybe an SD corruption issue?

helgeerbe, 2021-08-25:
@sapnho  Have a look at #159 It's closed, but I guess not solved. Workaround is to set `use_glx=true`

sapnho, 2021-08-25:
Hmm, it doesn't go black between the images. It goes black and stays black. Very rarely, but it happened twice now.

helgeerbe, 2021-08-25:
Hm, I have this issue, if my picframe is loosing its wifi connection. So it becomes simply unresponsive to mqtt requests from home assistant.

Do you see the current image in home assistant? Or can you access your picframe through the web interface?

sapnho, 2021-08-25:
Yes, in Homes Assistant the image is shown and the playback sequence continues. It just isn't shown on the screen.

helgeerbe, 2021-08-25:
If you shut down the PricframeViewer (`systemctl stop ...`). Does the screen appears again? Just to be sure, pi3d is causing that.

sapnho, 2021-08-25:
I will have to wait until it happens again and will try then.

jgodfrey, 2021-08-25:
Probably not related, but I had some issues with one my frames randomly going black soon after I built it. In that case, it was a bad HDMI cable...

sapnho, 2021-08-25:
Yes, it's probably one of these things that you don't think of at all.. 

In my case, probably not the HDMI cable as after rebooting everything works fine again, but who knows?
I was suspecting the SD card or even the power supply.

Is there any way at all that Pi3D could cause this from the programming logic?

helgeerbe, 2021-11-23:
@sapnho Is this solved?

sapnho, 2021-11-23:
Yes, I think it was a hardware issue. I mirrored the SD image to a new SD card and the problem has never appeared again.

helgeerbe, 2021-11-23:
OK, so I close this issue.

closed 2021-11-23 by helgeerbe