helgeerbe/picframe #166: Support for selecting multiple subdirectories to display

greensum, 2021-09-04:
First of all kudos to all those involved in this excellent project.  It has basically all the features that I want and some that really wow me.
I have all my pictures stored in my NAS and samba-mounted under `/mnt/photo` beneath which are subdirectories for each calendar year.  Now if I want to display multiple years of pictures, currently there is no simple way to do it.  Letting it see all the pictures and then use date filter is one possible way but I'll miss those pictures that are missing EXIF.  I could set `~/Pictures` as the picture directory in the configuration and then create several symlinks to point to those subdirectories under `/mnt/photo` that I want to display.  E.g.:
`~/Pictures/2019 -> /mnt/photo/2019`
`~/Pictures/2020 -> /mnt/photo/2020`
`~/Pictures/2021 -> /mnt/photo/2021`
But I found it wouldn't work unless I add the `followlinks=True` option to the `os.walk` call in the `__get_modified_folders` def in `image_cache.py` as follows:
```python
    def __get_modified_folders(self):
        out_of_date_folders = []
        sql_select = "SELECT * FROM folder WHERE name = ?"
        for dir in [d[0] for d in os.walk(self.__picture_dir, followlinks=True)]:
```
I know enhancing the selection subdirectory to support multiple ones is not straight forward, but in the meanwhile I think there is no reason that we don't want to follow links in the `os.walk` call, right?  I'd appreciate very much if this could be changed to support my use case, and presumably many other people's as well.
Thanks!

paddywwoof, 2021-09-05:
Hi @greensum, thanks for this. Something to cope with multiple not-necessarily-related-folders would be good, and I had wondered about this before. I wondered if it would be possible to change the SQL to allow a `LIKE` style lookup or else make it cope with parsing a list so `IN` would work. The client end would have to do something a bit cleverer as well

helgeerbe, 2021-09-07:
Be aware that setting followlinks to True can lead to infinite recursion if a link points to a parent directory of itself. walk() does not keep track of the directories it visited already.

So maybe this should be a config setting for users knowing what they do?

os.walk is also used in model.py `def get_directory_list(self)`

jgodfrey, 2021-09-07:
> Something to cope with multiple not-necessarily-related-folders would be good

Some sort of a tagging system would be really useful here. So, define one or more text-based tags per folder and then build a playlist based on specified tag matches. I guess the hardest part would be in providing a mechanism to define the tags...

helgeerbe, 2021-09-08:
I wrote a patch.

````YAML
model:
  follow_links: False                     # default=False, By default, picframe will not walk down into symbolic links that 
````

So you can configure, if picframe should follow symbolic links.

Have in mind, that the cache is not cleaned up, if you switch back to to `follow_links=False`. So picframe will continue to show the already cached images in the symlink directories.
This, and that you can infinite loops let me decide to set the default to `False`. It's more an expert flag. You should know what you are doing.

Commit: https://github.com/helgeerbe/picframe/commit/f4f7ee295e0754c3cf3dbae0a702af0b7b21805a

greensum, 2021-09-12:
I’ve tested the change and it’s working for me. Thanks for putting in the change so promptly.

helgeerbe, 2021-09-22:
Sybolic links will will work since release 2021.09.22. But the rest might be an enhancement.

baltusf, 2021-11-24:
I know this is nog the right way, But can you contact me? I have a question for you.. contact@fredbaltus.nl . Sorry that i miss use this function

Ascendin81, 2022-01-14:
Hi,

So, I have my NAS mounted via SMB.  I can access the files on '/mnt/nas/Photos'

There is many subdirectories under this Photos folder...
2010-01
2010-02
2010-03
and so forth...

I have the correct location in configuration.yaml as "/mnt/nas/Photos"
but, when I run the run_start.py, it keeps sending me to the blue coffee cup screen, with no photos found.

How do I fix this?

Thank you,

Ascendin81, 2022-01-14:
UPDATE: Never mind, it works, there are over 100K photos, it took about 10mins start the slideshow. NICE!

helgeerbe, 2022-01-14:
Could be an access rights issue? On a pi3 I assume, that you run picframe as user `pi` on a pi4 I assume that picframe is running as user `root`?

Ascendin81, 2023-11-30:
I just updated to the 2023.07.26.post1 version.  It is no longer looking into the subdirectories.

I have "/mnt/nas/Photos" as the path, and it is running under user, not root.

Fresh install of picframe, new config, just changed the path.  Should I leave subdirectory: as ""?

I can go to /mnt/nas/Photos easily, and list everything...

Thank you.

commit f4f7ee29 (2021-09-08): Fix #166 Make it configurable, that picframe
follows symbolic links.