helgeerbe/picframe #167: picframe does not survive short break in image availability

veepee78, 2021-09-16:
Thanks for awesome image viewer!

I noticed that if image directory is not available for few minutes, they are marked as 'missing' and wont come back. I was able to change manually the db 'missing' field back to 0 so it works again.
I have mounted network share as source for the images.  Sometimes image host requires a boot and that is enough to break the database. Would it be possible to have some sort of mechanism to allow short breaks to image availability? Or maybe crawler could figure out that images are back online?

paddywwoof, 2021-09-16:
I thought it would re-find missing files on reconstruction. Vaguely remember that being the intention...

jgodfrey, 2021-09-16:
@veepee78  - welcome!

Hmmm... I guess I would have expected this to work now. The folders will certainly be marked as "missing" when the image host is rebooted (as in your above scenario), but I'd expect them to be picked back up after they become available again. If that's not the case (and, apparently it's not), something seems broken.

I'll try to take a look sometime soon.

veepee78, 2021-09-17:
This has happened couple of times now and first time I waited like 2 days, second time was just before I created this report. It showed default image and it did refresh the image according to schedule, because I saw "data" as directory on the screen everytime it loaded image again. I have around 40k images in the database.

Thank you again.

jgodfrey, 2021-09-18:
So, looking at the code, I see the problem. Once a folder goes missing, it's marked that way in the db (`missing = 1` in the `folder` table). Any missing folder will be rediscovered almost immediately upon becoming available again, but the folder's `missing` status isn't set back to `0` unless something has changed inside it (so, a file has been added, deleted, or modified).

In the described scenario (a mount going offline for a period of time and then coming back), none of those things will have happened, so the folder's missing status is never updated in the db.

It should be relatively easy to fix. I'll look into it.

jgodfrey, 2021-09-18:
Just committed a fix in PR https://github.com/helgeerbe/picframe/pull/168. I'll wait a bit before merging in case anyone wants to have a look.

jgodfrey, 2021-09-19:
Just merged the mentioned PR. This should now be fixed in `dev`.

closed 2021-09-22 by helgeerbe