helgeerbe/picframe #169: picframe db issue after upgrade to latest release

bobcoup, 2021-09-20:
Hi

Since upgrading to 2021.08.30 the following error occurs whenever picframe is started:

picframe.service - Start PictureFrame on Pi
   Loaded: loaded (/etc/systemd/system/picframe.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-09-20 07:58:35 BST; 17s ago
  Process: 1016 ExecStart=/home/pi/.local/bin/picframe ~/picframe_data/config/configuration.yaml (code=exited, status=1/
 Main PID: 1016 (code=exited, status=1/FAILURE)

Sep 20 07:58:35 piframe picframe[1016]:     m = model.Model(args.configfile)
Sep 20 07:58:35 piframe picframe[1016]:   File "/home/pi/.local/lib/python3.7/site-packages/picframe/model.py", line 175
Sep 20 07:58:35 piframe picframe[1016]:     model_config['portrait_pairs'])
Sep 20 07:58:35 piframe picframe[1016]:   File "/home/pi/.local/lib/python3.7/site-packages/picframe/image_cache.py", li
Sep 20 07:58:35 piframe picframe[1016]:     self.__update_schema(3)
Sep 20 07:58:35 piframe picframe[1016]:   File "/home/pi/.local/lib/python3.7/site-packages/picframe/image_cache.py", li
Sep 20 07:58:35 piframe picframe[1016]:     self.__db.execute("ALTER TABLE folder ADD COLUMN missing INTEGER DEFAULT 0 N
Sep 20 07:58:35 piframe picframe[1016]: sqlite3.OperationalError: duplicate column name: missing
Sep 20 07:58:35 piframe systemd[1]: picframe.service: Main process exited, code=exited, status=1/FAILURE
Sep 20 07:58:35 piframe systemd[1]: picframe.service: Failed with result 'exit-code'.

Deleting the picframe db enables the service to start successfully, but any restart of the picframe service will result in the same error.

Any ideas how to resolve?

Thanks.

jgodfrey, 2021-09-20:
Hmmm... That's in the db's auto-migration code. What's the value of the `schema_version` field in the `db_info` table?

bobcoup, 2021-09-20:
Hi

The schema version is 'NULL'.

jgodfrey, 2021-09-20:
A `null` field in the `schema_version` field is certainly unexpected. The current schema version is `3`. If you look at the `file` table, does it contain both a `displayed_count` and a `last_displayed` field? If it does, you can probably fix your immediate issue by simply setting the `schema_version` field to `3` manually.

That said, it doesn't explain how we end up with the NULL value. That's unexpected in all circumstances.

bobcoup, 2021-09-20:
Thanks for the help. Manually putting a '3' in the schema field has resolved my issue.

helgeerbe, 2021-11-23:
@jgodfrey Are you still investigating? Or should we close this issue?

jgodfrey, 2021-11-23:
> @jgodfrey Are you still investigating? Or should we close this issue?

No, I'm not. I was never able to reproduce this and nothing in the code jumps out at me as being wrong. I'll close it and we can look again if it's reported in the future.

closed 2021-11-23 by jgodfrey