helgeerbe/picframe #170: Screen going black after removing and adding images

sapnho, 2021-09-20:
This "screen going black" seems to have several variations.

Yesterday, I removed about 10 images and added 40 shortly afterward. The screen went black in the process and Pi3D even froze as in Home Assistant the images stopped changing. 

While the other "screen going black" issue was with Pi3D apparently still working as the images in Home Assistant kept changing.

Anything I could provide to help filter out a possible bug?

> picframe version:  2021.07.05
Checking required packages......
PIL :  5.4.1
exifread :  2.3.2
pi3d :  2.46
yaml :  5.4.1
paho.mqtt :  1.5.1
iptcinfo3 :  2.1.4
numpy :  1.16.2
ninepatch : installed, but no version info

sapnho, 2021-09-20:
Not sure if that has anything to do with it, but I have 1,527 image records in the database but only 1,523 photos in the Pictures directory.

sapnho, 2021-09-22:
I deleted the database and now the numbers are the same.

helgeerbe, 2022-01-05:
Hi @sapnho is there anything to do, or can I close this issue?

sapnho, 2022-01-05:
Yes, you can close this issue. After changing the SD card, the problem went away.

closed 2022-01-05 by helgeerbe