helgeerbe/picframe #173: Problems installing HEIC/HEIF support

jeffalperin, 2021-09-23:
I tried to install HEIC/HEIF support using the instructions [here](https://www.thedigitalpictureframe.com/the-heic-photo-format-now-works-with-the-pi3d-digital-picture-frame-image-viewer/), but after running for about 15 minutes on a rPi 4 (8gb) ended up with the message `fatal: unable to access 'https://github.com/libffi/libffi.git/' Could not resolve host: github.com`  (Strange error because I don’t have any problem with that link when I enter it into a browser.  )

Only the `libde265 ` and `libheif `directories were created (both in `/home/pi/picframe_data/config`).  

I realized that I may have run the command from the wrong directory, so I changed to `/home/pi` and tried again.  I got a very similar error after 15 minutes or so, but this time, only the `libde265 ` directory was created:
`fatal: unable to access 'https://github.com/strukturag/libheif.git/': Could not resolve host: github.com`

Do you have any idea what may be going on?  Or how to “uninstall” what I’ve already installed?  (Can I just remove the new directories?)

Many thanks for any suggestions!

helgeerbe, 2021-09-23:
I can't tell, what went wrong. But you can try to complete the rest step by step.
start with ``git clone https://github.com/strukturag/libheif.git ``

jeffalperin, 2021-09-23:
Step by step seemed to work!  Now when I run `picframe -v`, it shows pyheif 0.5.1 as an installed optional package.

My successful installation is in `/home/pi`.  Earlier, I had an aborted install in `/home/pi/picframe_data/config`, so there are `libde265 ` and `libheif ` subdirectories there.  To clean up, can I just remove those directories or is there something else I need to do?

helgeerbe, 2021-09-23:
Good to hear. I think you can can delete everything under /home/pi related to your `libheif` istalltions. Those are only source code folders which were used to build the required libs. After installation they should be copied under /usr/lib or should exist in the python site packages.

jeffalperin, 2021-09-28:
All seems good, so I'm closing this issue.  Many thanks!

closed 2021-09-28 by jeffalperin