helgeerbe/picframe #174: pictures from deleted_pictures being displayed

junglebells, 2021-09-28:
Picframe is displaying pictures from the deleted pictures folder - is this considered normal?

I was expecting anything in the deleted_pictures folder to be excluded. Please let me know if there is anything I can provide to help debug.

Here's some settings in configuration.yaml

```
pic_dir: "/home/pi/pictures"   
deleted_pictures: "/home/pi/pictures/deleted"
follow_links: False
subdirectory: ""
```

```
picframe version:  2021.09.22

Checking required packages......
PIL :  5.4.1
exifread :  2.3.2
pi3d :  2.47
yaml :  5.4.1
paho.mqtt :  1.5.1
iptcinfo3 :  2.1.4
numpy :  1.16.2
ninepatch : installed, but no version info

Checking optional packages......
pyheif : Not found!
```

paddywwoof, 2021-09-28:
Picframe builds a database by walking down the directory structure from pic_dir. deleted_pictures is a location expected to be outside pic_dir to move pictures to (rather than actually deleting them, like the wastebin folder of desktops). Obviously we could put some complicated filtering in but you would need to make a convincing case for using the setup you have.

junglebells, 2021-09-28:
Thanks @paddywwoof - there is no strong case here :)
This setup was done for convenience - it is equally easy for me to move the deleted_pictures location outside the pic_dir - will do that!
Thanks for your quick response and a wonderful product!

closed 2021-09-28 by junglebells