helgeerbe/picframe #176: mqtt directory list shows sym links, even if follow_links=false

helgeerbe, 2021-10-04:
Fixed with https://github.com/helgeerbe/picframe/commit/a10e7f8d66129f540e529d0e9d501c93ddd2e3f8

helgeerbe, 2021-10-05:
> Fixed with https://github.com/helgeerbe/picframe/commit/a10e7f8d66129f540e529d0e9d501c93ddd2e3f8

Guess there are parenthesis missing in the Boolean operation. Have to check this.

boomesito, 2021-10-05:
@helgeerbe Checked this im my test setup from #175 and it worked.

helgeerbe, 2021-10-05:
Hm, I have some issues. Will post it on #175

helgeerbe, 2021-10-05:
OK, found something.

`usb`is a mount point, and there is no stick attached. Testing this not accessible directory is consuming the time. At the end `os.path.islink()` throws internally an OSError exception.

But I don't know how to test for such an invalid mount point without waiting too long. Any ideas?

boomesito, 2021-10-05:
Maybe you could test for mount point first: `os.path.ismount()`

helgeerbe, 2021-10-05:
> Maybe you could test for mount point first: `os.path.ismount()`

I already tried that. But result is the same. If there is no mount (USB stick) on the mount point, call to this function waits for an internal exception.

I removed my automount and the mount point. So everything works fine. If I want to use automount again, I have to set ` `follow_links=True`. Setting this flag avoids testing of directories.

boomesito, 2021-10-05:
I'm not shure if I understand you correct. When I create a symlink from `Pictures\usb` to `/mount/pi/<usb-stick>` and then remove the USB stick `os.walk()` does not list this symlink even if follow_links is True

helgeerbe, 2021-10-05:
No. My usb directory is not a symlink. See https://github.com/helgeerbe/picframe/wiki/FAQ%20:%20User#automount-usb-stick

helgeerbe, 2021-10-06:
Should be fixed with release 2021.10.06.

closed 2021-10-06 by helgeerbe