helgeerbe/picframe #185: Is the size of darkened background for show_text customizable?

junglebells, 2021-11-07:
When show_text is used to display title, date, location, etc., this text is overlayed on a slightly darkened gradient background.

The size of this darkened band remains same irrespective of the font size used. In my case I am using a small font size (25 or 30). As a result the darkened background covers a lot of area on the image, and is quite distinctly visible with lighter images.

Is it possible to adjust the height of this background? It would be ideal to have a configuration option, or better yet, have it self adjust depending upon the font size/ length of the text string.

jgodfrey, 2021-11-07:
Currently, the scrim texture (the gradient) is just statically generated as a percentage of the screen height (25% ATM I think). It'd be fairly easy to make that percentage configurable, but it still would not be dynamic based on a given image. It might be interesting to have it dynamically generated for each image, based on the current text requirements (font size, number of lines, ...). However, that could cause visual differences in the density of the scrim layer between successive images, which might not be desirable...

junglebells, 2021-11-07:
Awesome, having the scrim texture height configurable would be a great start.

Thanks @jgodfrey

jgodfrey, 2021-11-10:
I'm traveling this week but I'll make an update for this this when I return.

junglebells, 2021-11-10:
Sounds good Jeff. Safe travels.

jgodfrey, 2021-11-15:
Just committed https://github.com/helgeerbe/picframe/pull/189/commits/a8f0f82ada2deb268588d3cd85ddb2ae6a41f0a7 to `dev` for this issue, which adds a new `text_bkg_hgt` configuration property. It represents a _percentage of screen height_ for the scrim texture. It defaults to `0.25` to match the original behavior, but can be set to anything in the range of `0 < text_bkg_hgt <= 1`.

jgodfrey, 2021-11-15:
Also, for clarity, the background texture has the following properties:

- Width: screen width
- Height: screen height * `text_bkg_hgt`
- Position: Bottom of screen
- Color: Solid Black (0,0,0)
- Alpha, bottom edge: 120 (~53% transparent)
- Alpha, top edge: 0 (100% transparent)
- Alpha, middle: evenly distributed between top and bottom values

junglebells, 2021-11-15:
> Just committed a8f0f82 to dev for this issue, which adds a new text_bkg_hgt configuration property. It represents a percentage of screen height for the scrim texture. It defaults to 0.25 to match the original behavior, but can be set to anything in the range of 0 < text_bkg_hgt <= 1.

Great, thanks Jeff. I’ll check it out and let you know.

junglebells, 2021-11-15:
Thanks Jeff, this is helpful.

jgodfrey, 2022-03-24:
This was fixed last November. Seeing no more discussion, closing this issue.

closed 2022-03-24 by jgodfrey