helgeerbe/picframe #190: Issue mounting NAS share ?

gn3481, 2021-11-15:
I have been using the PictureFrame2020 version for a while and recently saw Wolfgang's update with the newest version and decided to give it a try.

It's installed on the same PI I use for 2020, but I have been unable to get the new version to recognize my picture directory as "/mnt/NAS-PICS/pics" - the same values I use for 2020. Within the "pics" sub-directory are more sub-directories holding the individual photos, organized by various categories like vacations, trips, family gatherings, etc.

However, I keep getting the "no pictures display".

Using 2020 version or file manager, I am able to display the pictures from the NAS, so I can't believe it's a network or permissions issue.

Obviously, I'm doing something wrong but can't imagine what...
Appreciate any guidance or suggestions you can provide.

Thanks,
George

gn3481, 2021-11-15:
typo - should have been "/mnt/NAS_PICS/pics"

helgeerbe, 2021-11-15:
Hi @gn3481 does your typo comment means, that this issue is solved?

gn3481, 2021-11-15:
No. Just correcting typo in initial post.
Problem still exists.
Thanks,
George

helgeerbe, 2021-11-15:
Hm I could think of a couple of reasons. Do you see something in the log (Debug Level), if you run picframe from the console?
1. File permissions. Are you accessing the dir as user `pi` or `root`?
2. Is `/mnt/NAS_PICS/pics` a symbolic link? In this case, you have to set `follow_links: True`

gn3481, 2021-11-15:
Have not checked log...

Follow links set to true

gn3481, 2021-11-15:
Accessing as pi, not root
Dir permissions allow any read

gn3481, 2021-11-21:
**Here is debug output...Control-C pressed once "No Images Found" display appears.**

pi@piframe:~ $ picframe ~/picframe_data/config/configuration.yaml
INFO:start.py:starting ['/home/pi/.local/bin/picframe', '/home/pi/picframe_data/config/configuration.yaml']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
DEBUG:image_cache.ImageCache:Creating an instance of ImageCache
DEBUG:image_cache.ImageCache:Updating cache
INFO:controller.Controller:creating an instance of Controller
DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
DEBUG:image_cache.ImageCache:Found 0 new files on disk
DEBUG:pi3d.Display:

  Pi3D module - version 2.49

  Copyright (c) Tim Skillman, 2012-2021
  Copyright (c) Patrick Gaunt, 2012-2021
  Copyright (c) Tom Ritchford, 2012-2021

  Updates available from www.github.com/tipam/pi3d

DEBUG:pi3d.Display:Display size is w=1360, h=768
DEBUG:pi3d.Shader:no file shaders//home/pi/picframe_data/data/shaders/blend_new.vs in pkg_resources trying
DEBUG:pi3d.Shader:shader(2) /home/pi/picframe_data/data/shaders/blend_new, b'Compiled'
DEBUG:pi3d.Shader:no file shaders//home/pi/picframe_data/data/shaders/blend_new.fs in pkg_resources trying
DEBUG:pi3d.Shader:shader(3) /home/pi/picframe_data/data/shaders/blend_new, b'Compiled'
DEBUG:pi3d.Shader:saving shader /home/pi/picframe_data/data/shaders/blend_new
DEBUG:pi3d.util.Loadable:__init__: <pi3d.shape.Sprite.Sprite object at 0x74b7f710>
DEBUG:pi3d.util.Loadable:__init__: <pi3d.Buffer.Buffer object at 0x74b7f850>
DEBUG:pi3d.util.Loadable:__init__: <pi3d.Texture.Texture object at 0x74b7f950>
DEBUG:pi3d.Shader:shader(5) uv_flat, b'Compiled'
DEBUG:pi3d.Shader:shader(6) uv_flat, b'Compiled'
DEBUG:pi3d.Shader:saving shader uv_flat
DEBUG:pi3d.util.Loadable:__init__: <pi3d.shape.Sprite.Sprite object at 0x74b7fb30>
DEBUG:pi3d.util.Loadable:__init__: <pi3d.Buffer.Buffer object at 0x74b7fad0>
DEBUG:image_cache.ImageCache:Updating cache
DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
DEBUG:image_cache.ImageCache:Found 0 new files on disk
DEBUG:mat_image.MatImage:No valid mat types defined - using: ['float', 'float_polaroid', 'float_color_wrap', 'single_bevel', 'double_bevel', 'double_flat']
DEBUG:PIL.PngImagePlugin:STREAM b'IHDR' 16 13
DEBUG:PIL.PngImagePlugin:STREAM b'sRGB' 41 1
DEBUG:PIL.PngImagePlugin:STREAM b'IDAT' 54 227
DEBUG:PIL.PngImagePlugin:STREAM b'IHDR' 16 13
DEBUG:PIL.PngImagePlugin:STREAM b'sRGB' 41 1
DEBUG:PIL.PngImagePlugin:STREAM b'IDAT' 54 601
DEBUG:PIL.PngImagePlugin:STREAM b'IHDR' 16 13
DEBUG:PIL.PngImagePlugin:STREAM b'sRGB' 41 1
DEBUG:PIL.PngImagePlugin:STREAM b'IDAT' 54 99
DEBUG:PIL.PngImagePlugin:STREAM b'IHDR' 16 13
DEBUG:PIL.PngImagePlugin:STREAM b'sRGB' 41 1
DEBUG:PIL.PngImagePlugin:STREAM b'IDAT' 54 93
DEBUG:image_cache.ImageCache:Updating cache
DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
DEBUG:image_cache.ImageCache:Found 0 new files on disk
DEBUG:pi3d.util.Loadable:__init__: <pi3d.Texture.Texture object at 0x74066670>
DEBUG:pi3d.Texture:Loading ...PIL.Image (RGB)
DEBUG:pi3d.util.Loadable:__init__: <pi3d.util.FixedString.FixedString object at 0x74066890>
DEBUG:pi3d.util.Loadable:__init__: <pi3d.shape.Sprite.Sprite object at 0x740668b0>
DEBUG:pi3d.util.Loadable:__init__: <pi3d.Buffer.Buffer object at 0x74066d90>
DEBUG:image_cache.ImageCache:Updating cache
DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
DEBUG:image_cache.ImageCache:Found 0 new files on disk
^CYou pressed Ctrl-c!
DEBUG:pi3d.util.Loadable:__del__: <pi3d.shape.Sprite.Sprite object at 0x74b7f710>
DEBUG:pi3d.util.Loadable:__del__: <pi3d.shape.Sprite.Sprite object at 0x74b7fb30>
DEBUG:pi3d.util.Loadable:__del__: <pi3d.Texture.Texture object at 0x74b7f950>
DEBUG:pi3d.util.Loadable:__del__: <pi3d.Texture.Texture object at 0x74066670>
DEBUG:pi3d.util.Loadable:__del__: <pi3d.util.FixedString.FixedString object at 0x74066890>
DEBUG:pi3d.util.Loadable:__del__: <pi3d.shape.Sprite.Sprite object at 0x740668b0>
pi@piframe:~ $

**Here is a copy of my changes to the configuration.yaml file for the mount:**

model:
  pic_dir: "mnt/NAS_PICS/pics"                   # mnt/NAS_PICS/pics"   was "~/Pictures"  CHANGED                 default="~/Pictures", root folder for images
  deleted_pictures: "~/DeletedPictures"   # move deleted pictures here
  follow_links: True                      # CHANGED  default=False, By default, picframe will not walk down into symbolic links that resolve to directories. Set follow$
  no_files_img: "~/picframe_data/data/no_pictures.jpg" # default="PictureFrame2020img.jpg", image to show if none selected
  subdirectory: ""                        # default="", subdir of pic_dir - can be changed by MQTT"
  recent_n: 7                             # default=7 (days), when shuffling file change date more recent than this number of days play before the rest
  reshuffle_num: 1                        # default=1, times through before reshuffling
  time_delay: 30.0                        # CHANGED default=200.0, time between consecutive slide starts - can be changed by MQTT
  fade_time: 6.0                          # CHANGED default=10.0, change time during which slides overlap - can be changed by MQTT"
  shuffle: True                           # default=True, shuffle on reloading image files - can be changed by MQTT"
[... cut ...] as

**Here is copy of changes to configuration.yaml for debug logging - nothing else in .yaml has been changed:**

  log_level: "DEBUG"                    # default=WARNING, could be DEBUG, INFO, WARNING, ERROR, CRITICAL
  log_file: "/home/pi/picframe_data/data/logs"                            # default="" for debugging set this to the path to a file. NB logging messages will
                                          # appended indefinitely so don't forget this. You will need to tidy it up later

gn3481, 2021-11-21:
Not sure why copy/paste from nano to the above post resulted in strike-through under "model:", but in the .yaml, there is no strike-through.

helgeerbe, 2021-11-22:
Hi @gn3481 ,

I'm wondering, if `pic_dir: "mnt/NAS_PICS/pics" ` is correct.

In you first post you wrote `mnt/NAS_PICS/pics`. Maybe there is a leading `/`missing.

helgeerbe, 2021-11-22:
Just to be sure. If you put some images on the local drive they were displayed (e. g. in /home/pi/Pictures)?

gn3481, 2021-11-22:
OK - some progress...

Now reviewing log via putty off another computer on the network.
The mount does need the leading "/" as it is finding my pictures now...

The issue is they are not displayed - the "No Pictures Found" screen remains on the display, despite the debug data showing it has resolved the path to a picture - some debug data below (it has cycled through several of my pictures, but they are never displayed) :

**DEBUG:image_cache.ImageCache:Inserting: /mnt/NAS_PICS/pics/Arizona 2017 A/IMGP0412.JPG**
DEBUG:exifread:JPEG format recognized data[0:2]=0xFFD8
DEBUG:exifread:data[2]=0xFF data[3]=0xE1 data[6:10]=b'Exif'
DEBUG:exifread: Segment base 0x2
DEBUG:exifread:  APP1 at base 0x2
DEBUG:exifread:  Length: 0xF5 0x35
DEBUG:exifread:  Code: b'Exif'
DEBUG:exifread:  Decrement base by 2 to get to pre-segment header (for compatibility with later code)
DEBUG:exifread:Endian format is I (Intel)
DEBUG:exifread:IFD 0 (Image) at offset 8:
DEBUG:exifread: Make: (0x010F) ASCII=PENTAX RICOH IMAGING    @ 146
DEBUG:exifread: Model: (0x0110) ASCII=PENTAX WG-3 @ 170
DEBUG:exifread: Orientation: (0x0112) Short=Horizontal (normal) @ 42
DEBUG:exifread: XResolution: (0x011A) Ratio=72 @ 194
DEBUG:exifread: YResolution: (0x011B) Ratio=72 @ 202
DEBUG:exifread: ResolutionUnit: (0x0128) Short=Pixels/Inch @ 78
DEBUG:exifread: Software: (0x0131) ASCII=PENTAX WG-3 Ver. 1.00 @ 210
DEBUG:exifread: DateTime: (0x0132) ASCII=0000:00:00 00:00:00 @ 232
DEBUG:exifread: YCbCrPositioning: (0x0213) Short=Co-sited @ 114
DEBUG:exifread: ExifOffset: (0x8769) Long=602 @ 126
DEBUG:exifread: PrintIM: (0xC4A5) Undefined=[80, 114, 105, 110, 116, 73, 77, 0, 48, 51, 48, 48, 0, 0, 33, 0, 1, 0, 22, 0, ... ] @ 252
DEBUG:exifread:IFD 1 (Thumbnail) at offset 51766:
DEBUG:exifread: Compression: (0x0103) Short=JPEG (old-style) @ 51776
DEBUG:exifread: XResolution: (0x011A) Ratio=72 @ 51844
DEBUG:exifread: YResolution: (0x011B) Ratio=72 @ 51852
DEBUG:exifread: ResolutionUnit: (0x0128) Short=Pixels/Inch @ 51812
DEBUG:exifread: JPEGInterchangeFormat: (0x0201) Long=52452 @ 51824
DEBUG:exifread: JPEGInterchangeFormatLength: (0x0202) Long=7377 @ 51836
DEBUG:exifread:Exif SubIFD at offset 602:
DEBUG:exifread: ExposureTime: (0x829A) Ratio=1/1250 @ 932
DEBUG:exifread: FNumber: (0x829D) Ratio=14/5 @ 940
DEBUG:exifread: ISOSpeedRatings: (0x8827) Short=125 @ 636
DEBUG:exifread: ExifVersion: (0x9000) Undefined=0230 @ 648
DEBUG:exifread: DateTimeOriginal: (0x9003) ASCII=0000:00:00 00:00:00 @ 948
DEBUG:exifread: DateTimeDigitized: (0x9004) ASCII=0000:00:00 00:00:00 @ 968
DEBUG:exifread: ComponentsConfiguration: (0x9101) Undefined=YCbCr @ 684
DEBUG:exifread: ExposureBiasValue: (0x9204) Signed Ratio=0 @ 988
DEBUG:exifread: MeteringMode: (0x9207) Short=Pattern @ 708
DEBUG:exifread: Flash: (0x9209) Short=Flash did not fire, auto mode @ 720
DEBUG:exifread: FocalLength: (0x920A) Ratio=9/2 @ 996
DEBUG:exifread: FlashPixVersion: (0xA000) Undefined=0100 @ 756
DEBUG:exifread: ColorSpace: (0xA001) Short=sRGB @ 768
DEBUG:exifread: ExifImageWidth: (0xA002) Long=4608 @ 780
DEBUG:exifread: ExifImageLength: (0xA003) Long=2592 @ 792
DEBUG:exifread:Interoperability SubIFD at offset 51716:
DEBUG:exifread: InteroperabilityIndex: (0x0001) ASCII=R98 @ 51746
DEBUG:exifread: InteroperabilityVersion: (0x0002) Undefined=[48, 49, 48, 48] @ 51738
DEBUG:exifread: InteroperabilityOffset: (0xA005) Long=51716 @ 804
DEBUG:exifread: CustomRendered: (0xA401) Short=Normal @ 816
DEBUG:exifread: ExposureMode: (0xA402) Short=Auto Exposure @ 828
DEBUG:exifread: WhiteBalance: (0xA403) Short=Auto @ 840
DEBUG:exifread: DigitalZoomRatio: (0xA404) Ratio=1 @ 1004
DEBUG:exifread: FocalLengthIn35mmFilm: (0xA405) Short=25 @ 864
DEBUG:exifread: SceneCaptureType: (0xA406) Short=Standard @ 876
DEBUG:exifread: Contrast: (0xA408) Short=Normal @ 888
DEBUG:exifread: Saturation: (0xA409) Short=Normal @ 900
DEBUG:exifread: Sharpness: (0xA40A) Short=Normal @ 912
DEBUG:exifread: SubjectDistanceRange: (0xA40C) Short=2 @ 924
**DEBUG:image_cache.ImageCache:Inserting: /mnt/NAS_PICS/pics/Arizona 2017 A/little rock on right.JPG**
DEBUG:exifread:JPEG format recognized data[0:2]=0xFFD8
DEBUG:exifread:data[2]=0xFF data[3]=0xE1 data[6:10]=b'Exif'
DEBUG:exifread: Segment base 0x2
DEBUG:exifread:  APP1 at base 0x2
[... 34 lines cut ...]
DEBUG:exifread: MaxApertureValue: (0x9205) Ratio=4 @ 4822
DEBUG:exifread: MeteringMode: (0x9207) Short=Pattern @ 2414
DEBUG:exifread: Flash: (0x9209) Short=Flash did not fire, auto mode @ 2426
DEBUG:exifread: FocalLength: (0x920A) Ratio=26749/1000 @ 4830
DEBUG:exifread: FlashPixVersion: (0xA000) Undefined=0100 @ 2474
DEBUG:exifread: ColorSpace: (0xA001) Short=sRGB @ 2486
DEBUG:exifread: ExifImageWidth: (0xA002) Short=1600 @ 2498
DEBUG:exifread: ExifImageLength: (0xA003) Short=1200 @ 2510
DEBUG:exifread:Interoperability SubIFD at offset 7240:
DEBUG:exifread: InteroperabilityVersion: (0x0002) Undefined=[48, 49, 48, 48] @ 7250
DEBUG:exifread: RelatedImageWidth: (0x1001) Short=1600 @ 7262
DEBUG:exifread: RelatedImageLength: (0x1002) Short=1200 @ 7274
DEBUG:exifread: InteroperabilityOffset: (0xA005) Long=7240 @ 2522
DEBUG:exifread: FocalPlaneXResolution: (0xA20E) Ratio=64000/9 @ 7284
DEBUG:exifread: FocalPlaneYResolution: (0xA20F) Ratio=1200000/169 @ 7292
DEBUG:exifread: FocalPlaneResolutionUnit: (0xA210) Short=2 @ 2558
DEBUG:exifread: SensingMethod: (0xA217) Short=One-chip color area @ 2570
DEBUG:exifread: FileSource: (0xA300) Undefined=Digital Camera @ 2582
DEBUG:exifread: CustomRendered: (0xA401) Short=Normal @ 2594
DEBUG:exifread: ExposureMode: (0xA402) Short=Auto Exposure @ 2606
DEBUG:exifread: WhiteBalance: (0xA403) Short=Auto @ 2618
DEBUG:exifread: DigitalZoomRatio: (0xA404) Ratio=1 @ 7300
DEBUG:exifread: SceneCaptureType: (0xA406) Short=Standard @ 2642
DEBUG:exifread: Padding: (0xEA1C) Undefined=[] @ 2674
DEBUG:exifread: OffsetSchema: (0xEA1D) Signed Long=4160 @ 2666

gn3481, 2021-11-22:
I also moved several pictures to the local ~/Pictures directory. Reset .yaml to point to ~/Pictures
Pictures are displaying fine...

Only other parameters I changed from base .yaml were the duration and fade times. Will try making them larger and see what happens. Perhaps some sort of latency causing this...???

gn3481, 2021-11-22:
OK - 
Seems it is working now but it displays the NO pictures found screen at startup. 
If I leave it running long enough, it displays the pictures off my NAS.

helgeerbe, 2021-11-23:
Glad to hear that it is working now. On the first run, the image cache has to be build. On the next run, I would expect that the No pictures image is not shown anymore.

gn3481, 2021-11-23:
I think that was what caused my issue from the start - a "caching" display would be better than no pictures found, causing me to believe a config issue existed.

Thanks so much for your help !!!

jgodfrey, 2021-11-23:
> On the first run, the image cache has to be build.

Even this seems odd to me. The slideshow *should* start fairly quickly, even with no existing cache data. In that case, it should begin by caching enough to get started and then continue to load in the background. However, perhaps just the initial collection of files and folders on the mounted NAS is slow for some reason - not sure...

> a "caching" display would be better than no pictures found

Agreed - this could be improved. Although, again, it's _not supposed_ to take very long... 

Glad you have it working.

closed 2021-11-24 by helgeerbe

gn3481, 2021-11-24:
Thanks for your help !