helgeerbe/picframe #196: MQTT error in Home Assistant log at every picture change

vinyli85, 2021-11-21:
Hi,
i have the following error Message at every picture change in HomeKit Assistant Core log.

```
2021-11-21 11:49:10 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: select picframe_directory
2021-11-21 11:49:10 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_select_mqtt': ({'name': 'picframe_directory', 'icon': 'mdi:folder-multiple-image', 'options': ['Pictures'], 'state_topic': 'homeassistant/sensor/picframe/state', 'command_topic': 'picframe/directory', 'value_template': '{{ value_json.directory}}', 'availability_topic': 'homeassistant/switch/picframe/available', 'unique_id': 'picframe_directory', 'device': {'identifiers': ['picframe']}, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable
    return schema(data)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/select.py", line 39, in validate_config
    raise vol.Invalid(f"'{CONF_OPTIONS}' must include at least 2 options")
voluptuous.error.Invalid: 'options' must include at least 2 options
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 198, in async_discover
    config = schema(discovery_payload)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec
    raise e if self.msg is None else AllInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec
    v = func(v)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 276, in __call__
    raise er.MultipleInvalid([e])
voluptuous.error.MultipleInvalid: 'options' must include at least 2 options
```
pictureframe and HA running on 2 different RPi4, both are fresh installs with HA OS 6.6 and Raspian OS Bullseye on pictureframe. Picturefame is Set up according to Wolfgang‘s website www.thedigitalpictureframe.com

Anybody an idea whats the Problem here?

helgeerbe, 2021-11-21:
Hi @vinyli85,

what happens, if you create an empty directory below Pictures? So you will have 2 options.

vinyli85, 2021-11-21:
Hi @helgeerbe,

that was an easy fix. I created an empty folder and now no more error messages, perfect. 
thanks

closed 2021-11-21 by vinyli85

helgeerbe, 2021-11-22:
Hm, seems to be a home assistant "feature".  I can't see any reason, why creating an input_select element via mqtt requires to options in minimum. I'm pretty sure a manually created input_select element is allowed to have just one option.

helgeerbe, 2021-11-24:
Will be fixed in home assistant( https://github.com/home-assistant/core/pull/60281)

helgeerbe, 2021-12-13:
Selects can now have a single or no options in Home Assistant (since release 2021.12)