helgeerbe/picframe #197: Ken Burns effect not working in 2021.11.04 (?)

CanaMike, 2021-11-22:
I can't seem to get the Ken Burns effect to do anything.  I've tried experimenting with long time_delay and fade_time parameters (300s and 60s respectively) and after setting kenburns: True and restarting the picframe service I don't see any movement in the photos at all.  

I'm running picframe 2021.11.04 and Raspian version 10 (Buster) on a Raspberry Pi 4 with my photos loaded onto a 32 GB SD card.  The photos are all sized for my display resolution, i.e., 1920 x 1200.  

Am I just missing something? - are there any other settings that need to be altered to get the Ken Burns effect to work?

paddywwoof, 2021-11-22:
Hi, I will try to check this tomorrow if I get chance. I'm not sure how but it could have been effected by the last change to allow instant fading.
Paddy

CanaMike, 2021-11-22:
Thanks Paddy, much appreciated.  

I've just started using Picframe (it's a great program, BTW) so there's I suspect there's still a chance that I may have something setup incorrectly in config.yaml - I'll look forward to your findings.

Cheers,

Mike

helgeerbe, 2022-01-27:
@paddywwoof any news on this issue?

CanaMike, 2022-03-18:
Thanks, Thomas - I did as you suggested and the Ken Burns effect started working for me, though not for the reason you might expect(!) 

It seems that there's another important consideration that needs to be taken into account if you want to use the Ken Burns effect, i.e., if the photos you're displaying are exactly matched to your screen resolution and aspect ratio, the effect does nothing.  

I had all of my photos sized to 1920 x 1200 (same resolution and aspect ratio as my monitor) and nothing happened... as soon as I uploaded a few photos with random aspect ratios, the effect came to life.  I suppose it makes sense, since the effect needs "somewhere to go" in order to move the photo across the screen, i.e., if the photo is already a perfect fit for the display, then the effect has no "undisplayed area" left to work with.

Cheers,

Mike

jgodfrey, 2022-03-18:
> I had the same issue, setting kern burns to True with the default config does not work. The reason is the matting. Disable it (set 0.1 to False) and Ken is your friend again...

As the creator of the matting system, but not a user of the Ken Burns feature, I've never noticed this. Good catch @Weinschenk. While I haven't looked at the code recently, I assume this is because the matting system produces an image that matches the display resolution - which leads to the case mentioned above by @CanaMike. 

We probably need to make these two options mutually exclusive in some reasonable way...

Tallman52, 2022-03-21:
Great thread. I was using Wolfgang's old 2017 digital picture frame setup and ran across the new Pic frame setup. I knew I had to give it a go. Thank you all who were part of this project.

I have successfully installed new version on pi3b and pi4. After fiddling around, I absolutely love it.

I tried the Ken Burns effect and could not get it working with the matting on (kuddos to jgodfrey). After reading this thread, I think the Ken Burns effect is working, kind of. The pictures pan up (only) and there appears to be no zoom happening. Is there something I am missing? I would assume the pan and zoom effect should work in tandem. I also do not see any settings to adjust the amount of pan/zoom.

Please don't read any of this as negative. The Pic frame project is really nice! Thanks.

helgeerbe, 2022-03-24:
> We probably need to make these two options mutually exclusive in some reasonable way...

and document this in the wiki.

closed 2026-05-06 by github-actions[bot]