helgeerbe/picframe #198: Adding HEIC images while frame is running

devvthedev, 2021-12-04:
Hi 👋 I've been using picframe alongside RCLONE to add and remove pictures from the directory picframe uses.

An issue I've been having with adding HEIC pictures from my iPhone is around missing metadata specifically the _location_ data on a picture. 

These pictures are being added to the directory whilst the frame is running on the pi. The pictures appear OK when added like this but the location meta data is missing and the date is sometimes incorrect.

I have however managed to get the meta data to appear by stopping the photo frame and then I add the photos to the directory before starting up the frame again.

It's not a massive issue but would be awesome to be able to carry on adding/removing photos with the location meta data still intact.

Are there any plans to support this use case?

Thank You!

paddywwoof, 2021-12-04:
At one stage I thought there was an issue with the python heic library getting meta data but it sounds like that's OK if it works after a restart. So it must be a bug in the mechanism for background checking and loading the database.

devvthedev, 2021-12-06:
Hi, so it doesn't work if the picture was added whilst picframe is running even if a restart is done. 

To get the location data to show I Have to :

1. Remove the photo, 
2. Stop picframe, 
3. Add the photo again 
4. And then start picframe again.

jgodfrey, 2021-12-07:
Hi @devvthedev,

A few questions / comments:

Do you know if you see this issue with other image types (JPG for example), or only with HIEC?

> The pictures appear OK when added like this but the location meta data is missing and the date is sometimes incorrect.

In this case, you just don't see ANY location data displayed? Related to the incorrect date, if we can't find a valid EXIF date, they code falls back to the timestamp on the file itself. Is that maybe what it's showing here?

Do you have any idea what's in the database for the offending files? For instance, the `all_data` view should contain all of the info in question for each file (exit date, lat, lon, location description, ...). It might be interesting to see what's stored for an offending file.

> To get the location data to show I Have to :

That's certainly not intended and seems like a bug. Though, I can't say that I've seen it personally. Though, I don't use HEIF/HEIC images (assuming that's related).

helgeerbe, 2022-01-27:
This seems to be a duplicate of issue #234 
Should hopefully fixed with pull request #235

helgeerbe, 2022-03-24:
Should be fixed with 2022.03.24.

closed 2022-03-24 by helgeerbe