helgeerbe/picframe #199: Starting succeeds at display resolution 2560x1600 only if the width is limited to 2048 pixel

Nerdiyde, 2021-12-07:
Hi Guys,

I'm trying to use your (btw awesome) project for one of my projects. 
Sadly I have the issue that picframe does not start with my display setup in case I set the correct display resolution. (More on that below)

**First:**
At starting picframe I get the following error report:
`picframe ~/picframe_data/config/configuration.yaml                                                                                                                                                                        INFO:start.py:starting ['/home/pi/.local/bin/picframe', '/home/pi/picframe_data/config/configuration.yaml']                                                                                                                                  INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:                                                                                                                                                         DEBUG:image_cache.ImageCache:Creating an instance of ImageCache                                                                                                                                                                              DEBUG:image_cache.ImageCache:Updating cache                                                                                                                                                                                                  INFO:controller.Controller:creating an instance of Controller                                                                                                                                                                                DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk                                                                                                                                                                   DEBUG:image_cache.ImageCache:Found 0 new files on disk                                                                                                                                                                                       DEBUG:pi3d.Display:

  Pi3D module - version 2.49

  Copyright (c) Tim Skillman, 2012-2021
  Copyright (c) Patrick Gaunt, 2012-2021
  Copyright (c) Tom Ritchford, 2012-2021

  Updates available from www.github.com/tipam/pi3d

DEBUG:pi3d.Display:Display size is w=2560, h=1600
Traceback (most recent call last):
  File "/home/pi/.local/bin/picframe", line 10, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.7/site-packages/picframe/start.py", line 136, in main
    c.start()
  File "/home/pi/.local/lib/python3.7/site-packages/picframe/controller.py", line 317, in start
    self.__viewer.slideshow_start()
  File "/home/pi/.local/lib/python3.7/site-packages/picframe/viewer_display.py", line 418, in slideshow_start
    display_config=pi3d.DISPLAY_CONFIG_HIDE_CURSOR, background=self.__background, use_glx=self.__use_glx)
  File "/home/pi/.local/lib/python3.7/site-packages/pi3d/Display.py", line 570, in create
    display_config=display_config, window_title=window_title, use_glx=use_glx)
  File "/home/pi/.local/lib/python3.7/site-packages/pi3d/util/DisplayOpenGL.py", line 122, in create_display
    self.create_surface(x, y, w, h, layer)
  File "/home/pi/.local/lib/python3.7/site-packages/pi3d/util/DisplayOpenGL.py", line 332, in create_surface
    assert r
AssertionError`

**My assumpation is that this is related to the (more or less) unique display I'm using. This is a 8.9" display with a resolution of 2560x1600 pixel. 
For this I made the following changes in the /boot/config.txt** 

`
.
.
.
gpu_mem=128                                                                                                                                                                                                                                  hdmi_group=2                                                                                                                                                                                                                                 hdmi_mode=87                                                                                                                                                                                                                                 hdmi_timings=2560 1 10 10 80 1600 1 4 4 4 0 0 0 50 0 128000000 5                                                                                                                                                                             #hdmi_timings=2560 1 10 10 80 1600 1 4 4 4 0 0 0 50 0 128000000 5                                                                                                                                                                            framebuffer_width=2560                                                                                                                                                                                                                       framebuffer_height=1600                                                                                                                                                                                                                      max_framebuffer_width=2560                                                                                                                                                                                                                   max_framebuffer_height=1600                                                                                                                                                                                                                  hdmi_pixel_freq_limit=128000000                                                                                                                                                                                                              config_hdmi_boost=4                                                                                                                                                                                                                          max_usb_current=1 `

**The display output is working fine.
Also I'm able to start picframe If i limit the width of the displayed area either in the rpi config file or in the picframe config file.**

For example picframe runs with the following adaptions in its config file:
`  display_w: 2048                         # width of display surface (null->None will use max returned by hardware)
  display_h: 1600                         # height of display surface`

but refues to start with this setting:

`  display_w: 2049                         # width of display surface (null->None will use max returned by hardware)
  display_h: 1600                         # height of display surface`

Basically all display width >2048 lead to the error shown at the top.

With the working setup picframe starts but is of course only displayed partly on the display. :)

Further details about my setup:
pi@fotoPhrame1:~ $ neofetch --stdout                                                                                                                                                                                                         pi@fotoPhrame1                                                                                                                                                                                                                               --------------                                                                                                                                                                                                                               OS: Raspbian GNU/Linux 10 (buster) armv7l                                                                                                                                                                                                    Host: Raspberry Pi Zero 2 W Rev 1.0                                                                                                                                                                                                          Kernel: 5.10.63-v7+                                                                                                      [... cut ...]    Terminal: /dev/pts/0                                                                                                                                                                                                                         CPU: BCM2835 (4) @ 1.000GHz                                                                                                                                                                                                                  Memory: 129MiB / 364MiB

helgeerbe, 2021-12-07:
Hi @Nerdiyde 

do you have more details of the display?  I just googled a similar display (https://www.waveshare.com/wiki/9inch_2560x1600_Monitor) and found this settings for it:

````
max_usb_current=1
gpu_mem=512     #If you are using Pi Zero, please change to gpu_mem=256
hdmi_group=2
hdmi_mode=87
hdmi_timings=2560 1 10 10 80 1600 1 4 4 4 0 0 0 60 0 128000000 5
framebuffer_width=2560
framebuffer_height=1600
max_framebuffer_width=2560
max_framebuffer_height=1600
hdmi_pixel_freq_limit=268500000
config_hdmi_boost=4
max_usb_current=1
````

Nerdiyde, 2021-12-07:
Hi Helgeerbe,

Thanks for your help. :)

The display you linked there is similar to the one I'm using. :) 
I got it from aliexpress. It's this one: https://de.aliexpress.com/item/1005001447618279.html

Also with the settings I'm using right now, the display is working fine. It shows the standard raspberrypi desktop environment (Picture is available here: https://www.instagram.com/p/CWWP4wnNg4U/ (I hope this works))

The only problem I have is that picframe does not start if I set it up to use the native display resolution.
It starts only if I limit the "display_w" parameter in the config file (of picframe) to a lower value. 
Here 2048 is the highest "display_w" value that is possible. In this case the picframe content is shown on the left display area but at the right side a stripe of the raspberrypi desktop is still visible. (If you like I can provide a picture tomorrow)
At higher values of "display_w" I get the shown assertion error. :/

Thanks and best regards
Fab

paddywwoof, 2021-12-07:
@Nerdiyde I will have a look through the code. For Texture2D samplers pi3d checks what the maximum size the GPU can cope with, on Pi<4 this is 2048 but Pi4 can go up to 4K. However I can't remember if this is used for the `pi3d.Display.width` There is a value returned by 'the display' i.e. for the older Pis (but including zero 2) it uses a [libbcm_host.so function](https://github.com/tipam/pi3d/blob/1f1c822dc3ac58344ad2d5468978d62e51710df4/pi3d/util/DisplayOpenGL.py#L56)

It would be interesting to know what value is being returned by the initial setup of the Display. In `pi3d.Display.__init__()` there is a value returned by the first link above, setting `Display.max_width` but, later, this can be overwritten by any value when `create_display()` [is called around here](https://github.com/tipam/pi3d/blob/1f1c822dc3ac58344ad2d5468978d62e51710df4/pi3d/Display.py#L538).

There doesn't look to be any verification of the values of w and h sent to `DisplayOpenGL.create_surface()` method [see here](https://github.com/tipam/pi3d/blob/1f1c822dc3ac58344ad2d5468978d62e51710df4/pi3d/util/DisplayOpenGL.py#L174)

To log the `max_width` you would have to hack the code a bit; In `/home/pi/.local/lib/python3.7/site-packages/pi3d/Display.py` Change line 548 to
```python
  LOGGER.debug('Display size is w=%d, h=%d, max_width=%d, max_height=%d', w, h, display.max_width, display.max_height)
```

Nerdiyde, 2021-12-08:
HI @paddywwoof 

thanks a lot for your help.
Your assumption sounds reasonable to me. Looks like it is somehow limited to these 2048 pixel. Would be awesome if there is a way to overwrite this limitation. :)

I integrated the additional debug output and tested the following scenarios/settings.
It looks like that the set value (or display resolution in case no other value is set) is forwarded properly to the location where the debug output is printed. 

1. Configuration:
>  display_w: 2560                         # width of display surface (null->None will use max returned by hardware)
>  display_h: 1600                         # height of display surface

Result:
```
INFO:start.py:starting ['/home/pi/.local/bin/picframe', '/home/pi/picframe_data/config/configuration.yaml']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
DEBUG:image_cache.ImageCache:Creating an instance of ImageCache
DEBUG:image_cache.ImageCache:Updating cache
INFO:controller.Controller:creating an instance of Controller
DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
DEBUG:image_cache.ImageCache:Found 0 new files on disk
DEBUG:pi3d.Display:

  Pi3D module - version 2.49

  Copyright (c) Tim Skillman, 2012-2021
  Copyright (c) Patrick Gaunt, 2012-2021
  Copyright (c) Tom Ritchford, 2012-2021

  Updates available from www.github.com/tipam/pi3d

DEBUG:pi3d.Display:Display size is w=2560, h=1600, max_width=2560, max_height=1600
DEBUG:image_cache.ImageCache:Updating cache
DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
DEBUG:image_cache.ImageCache:Found 0 new files on disk
DEBUG:image_cache.ImageCache:Updating cache
DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
DEBUG:image_cache.ImageCache:Found 0 new files on disk
DEBUG:image_cache.ImageCache:Updating cache
DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
DEBUG:image_cache.ImageCache:Found 0 new files on disk
DEBUG:image_cache.ImageCache:Updating cache
DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
DEBUG:image_cache.ImageCache:Found 0 new files on disk
DEBUG:image_cache.ImageCache:Updating cache
DEBUG:image_cache.ImageCache:No unprocessed files in memory, checking disk
DEBUG:image_cache.ImageCache:Found 0 new files on disk
Traceback (most recent call last):
  File "/home/pi/.local/bin/picframe", line 10, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.7/site-packages/picframe/start.py", line 136, in main
    c.start()
  File "/home/pi/.local/lib/python3.7/site-packages/picframe/controller.py", line 317, in start
    self.__viewer.slideshow_start()
  File "/home/pi/.local/lib/python3.7/site-packages/picframe/viewer_display.py", line 418, in slideshow_start
    display_config=pi3d.DISPLAY_CONFIG_HIDE_CURSOR, background=self.__background, use_glx=self.__use_glx)
  File "/home/pi/.local/lib/python3.7/site-packages/pi3d/Display.py", line 571, in create
    display_config=display_config, window_title=window_title, use_glx=use_glx)
  File "/home/pi/.local/lib/python3.7/site-packages/pi3d/util/DisplayOpenGL.py", line 122, in create_display
    self.create_surface(x, y, w, h, layer)
  File "/home/pi/.local/lib/python3.7/site-packages/pi3d/util/DisplayOpenGL.py", line 332, in create_surface
    assert r
AssertionError
Exception ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 1281, in _shutdown
    t.join()
  File "/usr/lib/python3.7/threading.py", line 1032, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock
[... 95 lines cut ...]
DEBUG:pi3d.util.Loadable:__init__: <pi3d.Texture.Texture object at 0x748fea50>
DEBUG:pi3d.Shader:shader(5) uv_flat, b'Compiled'
DEBUG:pi3d.Shader:shader(6) uv_flat, b'Compiled'
DEBUG:pi3d.Shader:saving shader uv_flat
DEBUG:pi3d.util.Loadable:__init__: <pi3d.shape.Sprite.Sprite object at 0x748fea70>
DEBUG:pi3d.util.Loadable:__init__: <pi3d.Buffer.Buffer object at 0x748fe950>
DEBUG:mat_image.MatImage:No valid mat types defined - using: ['float', 'float_polaroid', 'float_color_wrap', 'single_bevel', 'double_bevel', 'double_flat']
DEBUG:PIL.PngImagePlugin:STREAM b'IHDR' 16 13
DEBUG:PIL.PngImagePlugin:STREAM b'sRGB' 41 1
DEBUG:PIL.PngImagePlugin:STREAM b'IDAT' 54 227
DEBUG:PIL.PngImagePlugin:STREAM b'IHDR' 16 13
DEBUG:PIL.PngImagePlugin:STREAM b'sRGB' 41 1
DEBUG:PIL.PngImagePlugin:STREAM b'IDAT' 54 601
DEBUG:PIL.PngImagePlugin:STREAM b'IHDR' 16 13
DEBUG:PIL.PngImagePlugin:STREAM b'sRGB' 41 1
DEBUG:PIL.PngImagePlugin:STREAM b'IDAT' 54 99
DEBUG:PIL.PngImagePlugin:STREAM b'IHDR' 16 13
DEBUG:PIL.PngImagePlugin:STREAM b'sRGB' 41 1
DEBUG:PIL.PngImagePlugin:STREAM b'IDAT' 54 93
DEBUG:image_cache.ImageCache:Updating cache
```

I hope this helps.
Thanks and best regards
Fabian

paddywwoof, 2021-12-08:
Thanks for doing that, definitely a bit more to go on. It looks like `libbcm_host.bcm.graphics_get_display_size()` gets hold of the correct dimensions for the display but one of the functions in `create_surface()` is unhappy. The bcm stuff is a pretty opaque (hence the desire to switch to open source mesa stuff in bullseye) but there is a function here that wants the width bit shifted 16 places up, which might be overflowing something. Maybe?

I will see if there's any way to get more info on what might be going on.

Nerdiyde, 2021-12-08:
@paddywwoof 

Sounds good, let me know if you need anything else or I shall test anything. :)

Thanks and best regards
Fab

Nerdiyde, 2022-06-16:
Hey guys,
sorry for coming back to this. 
I'm just curious If I can help with any additional information here?
Would be cool to get this solved. :)
Thanks in advance and best regards
Fab

paddywwoof, 2022-06-16:
Fabian, sorry not to follow up on this. I'm still not really sure what might be causing the issue but @thomasedoff has made quite a good [set of instructions](https://gist.github.com/thomasedoff/3973846101da1e7496524be075f1c02f) for setting up various RPi models all using the latest OS (bullseye) whereas I had struggled and ended up reverting to the previous version. I don't know if the drivers will work on the zero Mk1. If the problem is with the broadcom driver lib_bcmhost.so then that might get round it, but if it's due to the GPU hardware it won't.

If I get a chance later I will try running on a zero Mk1 here. And set the screen to 2560!

Nerdiyde, 2022-06-18:
Hey @paddywwoof , 
no problem. I'm happy to get the support here so nothing to worry about. :)
I tried yesterday to see how the setup behaves on bullseye. It looks like that (even with the fake kms activated) my custom hdmi config (which is needed to drive the off spec resolution) does not work anymore.
I started a thread in the raspberrypi forum to see if they can help me with this. I will let you know as soon as I get any feedback there. :)
Thanks and best regards
Fab

Nerdiyde, 2022-06-26:
Hi Guys,
so I did some more testing in the last days. Not sure If I made a real progress but maybe this helps to find a starting point where to investigate further.
What I did:
- Started a fresh raspian image (bullseye, no desktop)
- Set up the config properly to support the custom hdmi config (activated fkms, raspi boots up properly in console view)
- performed the configuration mentioned here: https://gist.github.com/thomasedoff/3973846101da1e7496524be075f1c02f
- Started picframe

The following happened:
If I start picframe with the untouched configuration the example picture is displayed properly in the upper left corner but to small. Similar to the desktop view (screenshot 1) in this post: https://forums.raspberrypi.com/viewtopic.php?p=2014551#p2014551 (thats by the way the thread on the raspberrypi forum about the correct hdmi settings using bullseye)

If I change the resolution in the picframe configuration to 2560x1600pxl the displayed content is corrupted (shifted picture parts) and again displayed only in the top left corner of the display not using the whole possible resolution.

If I change the resolution in the picframe configuration to 2048x2048 it looks like the example picture included in picframe is properly resized to 2048x2048pxl but again only the upper left part is visible.

I'm not really familiar with X so this is only an assumption: 
For me it looks like the displayed area is already limited by X. Sadly I dont know how to resize this to the full screen resolution. This correlates also to the behaviour described in https://forums.raspberrypi.com/viewtopic.php?p=2014551#p2014551.
There the desktop environment is displayed properly but displayed only in the upper left corner. The resolution of the displayed area can be changed using the raspberry pi app "Screen configuration" but is limited in this app to 1368x768 only. So its not possible to set the full resolution here.

Do you have any Idea what else I could try? I can also deliver any addtional logfiles. Just let me know. :)

Thanks in advance and best regards
Fab

Nerdiyde, 2022-10-20:
Hey guys,
long time no see...
Just wanted to let you know that this does not work (from my point of view) due to the 3D HW limitation of the Pi Zero 2 to "only" 2048x2048 pixel.  I changed my design now and integrated and pi4 which works perfect.
Thanks again for that awesome piece of software and the support. :)
Best regards
Fab

closed 2022-10-20 by Nerdiyde