helgeerbe/picframe #203: Text fade-in is not smooth anymore

glenvorel, 2021-12-16:
The commit 3bd1fb73c97ef735727b5b9ee63881e2a39095da changed the behavior of text fade-in and fade-out.

Using the previous logic, both fade-in and fade-out were relatively smooth.

The new logic seems to have two issues:
1. The text appears suddenly which is somewhat disturbing when all other effects are smooth.
2. By the end of the `show_text_tm` period, there are several frames (seconds) without displaying text which effectively means that picframe doesn't respect the `show_text_tm` value defined in config.

This commit uses for `dt` a value range [0.1, 1.1]. If it was [0, 1] - as before the commit - both issues would be solved.
I could write a PR but looking at the comment on line [viewer_display.py#L510](https://github.com/helgeerbe/picframe/commit/3bd1fb73c97ef735727b5b9ee63881e2a39095da#diff-e4c9ed0fdd9af33983135f8553dbd3d0961afe171313e281c76e0bd90d43c357R510), it appears that this shift is intentional.

I would like to ask is it really intentional and what is the rationale? Thank you

paddywwoof, 2021-12-16:
I don't think that line was supposed to change materially and that it's probably a typo a.k.a. error on my part (^_^;).
```python
            dt = (self.__show_text_tm - self.__name_tm + tm + 0.1) / self.__show_text_tm
# expands to ==>
            dt = self.__show_text_tm / self.__show_text_tm - self.__name_tm / self.__show_text_tm + tm / self.__show_text_tm + 0.1 / self.__show_text_tm
# simplifies to ==>
            dt = 1.0 + 0.1 / self.__show_text_tm - (self.__name_tm - tm) / self.__show_text_tm # i.e. around 0.01 to 1.01
  ```
I look a bit more at the logic to see why it needs the 0.1, presumably something bad happens when dt is 0.0, like divide by zero.

glenvorel, 2021-12-16:
Got it, thank you! It actually seems to be working well after simply changing the line from

```python
dt = 1.1 - (self.__name_tm - tm) / self.__show_text_tm
```

to

```python
dt = 1.0 - (self.__name_tm - tm) / self.__show_text_tm
```

The function creates a linear space which starts negative, then equals 0.0 at the end of `fade_time` and reaches 1.0 at `__name_tm`.
Explicitly setting the value of `dt` to 0.0 doesn't seem to cause any issues down the line either.

jgodfrey, 2022-01-01:
Regarding this code (shown above):

`dt = 1.0 - (self.__name_tm - tm) / self.__show_text_tm`

While it does allow the text to fade in nicely, it also causes the text's `alpha`  value to never quite reach zero as it fades out. If it's displayed over a significantly dark background, it's still faintly visible after it should be gone.

I also notice that the text still displays _abruptly_ when shown on matted images. I haven't looked into why that is yet, but I suspect it may be due to the extra processing time required to generate the mat (?). Though, if that's the case, it would seem to be a bug...

jgodfrey, 2022-01-02:
> I haven't looked into why that is yet, but I suspect it may be due to the extra processing time required to generate the mat (?).

This assumption is correct. At the beginning of [slideshow_is_running()](https://github.com/helgeerbe/picframe/blob/5574fde4d198afd6a1a83d22b24fdaa63d53d64d/picframe/viewer_display.py#L440), we get the current time stamp, which is used as a baseline for timed events for the current image. Soon after, we calculate a _future_ time stamp for when the [text fade-in should complete](https://github.com/helgeerbe/picframe/blob/5574fde4d198afd6a1a83d22b24fdaa63d53d64d/picframe/viewer_display.py#L444). Only at that point do we actually begin to load the next image. If the image(s) are to be matted, that process takes a few cycles to complete. So, once the matted image is returned, we've already eaten up some of the time that was intended for the text fade in. Later, we calculate the current percentage through the _fade time_ and use it to set the text's alpha value. However, in this case, we're already a good way through the fade time right out of the gate, due to the matting delay. So, the initial alpha is calculated at a much higher value that expected, causing the text to be displayed abruptly. 

Really, this is also a problem with non-matted images, but it's much less noticeable since the initial load time is quicker in that case.

I've reorganized some of the code near the top of `slideshow_is_running()` so that it doesn't calculate the frame's timer values until after the image is loaded (matted or not). This fixes the abrupt display of text on matted images (so it fades in smoothly as expected), but I'm not sure if I may have caused some subtle timing issues elsewhere. That, and it's just messy. It'd be nice to have a cleaner solution here...

@paddywwoof  - any thoughts here?

paddywwoof, 2022-01-03:
Yes this has needed a proper re-think for a long time. Your fix sounds sensible - ideally the fading of images and text should happen when other processes are not happening and the delta alpha values calculated at the start of the fade. The only thing I can think of is where the fade times won't fit in the time available (because the image loading took a chunk of the time up) should the next slide be delayed, or should the delta alpha be increased.

I can't remember why the time stamp is taken once. I suppose it saves calling the `time.time()` several times but that's probably trivial in the scale of the other processing.

jgodfrey, 2022-01-15:
Just merged https://github.com/helgeerbe/picframe/pull/229 into `dev` for this.

helgeerbe, 2022-01-15:
Fix works

helgeerbe, 2022-01-20:
Fixed in 2022.01.20

closed 2022-01-20 by helgeerbe