helgeerbe/picframe #207: Filtering on a date range independent of the year possible?

sapnho, 2021-12-27:
I was wondering if I can filter on a date range, e.g., a birthday or a public holiday regardless of the year.

For example, it would be nice to show only photos taken on a person's birthday but from all the years.

Is this at all possible? Thanks!

paddywwoof, 2022-01-02:
Hi, sorry for the delay. I'm not sure whether picframe allows SQL injection via some route or other, or if it would need a cascade of additional methods adding to `controller` and `interface_mqtt` but you basically need to set something up like:
```python
...
    self.__model.set_where_clause('date_day', "strftime('%d', exif_datetime = '{:02d}'".format(day_val))
...
    self.__model.set_where_clause('date_month', "strftime('%m', exif_datetime = '{:02d}'".format(month_val))
...
```
Which would probably just require a bit of copying and pasting. Doing a range, i.e. 24th Dec to 1st of Jan every year would be **much** harder to do.

Paddy

paddywwoof, 2022-01-03:
PS did a quick hack and couldn't get this to work. Picframe got stuck at the no images image. Couldn't escape without stopping and starting the systemd process! Will look at it again later.

sapnho, 2022-01-03:
Thanks for checking, Paddy!

helgeerbe, 2022-01-05:
@paddywwoof Just thinking out loud.
Can we use the  _ wildcard in date from string? 
If we detect this wildcard in date from, we will ignore date_to

Query would be something like
````sql
SELECT *
FROM images
WHERE date LIKE date_from;
````

paddywwoof, 2022-01-05:
Yes, I think it's a good idea to piggy back on the `date_from` setter/getter code. However the field is stored in the db as a REAL as I don't think sqlite has 'proper' date specific fields. It should be possible to use a date function in the where clause (worked on a different test when I tried it) but I've failed to get it to work on the image db after a little bit of testing. I will have another go later.

jgodfrey, 2022-01-05:
>  I don't think sqlite has 'proper' date specific fields

That's correct. In SQLite, dates can be stored as TEXT, REAL, and/or INTEGER values. Applications are free to use any/all of these data types for date storage and then use the provided [date and time functions](https://www.sqlite.org/lang_datefunc.html) to convert between them as necessary.

jgodfrey, 2022-01-05:
Related to the general thread, it'd be nice store named SQL "search fragments" in the DB and then be able to switch between them via MQTT. So, rather than the ad hoc creation of a search query that'd show "Christmas" photos (for example), a "Christmas" search fragment could be stored in the db. Then, the User could simply select "Christmas" for the current search.

It'd take some rework of the existing image selection features, but it should be doable. I've done similar things (custom, fragment-based searches) in other SQLite-based apps, and it's really flexible in practice.

Of course, the fragments themselves would have to be inserted into the db manually - at least for now.