helgeerbe/picframe #221: picframe not searching NFS

sedwards58, 2022-01-12:
It doesn't appear that picframe is finding photos on an NFS share.

Ideally, I'd like picframe to recurse through the subdirectories, but ignore directories that start with a dot. For example, recurse /photos/ into /photos/2022/2022-01/full/ but ignore  /photos/2022/2022-01/.thunbs/.

Any suggestions on what configuration settings I should use?

helgeerbe, 2022-01-14:
If I remember right, picframe already ignores all directories starting with a dot.
If it is not finding the pictures check your access rights. On pi3 I assume picframe is running as root. On pi3 I assume it is running as user pi.

jgodfrey, 2022-01-14:
> If I remember right, picframe already ignores all directories starting with a dot.

I don't think that's true for _folders_, although we do skip _files_ starting with `.`

helgeerbe, 2022-01-15:
There seems to be an inconsistency between cache and model subdirectory

````python
def get_directory_list(self):
        _, root = os.path.split(self.__pic_dir)
        actual_dir = root
        if self.subdirectory != '':
            actual_dir = self.subdirectory
        follow_links = self.get_model_config()['follow_links']
        subdir_list = next(os.walk(self.__pic_dir, followlinks=follow_links))[1]
        subdir_list[:] = [d for d in subdir_list if not d[0] == '.']
        if not follow_links:
            subdir_list[:] = [d for d in subdir_list if not os.path.islink(self.__pic_dir + '/' + d)]
        subdir_list.insert(0,root)
        return actual_dir, subdir_list
`````

Maybe we should exclude hidden folders also from cache?

jgodfrey, 2022-01-15:
> Maybe we should exclude hidden folders also from cache?

Yeah, agreed. The image loader should probably ignore hidden folders. I'll commit something.

jgodfrey, 2022-01-15:
Just merged https://github.com/helgeerbe/picframe/pull/230 to skip hidden folders when loading images.

helgeerbe, 2022-01-20:
Skip hidden folders will be available with release 2022.01.20. But were you able to use your NFS?

Gabvoir, 2022-02-22:
Sorry for my newbies. I'm under Pi Zero, and I use pip instead pip3.
With a "pip3 install picframe --upgrade" all i okey !

closed 2022-03-24 by helgeerbe