helgeerbe/picframe #224: Getting error - "sqlite3.OperationalError: database is locked"

Ascendin81, 2022-01-14:
Hi,

I am executing command: 

sudo python3 run_start.py /home/pi/picframe/picframe_data/config/configuration.yaml

I get this output:
pi@raspberrypi:~/picframe/picframe_data $ sudo python3 run_start.py /home/pi/pic                                           frame/picframe_data/config/configuration.yaml
INFO:start.py:starting ['run_start.py', '/home/pi/picframe/picframe_data/config/                                           configuration.yaml']
INFO:model.Model:Open config file: /home/pi/picframe/picframe_data/config/config                                           uration.yaml:
Traceback (most recent call last):
  File "run_start.py", line 2, in <module>
    start.main()
  File "/home/pi/picframe/picframe/start.py", line 130, in main
    m = model.Model(args.configfile)
  File "/home/pi/picframe/picframe/model.py", line 177, in __init__
    model_config['portrait_pairs'])
  File "/home/pi/picframe/picframe/image_cache.py", line 41, in __init__
    self.__update_schema(3)
  File "/home/pi/picframe/picframe/image_cache.py", line 318, in __update_schema
    self.__db.execute("DROP VIEW all_data")
sqlite3.OperationalError: database is locked

What does it mean by "database is locked"?

Thank you,

Ascendin81, 2022-01-14:
UPDATE: I was able to execute the command once, and I have that blue screen with "No Pictures Selected" text on it.  But, I do not know how to close out of that running instance of the program.  I tried Ctrl+C, Ctrl+Z.  Maybe, this is why the database is locked, because another instance is using it?  How do I stop the other instance?

Thank you,

Ascendin81, 2022-01-14:
UPDATE 2: I was able to type "exit" in the terminal and get back to the desktop, but now when I use the same command from the original post, I get this output and error:

pi@raspberrypi:~/picframe/picframe_data $ sudo python3 run_start.py /home/pi/picframe/picframe_data/config/configuration.yaml
INFO:start.py:starting ['run_start.py', '/home/pi/picframe/picframe_data/config/configuration.yaml']
INFO:model.Model:Open config file: /home/pi/picframe/picframe_data/config/configuration.yaml:
Traceback (most recent call last):
  File "run_start.py", line 2, in <module>
    start.main()
  File "/home/pi/picframe/picframe/start.py", line 130, in main
    m = model.Model(args.configfile)
  File "/home/pi/picframe/picframe/model.py", line 177, in __init__
    model_config['portrait_pairs'])
  File "/home/pi/picframe/picframe/image_cache.py", line 41, in __init__
    self.__update_schema(3)
  File "/home/pi/picframe/picframe/image_cache.py", line 319, in __update_schema
    self.__db.execute("ALTER TABLE folder ADD COLUMN missing INTEGER DEFAULT 0 NOT NULL")
sqlite3.OperationalError: duplicate column name: missing

helgeerbe, 2022-01-14:
Hi @Ascendin81. I would propose you stop picframe and delete the database (`~/picframe_data/data/pictureframe.db3`). If you restart picframe the database will be recreated.

Ascendin81, 2022-01-14:
Yes, ok that worked.  I got the blue screen again with the coffee cup.  How do I exit it correctly? :)  so the database does not get corrupted again.

Thanks!

helgeerbe, 2022-01-14:
If you enable the http interface, there is a stop button. Or run picframe as a systemd service https://github.com/helgeerbe/picframe/wiki/Running%20As%20Service

helgeerbe, 2022-01-14:
There is a pending pull request for mouse, touchscreen and keyboard. So chances are good, that keyboard will work in near future.

jgodfrey, 2022-01-14:
> self.__db.execute("ALTER TABLE folder ADD COLUMN missing INTEGER DEFAULT 0 NOT NULL")

Related, the cause of the above error has been fixed in `dev` also. It's caused by the following situation:

- picframe is started for the first time (so, no database exists)
- The initial db is created and upgraded to the current schema (but not committed, this is the bug that has been fixed in `dev`)
- Something (anything) goes wrong later in the execution (*prior* to any images being loaded) that causes a crash

- From this point on, when `picframe` is restarted, you'll get the mentioned error. Without the bug fix, the best way around it is to remove the database file.

helgeerbe, 2022-01-14:
Well I guess it's time to make a new release.

Ascendin81, 2022-01-14:
Ah, thank you guys, everything is working great now.  I love it!

closed 2022-01-14 by helgeerbe

jgodfrey, 2022-01-15:
> Well I guess it's time to make a new release.

@helgeerbe - how soon are you cutting a new release? I still have some WIP changes here to address some of the things discussed in https://github.com/helgeerbe/picframe/issues/203, but they're not quite ready. I might be able to get them ready this weekend if it matters, but they're not critical and probably don't need to hold up your plans. Let me know.

helgeerbe, 2022-01-15:
@jgodfrey I'm not in a rush. Just tell me when you are done.