helgeerbe/picframe #227: HDMI-CEC option

dexn, 2022-01-15:
Hi,

A feature request I have is to have an option to utilise CEC with "display_is_on".

This would require "cec-utils" from apt to be installed. 

I was thinking of an option in "configuration.yaml" like "use_cec: True". When "display_is_on=true" it runs an addition step and executes "echo 'on 0.0.0.0' | cec-client -s -d 1" or for "display_is_on=false" it executes "echo 'standby 0.0.0.0 | cec-client -s -d 1".

Thank you for your great work on this.

helgeerbe, 2022-01-15:
Hi @dexn,

picframe already publishes display state changes by mqtt. Are you already running a mqtt broker like mosquitto? You could run a simple script subscribing to that topic and execute the command. Or you use node-red. Both could be installed parallel to picframe on a pi3.

helgeerbe, 2022-03-24:
Close this issue, because of no feedback.

closed 2022-03-24 by helgeerbe