helgeerbe/picframe #228: Home Assistant - Picture shown on mobile but not on desktop

sapnho, 2022-01-15:
This is one for @helgeerbe and it's about the Home Assistant integration, which works really great, and I am glad every day that we have it!

But, here is a curious thing: Some time ago, the image stopped appearing in HA when using the iMac browser. It does appear when accessed through the mobile apps and even through the mobile browsers (iPad and iPhone).

On my iMac browser, I just see this:
[image]

This worked a few months before and I don't remember that I changed anything. Could this be a firewall issue?

sapnho, 2022-01-15:
I just tried it with another browser than Chrome, and sure enough, it works on Safari, Opera, and Firefox. But not on Chrome desktop. It does work on Chrome mobile though.

helgeerbe, 2022-01-15:
I don't think that it belongs to Home Assistant. What happens if you use the picframe's webinterface and click on "see current image". The url used is the same as in Home Assistant.

sapnho, 2022-01-15:
Yes indeed, it must be a Chrome browser issue. But I have been searching high and low for a possible reason. Maybe somebody on this forum has figured it out. I deactivated all extensions, checked any settings that might forbid images but to no avail.

sapnho, 2022-01-15:
When I enter something like
`http://192.168.178.67:9000/2013.%2008.18%20-%2011.30.%2029%20Bilderrahmen.jpg`
the picture is shown in Chrome.

sapnho, 2022-01-15:
And even when I right-click on the broken image icon, and say "open image in new tab", it works. It just refuses to show it in Home Assistant.

helgeerbe, 2022-01-15:
The rendered webpage from home assistant looks totally fine to me. Which version have you installed? I'm running Home Assistant 2021.12.8

sapnho, 2022-01-15:
Running 2021.12.9 And just to make sure, you are using Google Chrome on a Mac?

sapnho, 2022-02-06:
This issue can be closed. 
I have no idea why it isn't working on Chrome but since it's working in all other browsers, it clearly is an issue outside of HA and Pi3D.

closed 2022-02-07 by helgeerbe

sapnho, 2022-02-09:
I found it! It was a specific site setting in Google Chrome where you had to explicitly allow images. I made a note in the article: https://www.thedigitalpictureframe.com/integrate-raspberry-pi-picture-frame-home-assistant/#Troubleshooting

For some strange reason it wasn't allowed for the Home Assistant page.

jgodfrey, 2022-02-09:
Nice find @sapnho!