helgeerbe/picframe #231: Illegal instruction error

nimitjainFireLab, 2022-01-17:
Hello

I was following the instructions for installation from here (https://www.thedigitalpictureframe.com/how-to-add-crossfading-slide-transitions-to-your-digital-picture-frame-using-pi3d/) for Raspberry pi 2, running buster. Everything went fine till I tried to launch picframe using "picframe -i ." for initial configuration. Entering that just gives me an Illegal instruction error. Why would this be?

Thanks

paddywwoof, 2022-01-17:
What error message do you get when you try to import on command line

pip3 install pi3d --user

Say, or ninepatch?

nimitjainFireLab, 2022-01-17:
pi3d and ninepatch can be installed fine using pip3. I meant that when I open the python3 interpreter (by just typing python3 on the command line) and type in "import pi3d" or "import ninepatch," I get an illegal instruction error and the python3 interpreter quits.

paddywwoof, 2022-01-17:
I wonder if it's related to https://github.com/numpy/numpy/issues/18131#issuecomment-1003746152

What version of numpy do you now have installed (and what did pip3 get you)? And does the pi3d error happen now you have a different numpy?

nimitjainFireLab, 2022-01-17:
When I installed picframe initially, numpy-1.21.5 was installed as a dependency. This was giving an illegal instruction error when I tried importing numpy as a module using the python3 interpreter. I then did "pip3 uninstall numpy" followed by "sudo apt install python3-numpy." Now, when I import numpy using the python3 interpreter, it works fine. I did not test whether pi3d or ninepatch were importing fine before I changed the numpy version. 

Thanks for checking about PIL further. It indeed is not working correcting:
>>> from PIL import image
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'image' from 'PIL' (/home/pi/.local/lib/python3.7/site-packages/PIL/__init__.py)

paddywwoof, 2022-01-17:
Hmm, that looks like just the difference between `image` and `Image`. Could you try with an upper case i? I only suggested checking that because `PIL.Image` is a dependency of ninepatch.

I take it from your comment

> I did not test whether pi3d or ninepatch were importing fine before I changed the numpy version.

that when you try to import either of them now you get the 'illegal instruction' and interpreter crash?

nimitjainFireLab, 2022-01-17:
Sorry about the case error! "from PIL import Image" also gives me an illegal instruction error.
And yes, when I now import either pi3d or ninepatch, I get 'illegal instruction' and interpreter crashing. 

pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pi3d
Illegal instruction
pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ninepatch
Illegal instruction
pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from PIL import Image
Illegal instruction
pi@raspberrypi:~ $ 

How can I check that my hardware architecture is compatible with all these modules?

paddywwoof, 2022-01-17:
It looks like it's the PIL Image import that's causing the problem as pi3d and ninepatch both use that (as does picframe in lots of places). It is possible to set up pi3d to not use PIL but that seems to being circumvented by the the fact that it produces too serious an error to catch in a try block https://github.com/tipam/pi3d/blob/1f1c822dc3ac58344ad2d5468978d62e51710df4/pi3d/constants/__init__.py#L63

What version do you get if you do
```python
>>> import PIL
>>> PIL.__version__
'4.3.0'
>>> 
```
The whole thing does seem odd. You can (or used to be able. ED - just checked and it seems to be in buster https://packages.debian.org/source/buster/pillow) install a debian package for PIL
```terminal
sudo apt install python3_pil
```

nimitjainFireLab, 2022-01-17:
I get
pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import PIL
>>> PIL.__version__
'9.0.0'
>>> 

Should I uninstall this PIL and get the debian package instead? That's kind of what I did for numpy

paddywwoof, 2022-01-17:
Well it would give a bit more info whether it works or not.

nimitjainFireLab, 2022-01-17:
Sweet! Uninstalling that pillow version and installing the debian package resolved everything. I also got the "Have a Break" photo with picframe. Thanks so much for your help and patience!

paddywwoof, 2022-01-18:
Thanks for reporting the problem. It looks like there might be an issue with PIL 9.0.0 on arm 6 at least so I had better do a bit more checking on that. But the fact that you had a problem with numpy as well makes me think that both are related to something else.

helgeerbe, 2022-01-18:
I'm running PIL 9.0.0 without an issue. For the next release it changed the dependency to this version because it fixes #182 

I simply run `pip3 install Pillow --upgrade`

helgeerbe, 2022-03-24:
Close this, because of no feedback.

closed 2022-03-24 by helgeerbe