helgeerbe/picframe #234: Image meta info not found, when updating cache and new images not completely written to folder.

helgeerbe, 2022-01-24:
I added yesterday some new images to my picture folders. 
After adding new images to the cache, I found that some images don not have any meta info in the db.
I assume, that those images are not completly written, while they are imported to the db in parallel. 
Workaround is to stop picframe, move images to picture folder and restart picframe.

helgeerbe, 2022-01-24:
Maybe somehow related to this? https://stackoverflow.com/questions/19059877/python-os-path-getmtime-time-not-changing/19159371
Would It be good to store also the file size?

jgodfrey, 2022-01-24:
So, the offending images were brand new to the system? And, they were added to the `file` table, but didn't nave any `meta` table info? That's odd, as the meta info should be collected and inserted at the time the file record is added. The only thing that's delayed is the location look-up, which doesn't happen until the first time the image is displayed (unless we already have location info from another image taken from the same geoloc).

If the above description is correct, I'm not sure why restarting the app would make any difference. Did I misunderstand the situation?

helgeerbe, 2022-01-24:
What do you think about a delayed (e. g. 2 seconds) caching. So modification time must be older than 2 seconds before caching?

jgodfrey, 2022-01-24:
> What do you think about a delayed (e. g. 2 seconds) caching.

Yeah, based on the SO link you provided above, that could make some sense. That said, it'll definitely take some jiggery-pokery in the caching mechanism to support such a delay.

helgeerbe, 2022-01-25:
In 

````python
def __get_modified_folders(self):
        out_of_date_folders = []
        sql_select = "SELECT * FROM folder WHERE name = ?"
        ts = int(time.time()) - 10
        for dir in [d[0] for d in os.walk(self.__picture_dir, followlinks=self.__follow_links)]:
            if os.path.basename(dir)[0] == '.': continue # ignore hidden folders
            mod_tm = int(os.stat(dir).st_mtime)
            found = self.__db.execute(sql_select, (dir,)).fetchone()
            if mod_tm < ts and (not found or found['last_modified'] < mod_tm or found['missing'] == 1):
                out_of_date_folders.append((dir, mod_tm))
        return out_of_date_folders
````

You are casting `mod_tm` to `int`. Is there a reason why? So files in this folder could changed for nearly a second without updating the cache.

jgodfrey, 2022-01-25:
> You are casting mod_tm to int. Is there a reason why?

Hmmm... Good question. I can't think of a good reason. Looking, that cast exists even in my original experimental repo for this code - prior to it being added to `picframe`. No idea really, but it seems incorrect to me too...

paddywwoof, 2022-01-25:
Not sure what it gives. I think `time_t` which is a signed integer anyway.

I suppose there's nothing to stop `__get_modified_files()` from being called while the os is copying files to the directory. Maybe the `self.__modified_folders` list should be left until the next call of `update_cache()` we're storing it as a class property, probably that was the original idea. We might be able to just
```python
        if not self.__modified_files:
            self.__logger.debug('No unprocessed files in memory, checking disk')
            self.__modified_files = self.__get_modified_files(self.__modified_folders)
            self.__modified_folders = self.__get_modified_folders() # these will be processed next time!
            self.__logger.debug('Found %d new files on disk', len(self.__modified_files))

```

helgeerbe, 2022-01-26:
`time.time()` is float. In Python 3, there's a new a attribute called st_mtime_ns, which is st_mtime in nanoseconds. try it.

>>> os.stat('.').st_mtime
1381571932.044594
>>> os.stat('.').st_mtime_ns
1381571932044593972

Changing precision and moving file update to the next update cycle seems reasonable to me. Maybe to stay with float could do it, if the delay is long enough. So, we don't have to refresh the timestamps in the database.

paddywwoof, 2022-01-26:
Hmm, just to check how the values changed I just tried
```python
import os
import time

for i in range(20):
    os.system('touch junk/junk.txt')
    #with open('junk/junk.txt', 'a') as f:
    #    f.write('x')
    folder_stats = os.stat('junk')
    file_stats = os.stat('junk/junk.txt')
    print(folder_stats.st_mtime, folder_stats.st_mtime_ns, file_stats.st_mtime)
    time.sleep(0.1)
```
Which seems odd. Running the script several times doesn't change the directory mtime, and appending characters to the file rather than simple `touch` doesn't make any difference (even with `f.flush()`). If I touch the file in command line it doesn't change from previous value, but if I open the file with nano it does change, so long as I add and delete a character. So my conclusion is that there is probably more to the folder stats than meets the eye.

helgeerbe, 2022-01-26:
I just tested it. Changing existing files does not have any impact on the folder stats. Changes are only reflected on file level. Just adding/deleting/renaming files will be detected.

What does that mean:
- changing exif info e. g. with exif_read will not be detected unless you call `touch parent_folder`
- update files somewhere else and copy it to the picture folder will be detected, even when saved with an existing name

So we need a solution to handle already created files, which are not completely written while updating the cache.

jgodfrey, 2022-01-26:
Some related discussions can be found here -> https://github.com/helgeerbe/picframe/issues/96#issuecomment-913706728

helgeerbe, 2022-01-27:
I don't know if we should store `last_modified` in nano seconds instead of float of a second. Or maybe we should also store file size as a second dirty cache indicator?

helgeerbe, 2022-03-24:
Should be fixed with 2022.03.24.

closed 2022-03-24 by helgeerbe