helgeerbe/picframe #251: FR: Stop slideshow while screen is off

wormiedk, 2022-03-02:
From the MQTT state it seems as if picframe continues to change the background image even though the screen is off. It would be great if the slideshow was paused while the screen is turned off :)

paddywwoof, 2022-03-02:
Yes, it should pause automatically IMO. I'm away now but might submit a change along these lines unless anyone can think why it would be a bad idea.

jgodfrey, 2022-03-02:
Nope, I agree 100%. In fact, I run a cron job that turns my frame ON in the morning and OFF at night. In each case, I send two MQTT commands - one to pause/unpause the slideshow and one to turn off/on the screen.

helgeerbe, 2022-03-02:
I don't care. During night I shut down my frame and I have always shuffled play enabled. So I Never know where in the queue my frame is.

jgodfrey, 2022-03-02:
I don't really care about the slideshow itself (like, what picture I last saw and what I might see first thing in the morning). Really, it's more a matter of "there's no reason for the rpi to work hard when the screen is off". That, and the reduction in unnecessary memory card I/O can't be bad either.

I guess the real question would be - is there any legitimate reason to keep the slideshow running when the screen is off? I can't really think of one...

helgeerbe, 2022-03-07:
I changed  `controller.py`

````python
@display_is_on.setter
    def display_is_on(self, on_off):
        self.paused = not on_off
        self.__viewer.display_is_on = on_off
````

helgeerbe, 2022-03-24:
Implemented in 2022.03.24.

closed 2022-03-24 by helgeerbe