helgeerbe/picframe #252: MQTT command picframe/brightness changes brightness but also changes image.

koto60, 2022-03-04:
I was running an old version, and my MQTT setup worked fine. Updated the version, and now the image changes every time in change the brightness. Even when done through command line with: 

`mosquitto_pub -h localhost -t picframe/brightness -m "0.3"`

[config.txt](https://github.com/helgeerbe/picframe/files/8182871/config.txt)

koto60, 2022-03-04:
It is due to some change made in 2022.01.20

I downgraded and now it works as expected

helgeerbe, 2022-03-07:
I changed this behaviour on purpose. On image change, picframe sends out notifications on image meta info, and other settings like brightness. I thought one will change rarely the brightness, so it might be OK, to use the notification message on image change. This is needed to inform all other subscribers, of the change.

Do you have an use case, where your freqently change the brightness or other parameters? Then it might be a good idea to separate image meta info, from config info. This requires an additional communication flow.

koto60, 2022-03-07:
Ah I can see the change is even described in the change log, sorry for that. I do not often adjust brightness, but when I do it is in the evening when the screens might become too bright. Especially when I turn off lights in the room.

The main issue is that I use the MQTT Dash app for control. And the Range/Progress button there only works(as far as i know) with a "on_display" event. This means that It will send many brightness change requests, as long as I have the screen open, and this makes the display completely spas out.

Also I use the screens to display art I enjoy, and because of this I have set a pretty big "time_delay" value. If I want to adjust the brightness a few times to get the right value, I will skip past the image I am currently viewing, and others I might like to view. It is not so much a problem, as an annoyance for me since I will have to wait a long time to see the images I skipped. 

But I can see the reason for the change. Maybe I should just try some other similar MQTT service that does not work this way.

helgeerbe, 2022-08-02:
Should be solved with [2022.06.03](https://github.com/helgeerbe/picframe/wiki/Changelog#20220603)

closed 2022-08-02 by helgeerbe