helgeerbe/picframe #255: AttributeError: 'DisplayOpenGL' object has no attribute 'context'

cnichte, 2022-03-22:
I have setup pic frame on a fresh raspberry pi4, with an Display on the HDMI Port:

- PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
- NAME="Raspbian GNU/Linux"
- VERSION_ID="11"
- VERSION="11 (bullseye)"
- VERSION_CODENAME=bullseye
- ID=raspbian
- ID_LIKE=debian

First launch from the Terminal gets me the following Errors:

```
picframe
INFO:start.py:starting ['/usr/local/bin/picframe']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
Traceback (most recent call last):
  File "/usr/local/bin/picframe", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/picframe/start.py", line 136, in main
    c.start()
  File "/usr/local/lib/python3.9/dist-packages/picframe/controller.py", line 326, in start
    self.__viewer.slideshow_start()
  File "/usr/local/lib/python3.9/dist-packages/picframe/viewer_display.py", line 426, in slideshow_start
    self.__display = pi3d.Display.create(x=self.__display_x, y=self.__display_y,
  File "/usr/local/lib/python3.9/dist-packages/pi3d/Display.py", line 569, in create
    display.opengl.create_display(x, y, w, h, depth=depth, samples=samples, layer=layer,
  File "/usr/local/lib/python3.9/dist-packages/pi3d/util/DisplayOpenGL.py", line 120, in create_display
    assert self.context != EGL_NO_CONTEXT and self.context is not None
AttributeError: 'DisplayOpenGL' object has no attribute 'context'
```

What's going wrong here?

best regards

paddywwoof, 2022-03-22:
Hi, is the problem related to this discussion https://github.com/helgeerbe/picframe/issues/195 i.e. needing to setup the driver to work with bullseye?

helgeerbe, 2022-03-23:
I guess, you should try this:
https://github.com/helgeerbe/picframe/issues/195#issuecomment-982999227 if you want to keep bullseye.

TopDown71, 2024-06-03:
Yesterday I tried to do a clean build on Pi3B with Buster using the instructions at TheDigitalPictureFrame. I am getting similar messages as reported here and elsewhere. The OS is 2021-05-07-raspios-buster-arm64

cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian

### Here is the output from the picframe install:
**sudo python3 -m pip install -U pip**
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pip
  Downloading https://files.pythonhosted.org/packages/8a/6a/19e9fe04fca059ccf770861c7d5721ab4c2aebc539889e97c7977528a53b/pip-24.0-py3-none-any.whl (2.1MB)
    100% |████████████████████████████████| 2.1MB 135kB/s 
Installing collected packages: pip
  Found existing installation: pip 18.1
    Not uninstalling pip at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'pip'. No files were found to uninstall.
Successfully installed pip-24.0
Dan@PhotoShow:~ $ sudo python3 -m pip install pi-heif
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pi-heif
  Downloading pi_heif-0.12.0-cp37-cp37m-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (5.5 kB)
Collecting pillow>=8.4.0 (from pi-heif)
  Downloading Pillow-9.5.0-cp37-cp37m-manylinux_2_28_aarch64.whl.metadata (9.5 kB)
Downloading pi_heif-0.12.0-cp37-cp37m-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (817 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 817.9/817.9 kB 2.3 MB/s eta 0:00:00
Downloading Pillow-9.5.0-cp37-cp37m-manylinux_2_28_aarch64.whl (3.2 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.2/3.2 MB 3.5 MB/s eta 0:00:00
Installing collected packages: pillow, pi-heif
  Attempting uninstall: pillow
    Found existing installation: Pillow 5.4.1
    Uninstalling Pillow-5.4.1:
      Successfully uninstalled Pillow-5.4.1
Successfully installed pi-heif-0.12.0 pillow-9.5.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Dan@PhotoShow:~ $ pip3 install picframe
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting picframe
  Downloading https://www.piwheels.org/simple/picframe/picframe-2024.5.31-py3-none-any.whl (3.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.2/3.2 MB 2.5 MB/s eta 0:00:00
INFO: pip is looking at multiple versions of picframe to determine which version is compatible with other requirements. This could take a while.
  Downloading https://www.piwheels.org/simple/picframe/picframe-2024.2.19-py3-none-any.whl (3.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.2/3.2 MB 1.6 MB/s eta 0:00:00
  Downloading https://www.piwheels.org/simple/picframe/picframe-2023.7.26.post1-py3-none-any.whl (3.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.2/3.2 MB 2.2 MB/s eta 0:00:00
Requirement already satisfied: Pillow==9.5.0 in /usr/local/lib/python3.7/dist-packages (from picframe) (9.5.0)
Collecting defusedxml (from picframe)
  Downloading https://www.piwheels.org/simple/defusedxml/defusedxml-0.7.1-py2.py3-none-any.whl (25 kB)
Collecting pi3d>=2.49 (from picframe)
  Downloading pi3d-2.51-py3-none-any.whl.metadata (686 bytes)
Collecting PyYAML (from picframe)
  Downloading PyYAML-6.0.1-cp37-cp37m-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (2.1 kB)
Collecting paho-mqtt (from picframe)
  Downloading https://www.piwheels.org/simple/paho-mqtt/paho_mqtt-2.1.0-py3-none-any.whl (67 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 67.2/67.2 kB 396.5 kB/s eta 0:00:00
Collecting IPTCInfo3 (from picframe)
  Downloading https://www.piwheels.org/simple/iptcinfo3/IPTCInfo3-2.1.4-py3-none-any.whl (12 kB)
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from picframe) (1.16.2)
Collecting ninepatch (from picframe)
  Downloading ninepatch-0.1.22-py3-none-any.whl.metadata (6.0 kB)
Requirement already satisfied: pi-heif>=0.8.0 in /usr/local/lib/python3.7/dist-packages (from picframe) (0.12.0)
Collecting pysdl2 (from pi3d>=2.49->picframe)
  Downloading https://www.piwheels.org/simple/pysdl2/PySDL2-0.9.16-py3-none-any.whl (580 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 580.3/580.3 kB 1.1 MB/s eta 0:00:00
Downloading pi3d-2.51-py3-none-any.whl (256 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 256.6/256.6 kB 1.7 MB/s eta 0:00:00
[... 31 lines cut ...]
    self.__viewer.slideshow_start()
  File "/home/Dan/.local/lib/python3.7/site-packages/picframe/viewer_display.py", line 433, in slideshow_start
    background=self.__background, use_glx=self.__use_glx)
  File "/home/Dan/.local/lib/python3.7/site-packages/pi3d/Display.py", line 624, in create
    use_glx=use_glx, use_sdl2=use_sdl2)
  File "/home/Dan/.local/lib/python3.7/site-packages/pi3d/util/DisplayOpenGL.py", line 116, in create_display
    assert self.context != EGL_NO_CONTEXT and self.context is not None
AttributeError: 'DisplayOpenGL' object has no attribute 'context'
^CException ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 1281, in _shutdown
    t.join()
  File "/usr/lib/python3.7/threading.py", line 1032, in join
    self._wait_for_tstate_lock()
[... cut ...] 7/threading.py", line 1048, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt

_**From Native Terminal**_
**picframe ~/picframe_data/config/configuration.yaml**
INFO:start.py:starting ['/home/Dan/.local/bin/picframe', '/home/Dan/picframe_data/config/configuration.yaml']
INFO:model.Model:Open config file: /home/Dan/picframe_data/config/configuration.yaml:
libEGL warning: DRI2: failed to authenticate
gl_id  b'GLES3' b'OpenGL ES 3.1 Mesa 19.3.2'
^CYou pressed Ctrl-c!

TopDown71, 2024-06-04:
On a whim I tried Heelgerbe's 3B/Bullseye configuration from [issue 195](https://github.com/helgeerbe/picframe/issues/195) with my Buster install (per the 3B installation instructions). In other words I just ran raspi-config to change to full KMS and changed use_glx to True. This solved the startup problems from a native terminal window and the images cycle as expected.

But: enabling a mouse for input does not work, picframe 'hangs' when starting. Note - this is a Logitech wireless mouse (I no longer have any wired mice😉)

However, enabling a keyboard does work

TopDown71, 2024-06-05:
My working config for a fresh install of PicFrame 2024.5.31 on a 3B using Wolfgang's [instructions](https://www.thedigitalpictureframe.com/how-to-add-crossfading-slide-transitions-to-your-digital-picture-frame-using-pi3d/#For_Raspberry_Pi_3_Users)  required two things. First was using full KMS in raspi-config per issue 195, while use_glx in the yaml file stayed false. The second thing fixed the mouse peripheral issue. the file 'pointer.png' was somehow missing from the pi3d icons folder located at ~/.local/lib/python3.7/site-packages/pi3d/util/icons. I found this file at the pi3d GitHub site and copied it to my pi. Now picframe starts without error and the mouse works.

Performance seems significantly slower so following suggestions in issue #100 I adjusted fps and got a reasonable level of performance back. It seems like something under the hood has changed and affected performance on the Pi3B.

sapnho, 2024-06-05:
If you like, you can try the Bookworm way: https://www.thedigitalpictureframe.com/how-to-build-the-best-raspberry-pi-digital-picture-frame-with-bookworm-and-pi3d-picframe-2024-edition/

TopDown71, 2024-06-05:
I hadn’t realized there was a better starting point now with bookworm, Thanks for pointing this out. Itt works pretty well. I had to use sudo when I issued the journalctl command to vacuum the logs. That may be because I built the system with my name as user instead of ‘pi’ but it all worked.

The pointer.png file is still missing in the icons folder so picframe hangs when you start it with the mouse enabled. I managed to copy pointer.png to the correct place and the mouse works but I did notice that all the icons that used to be png files are now gif files. In any event it still enabled the mouse and use of the menus.

Would all of this still work with the full desktop version of Bookworm?

Many thanks,
Dan