helgeerbe/picframe #258: Crash when no contact to MQTT server

wormiedk, 2022-04-06:
**Describe the bug**
Whenever picframe looses contact to the MQTT server it crashes or refuses to start

**To Reproduce**
Steps to reproduce the behavior:
1. Disconnect network from picframe or shutdown Matt server
2. Try to start picframe / wait for picture change

**Expected behavior**
Log warning that there is no contact to mqtt server

**What are your OS, Python and picframe versions(please complete the following information):?**
2022.03.24

- Hardware: raspberry pi zero
- OS: raspbian 10.0 (buster)
- Python: 2.7/3.7
- picframe (run `picframe -v` if possible): 
INFO:start.py:starting ['/home/pi/.local/bin/picframe', '-v']
picframe version:  2022.03.24

Checking required packages......
PIL :  9.0.1
exifread :  2.3.2
pi3d :  2.49
yaml :  5.4.1
paho.mqtt :  1.5.1
iptcinfo3 :  2.1.4
numpy :  1.16.2
ninepatch : installed, but no version info

helgeerbe, 2022-04-11:
Hi @wormiedk can you provide a log file with log level "DEBUG".

madface303, 2022-04-24:
I have the same problem and jump in here. At least starting without the mqtt server just shows me a blank screen, here is the debug log:
[piframe_log.zip](https://github.com/helgeerbe/picframe/files/8549703/piframe_log.zip)
I noticed picframe showed me the "no pictures screen" sometimes when i rebootet my ha server (and the mqtt server on it), but i can't reproduce this at the moment.

helgeerbe, 2022-04-24:
hi @madface303 

If you enable mqtt picframe requires the mqtt broker to be up and to be reachable on startup. I see in your logfile that the broker was not available. Problem / Bug is, that picframe doesn't quit.

Later on, temporary unavailability of the broker shouldn't be a problem. Picframe will reconnect. Of course, if the broker is down one can't control picframe by mqtt. So a reliable connection is a must have.

I pushed a patch on dev. Picframe quits now, if mqtt interface couldn't be initialize on startup.

wormiedk, 2022-04-25:
I would much prefer a "soft" crash so that the frame would start no matter if the mqtt server was available but maybe write the failure in the log.

Actually I brought the frame to a party once and had to set up mqtt and internet to get it to boot and then disable mqtt...

closed 2022-11-24 by helgeerbe